pymeasure/pymeasure #222: Keithley 2410 "NotImplementedError: Please upgrade PyVISA to version 1.8 or later."

Hendrixb5, 2019-09-03:
I am trying to interface with the Keithley 2410 Sourcemeter using an NI GPIB-USB cable, but I am getting the error above.

I have the NI-VISA installed as well as the NI-488.2 drivers. Windows and NI MAX are detecting the GPIB connection fine. I am running Jupyter Notebook in Windows 10 32-Bit.

     import pyvisa
     from pymeasure.instruments.keithley import Keithley2400

     rm = pyvisa.ResourceManager()
     print(rm.list_resources())

     sourcemeter = Keithley2400("GPIB::24")`

The last line returns the error

`NotImplementedError: Please upgrade PyVISA to version 1.8 or later.`

Any insight would be appreciated.

bilderbuchi, 2019-09-04:
What is the pyvisa version you have installed?

Hendrixb5, 2019-09-04:
It is version 1.9.0

Hendrixb5, 2019-09-04:
I attempted to just use pyvisa to open the sourcemeter and I get the error

     VisaIOError: VI_ERROR_BERR (-1073807304): Bus error occurred during transfer.

bilderbuchi, 2019-09-04:
@cjermain @hlgirard any ideas?

hlgirard, 2019-09-04:
The pyvisa message seems to suggest the error is upstream of pymeasure. If pyvisa can't even open the connection there might be something wrong with the GPIB to USB converter (my Prologix adapter has a number of internal settings that can be changed and can break the connection to pyvisa).
I would check out a few things:

1) Check the version of pyvisa that is installed in the environment used in your notebook.
`pyvisa.__version__`
It has happened to me before to install packages in the wrong environment and get frustrated that the versions didn't match.

2) Sometimes, when the GPIB connection has been opened in another program before, pyvisa won't be able to bind to it anymore. So restart your computer and Keithley. Then, before opening anything else, open your notebook and run:
```python
import pyvisa

rm = pyvisa.ResourceManager()
rm.list_resources()
scm = rm.open_resource("GPIB::24") # Or whatever address is returned by the previous command
scm.query("*IDN?")
```
If that still doesn't work then there is a problem between pyvisa and the GPIB adapter. You should probably ask in the NI forums or pyvisa github issues.

3) If the GPIB adapter doesn't work you can fall back on a serial connection if you have a USB to serial adapter lying around. For that you will have to implement the adapter -> instrument logic from pymeasure like:
```python
from pymeasure.adapters import SerialAdapter
from pymeasure.instruments.keithley import Keithley2400

adapter = SerialAdapter("COM2") # Or whatever port the connection shows up as 
sourcemeter = Keithley2400(adapter)
```
The Keithley will have to be in Serial mode and the serial communication parameters should be the default for pySerial which you can find here:
https://pyserial.readthedocs.io/en/latest/pyserial_api.html
You can pass kwargs to the `SerialAdapter()` constructor if you want to stray from the defaults.

Hendrixb5, 2019-09-09:
Thanks for the replies. The computer I was working on died and I just got the replacement. I tried again and got a different error. Something to do with a bus error, BUT I changed out the GPIB cable and it all works the way it is supposed to now. You would think for the price, these things would tell you when they are bad.

Anyway, the issue is resolved.

Hendrixb5, 2019-09-09:
Sorry to reopen this, but I tested the cable with a SourceMeter in LabVIEW on a different machine and it works fine. I am wondering if anyone has any idea as to why this would be the case?

Thanks.

adrzazga, 2021-01-13:
Hello,

This problem still exist with the latest pymeasure version 0.8.0 when the Keithley KUSB-488 GPIB/USB is connected with an instrument (e.g. 2400 model).

Kind regards

Adrian

bilderbuchi, 2021-01-13:
Can you check if this goes away if you install pyvisa 1.10 (cf https://github.com/pymeasure/pymeasure/issues/381#issuecomment-745120237) until the next pymeasure version is released?

wolfg-2020, 2021-01-15:
Now that I'm able to communicate Keithley 2000 with pyvisa, but not with pymeasure. I'm having a similar error message: NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

Here's the lines I ran with pyvisa, and it works well with the os.add_dll_directory():

import pyvisa as visa
import os
os.add_dll_directory("C:/Program Files/Keysight/IO Libraries Suite/bin")
os.add_dll_directory("C:/Program Files (x86)/Keysight/IO Libraries Suite/bin")
rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")
print(rm.list_resources())
my_instrument = rm.open_resource('GPIB1::6::INSTR')
print(my_instrument.query('*IDN?'))

but when it comes to call pymeasure commands:

from pymeasure.adapters import VISAAdapter
from pymeasure.instruments.keithley import Keithley2000
adapter = VISAAdapter("GPIB::6")
multimeter = Keithley2000(adapter)
multimeter.id

I have this error message:
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

details:
Traceback (most recent call last):
File "D:/Python/myProjects/InstrumentCont/Model2000.py", line 15, in
adapter = VISAAdapter("GPIB::6")
File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pymeasure\adapters\visa.py", line 51, in init
raise NotImplementedError("Please upgrade PyVISA to version 1.8 or later.")
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

My PyVISA was downgraded to 1.10 as Christoph advised last time. But the same error occured no matter PyVisa was 1.10 or upgraded to 1.11.3. It doesn't look like a problem of PyVISA any more. Would you please give me some suggestion ? Thank you.

bilderbuchi, 2021-01-15:
> NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

> My PyVISA was downgraded to 1.10

Just so that I understand: You get this error when you have installed pyvisa 1.10 in your environment?
With pyvisa 1.11+ this error will always occur because of the removal mentioned in the comment linked above - currently released pymeasure is not compatible with pyvisa 1.11+, but this is already fixed in master.

bilderbuchi, 2021-01-15:
Also, if pyvisa does not work without the `os.add_dll_directory` calls, I think you also have to do the same before constructing the pymeasure instruments (at least until the dll lookup is fixed/improved on the pyvisa side, if that is planned).

wolfg-2020, 2021-01-18:
Yes, I put those os.add_dll_directory calls as the first lines, and have pyvvisa vbut got the error message.
Here are the lines I ran:

import os
import pyvisa as visa

from pymeasure.adapters import VISAAdapter
from pymeasure.instruments.keithley import Keithley2000

os.add_dll_directory("C:/Program Files/Keysight/IO Libraries Suite/bin")
os.add_dll_directory("C:/Program Files (x86)/Keysight/IO Libraries Suite/bin")
rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")

print(rm.list_resources())
adapter = VISAAdapter('GPIB1::6::INSTR')
multimeter = Keithley2000(adapter)

multimeter.id

and the error message :
Traceback (most recent call last):
  File "D:/Python/myProjects/InstrumentCont/Model2000.py", line 15, in <module>
    adapter = VISAAdapter('GPIB1::6::INSTR')
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pymeasure\adapters\visa.py", line 51, in __init__
    raise NotImplementedError("Please upgrade PyVISA to version 1.8 or later.")
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

Process finished with exit code 1

and here is the pyvisa and pymeasure information I can see from pyCharm's python console:

Python 3.8.3 (tags/v3.8.3:6f8c832, May 13 2020, 22:20:19) [MSC v.1925 32 bit (Intel)] on win32
import pyvisa
pyvisa.__version__
'1.10.0'

from pymeasure.adapters import VISAAdapter
VISAAdapter.has_supported_version()
False
[image]
[image]

with the captured screen information FYI. 
Would you please give me some advice on this problem ? Thank you so much!

wolfg-2020, 2021-01-18:
It's strange that pyvisa-info shows that function 'viOpen' was not found in my visa32.dll. Is this what has caused the problem? 
Please see details from the image:
[image]

Things are a little complicated in my case: I'm working with a Keithley model 2000 multimeter, with a Keysight 82357B GPIB adapter. So I have installed driver for both. At the same time, I have installed NI-MAX on a same computer. Shall I uninstall or re-install any of those programs/drivers ? Appreciate for any suggestion. Thank you!

bilderbuchi, 2021-01-18:
And what happens if you do 
```
import visa
visa.__version__
```
There was a transition from `visa` to `pyvisa` which is where some of this complication comes from. I am trying to find out why that check would give False when it should not...

The viOpen thing is something you have to (and already did I think, cf. https://github.com/pyvisa/pyvisa/issues/385) clear up with pyvisa, there is nothing pymeasure can do about it.
Remember, pyvisa-info does not do the add_dll_directory trick that you do in your code.

bilderbuchi, 2021-01-18:
As you are not using the default library when using pyvisa directly, and it works, can you try doing this instead when creating your VISAAdapter, to use the library you need (plus of course the other stuff as you had)
```
adapter = VISAAdapter('GPIB1::6::INSTR', visa_library='C:/Windows/System32/visa32.dll')
```

wolfg-2020, 2021-01-19:
Hi, Christoph, thank you for replying.
I guess there's something wrong with my visa.  it has no attribute '__version__'.

Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: module 'visa' has no attribute '__version__'
[image]

Yes, I tried to add visa_library to VISAAdapter():

[image]

but still have this error message:
Traceback (most recent call last):
  File "D:/Python/myProjects/InstrumentCont/Model2000.py", line 15, in <module>
    adapter = VISAAdapter('GPIB1::6::INSTR', visa_library="C:/Windows/System32/visa32.dll")
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pymeasure\adapters\visa.py", line 51, in __init__
    raise NotImplementedError("Please upgrade PyVISA to version 1.8 or later.")
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.
('GPIB0::11::INSTR', 'GPIB0::12::INSTR', 'GPIB0::13::INSTR', 'GPIB1::11::INSTR', 'GPIB1::5::INSTR', 'GPIB1::6::INSTR')

Process finished with exit code 1
[image]

bilderbuchi, 2021-01-19:
Is it possible that you accidentally have the "other" visa package (https://github.com/visa-sdk/visa-python, for the credit card API) installed? Can you check? 

If yes (check with your package manager - probably pip), please uninstall it if you can.

If no, I am at a loss what is going on, and I can only ask you to wait for the soon-to-come pymeasure 0.9 release which at least fixed that troublesome version check.
If you feel adventurous, you can in this case also navigate to [this method](https://github.com/pymeasure/pymeasure/blob/v0.8/pymeasure/adapters/visa.py#L70-L76) in your pymeasure installation files, comment the content out and make the function always return `True`. This way, under the assumption that the check is erroneously malfunctioning, we can at least check if your viOpen issue is fixed by the add dll invocations

wolfg-2020, 2021-01-22:
Hi, Christoph,

As checked, my visa directory is in this folder:
D:\Python\myProjects\InstrumentCont\venv\Lib\site-packages\visa
and the files in it are exactly the same as the page you mentioned. (I've even checked the program lines in those sub folders, nothing particular has been noticed.
Is this all that you wanted to confirm ? or there's any other to check ?

Yes, I installed pyvisa/pymeasure by pip. Do you mean to uninstall my pip? Shall it be reinstalled after that ?

as for pymeasure, the VISAAdapter() and has_supported_version() is in this file:
D:\Python\myProjects\InstrumentCont\venv\Lib\site-packages\pymeasure\adapters\visa.py

Is this what you expect ?

from 
    def has_supported_version():
        """ Returns True if the PyVISA version is greater than 1.8 """
        if hasattr(visa, '__version__'):
            return parse_version(visa.__version__) >= parse_version('1.8')
        else:
            return False

I'm not surprised when I got a "False", because as checked from
D:\Python\myProjects\InstrumentCont\venv\Lib\site-packages\visa\version.py
my visa(not pyvisa) version is 1.0.0. 

as a matter of fact, I'm quite confused that there're more than one "visa" on my computer. Which one will this "has_supported_version()" check ? 

Would you please tell me where I can check this point ? Thank you ?

wolfg-2020, 2021-01-22:
Hi, Christoph,

After editting has_supported_version() and always give a TRUE to its output, I got something new. This is the program I tried to run:
import os
import pyvisa as visa

from pymeasure.adapters import VISAAdapter
from pymeasure.instruments.keithley import Keithley2000

os.add_dll_directory("C:/Program Files/Keysight/IO Libraries Suite/bin")
os.add_dll_directory("C:/Program Files (x86)/Keysight/IO Libraries Suite/bin")
rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")

print(rm.list_resources())
adapter = VISAAdapter('GPIB1::6::INSTR', visa_library = "C:/Program Files/Keysight/IO Libraries Suite/bin")
multimeter = Keithley2000(adapter)

multimeter.id

[image]

and it resulted as:
D:\Python\myProjects\InstrumentCont\venv\Scripts\python.exe D:/Python/myProjects/InstrumentCont/Model2000.py
('GPIB0::11::INSTR', 'GPIB0::12::INSTR', 'GPIB0::13::INSTR', 'GPIB1::11::INSTR', 'GPIB1::5::INSTR', 'GPIB1::6::INSTR')
Traceback (most recent call last):
  File "D:/Python/myProjects/InstrumentCont/Model2000.py", line 15, in <module>
    adapter = VISAAdapter('GPIB1::6::INSTR', visa_library = "C:/Program Files/Keysight/IO Libraries Suite/bin")
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pymeasure\adapters\visa.py", line 57, in __init__
    self.manager = visa.ResourceManager(visa_library)
AttributeError: module 'visa' has no attribute 'ResourceManager'

Process finished with exit code 1
[image]

This means it is not calling pyvisa but the visa that has no "ResouceManager" although I'm importing pyvisa as visa, doesn't it? 
It's becoming interesting. What would you suggest me to do next ? Thank you for the comments.~~

bilderbuchi, 2021-01-22:
Good troubleshooting!

One problem here is that this visa-python package (the credit card one) is installed and is. Use `pip list` to check if pip sees this, and what the exact name is (I suspect `visa-python`), then uninstall it using pip (this is what I meant above, sorry for being unclear) - `pip uninstall name-here`. This should already avoid a couple of failure scenarios.

Another thing you need to change is give the `VISAAdapter` initialiser `visa_library` argument the same string that you give to `ResourceManager` (it gets passed along internally) - `"C:/Windows/System32/visa32.dll"`

Btw, you can use three backticks(```) to start and end code blocks which get nicely formatted, then there is no need for screenshots, see https://docs.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax#quoting-code

wolfg-2020, 2021-01-22:
Hi, Christoph,

This is the list pip sees: there are a pyvisa 1.10.0 and a visa 1.0.0 as mentioned before . I think visa 1.0.0 is the one has no ResouceManager(), and has caused FALSE in has_supported_version(). Do you mean to uninstall this visa 1.0.0 ?

Package           Version
----------------- ---------
-ip               20.3.1
certifi           2020.12.5
chardet           3.0.4
idna              2.10
numpy             1.19.4
pandas            1.1.5
pip               20.3.3
PyMeasure         0.8.0
PyQt6             0.1
pyqtgraph         0.11.0
pyserial          3.5
python-dateutil   2.8.1
pytz              2020.4
PyVISA            1.10.0
requests          2.25.0
setuptools        51.1.0
six               1.15.0
typing-extensions 3.7.4.3
urllib3           1.26.2
visa              1.0.0
wheel             0.36.2

[image]

and thanks for the tips.~~

wolfg-2020, 2021-01-22:
Hi, Christoph,

I tried to uninstall visa 1.0.0, and modified my code as
'''
import os
import pyvisa as visa

from pymeasure.adapters import VISAAdapter
from pymeasure.instruments.keithley import Keithley2000

os.add_dll_directory("C:/Program Files/Keysight/IO Libraries Suite/bin")
os.add_dll_directory("C:/Program Files (x86)/Keysight/IO Libraries Suite/bin")
rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")

print(rm.list_resources())
adapter = VISAAdapter('GPIB1::6::INSTR', visa_library = "C:/Windows/System32/visa32.dll")
multimeter = Keithley2000(adapter)

multimeter.id
'''

I didn't get the error message, but just a future warning: 

D:\Python\myProjects\InstrumentCont\venv\Scripts\python.exe D:/Python/myProjects/InstrumentCont/Model2000.py
('GPIB0::11::INSTR', 'GPIB0::12::INSTR', 'GPIB0::13::INSTR', 'GPIB1::11::INSTR', 'GPIB1::5::INSTR', 'GPIB1::6::INSTR')
D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\resources\messagebased.py:106: FutureWarning: values_format is deprecated and will be removed in 1.10
  warnings.warn('values_format is deprecated and will be removed in '

Process finished with exit code 0

bilderbuchi, 2021-01-22:
Yes, exactly, uninstall `visa`! Glad to see that it works now! 
(Note: you'll need to use `print(multimeter.id)` that the script shows you the ID)

closed 2021-01-22 by bilderbuchi