pymeasure/pymeasure #225: Error creating QT GUI when FloatParameter minimum >=0.005

jeffmincey, 2019-09-11:
pymeasure==[0.7.0](https://github.com/ralph-group/pymeasure/tree/v0.7)

if a FloatParameter param "minimum" is set to a value greater than 0.005 as the following: 
```
delay = FloatParameter('Delay Time', units='s', minimum=.1, maximum=5, default=0.2)
```

then QT gui will fail with the following error:
```
Traceback (most recent call last):
  File "...\lib\site-packages\pymeasure\display\inputs.py", line 250, in validate
    if self._parameter.units:
AttributeError: 'NoneType' object has no attribute 'units'
```

---

Also, likely related, if minimum is set >0 and <0.005 by adding the following test case to TestScientificInput parametrize'd inputs
```
[0.004, 5.5, 3.3],
```
then this error is thrown in pytest, however the Qt GUI behaves correctly.
```
>       assert sci_input.minimum() == min_
E       assert 0.0 == 0.004
E        +  where 0.0 = <built-in method minimum of ScientificInput object at 0x0000022AA25C5288>()
E        +    where <built-in method minimum of ScientificInput object at 0x0000022AA25C5288> = <pymeasure.display.inputs.ScientificInput object at 0x0000022AA25C5288>.minimum

tests\display\test_inputs.py:155: AssertionError
```

related to:  #118, #115

jeffmincey, 2019-09-11:
just in case, here's the version of Qt I'm using:
```
In [2]: pyqtgraph.Qt.VERSION_INFO
Out[2]: 'PyQt5 5.9.2 Qt 5.9.6'
```

cjermain, 2020-01-17:
Thanks for reporting this @jeffmincey. I'll take a look at adding these test as soon as time allows. If anyone else is interested, PRs are welcome.

bilderbuchi, 2020-10-20:
I think this is also fixed with #278.

closed 2020-10-20 by bilderbuchi