pymeasure/pymeasure #229: [Keithley 2450] Cannot apply or measure current, but connection works

Synthos, 2019-10-18:
Hi everyone,

I have an issue with a Keithley 2450. I am able to open a connection via pyVISA, functions like beep() work just find. Here is my code

`self.instrument = Keithley2450('USB0::0x****::0x****::********::INSTR')`
`self.instrument.beep(2000,1)`
`self.instrument.measure_voltage()`

But I read back Rone. The Keithley display, during this remote operation does not show any data. (see image)

When deactivate the remote control and activate e.g. Ohmmeter in the quick settings, the data is displayed as expected. Sending a remote command:
`self.instrument.measure_resistance()` returns None, while `self.instrument.write(":READ?")` returns the value displayed before on the screen.

Summarised: basic commands like ":READ?" or enable_source() work, but not the read back functions nor the apply voltage/current functions. Does anybody know why? I would very much appreciate your help :)

Note: the image shows how I was able to set the source_voltage and having the out put enabled without any current being actually applied...
[image]

cjermain, 2020-01-17:
@Synthos, are you seeing any log messages when running `self.instrument.measure_resistance()`?

bilderbuchi, 2021-02-16:
Closing due to no reply from reporter, feel free to re-open.

closed 2021-02-16 by bilderbuchi