pymeasure/pymeasure #233: FloatParameter default<0.01 changes to 0

moritzj29, 2019-11-08:
When setting the default value of a `FloatParameter` to values smaller than `0.01`, the initial value in the GUI just states `0`. This happens also for exponential notations, e.g. `1e-6`. When setting the value in the GUI everything works as expected. So it has to be related to the initial setting of the value.

I tried tracing down the error but I had no success so far. The change happens in the `set_parameter` method of the `Input` class in `inputs.py`. The value passed to `self.setValue(parameter.default)` is fine. Reading the set value with `self.value()` directly afterwards gives `0`.

relevant part of the Input class code:
```
class Input(object):
    """
    Mix-in class that connects a :mod:`Parameter <.parameters>` object to a GUI
    input box.
    :param parameter: The parameter to connect to this input box.
    :attr parameter: Read-only property to access the associated parameter.
    """

    def __init__(self, parameter, **kwargs):
        super().__init__(**kwargs)
        self._parameter = None
        self.set_parameter(parameter)

    def set_parameter(self, parameter):
        """
        Connects a new parameter to the input box, and initializes the box
        value.
        :param parameter: parameter to connect.
        """
        self._parameter = parameter

        if parameter.default is not None:
            # here the parameter.default value is correct
            self.setValue(parameter.default)
            # here self.value() gives 0

        if hasattr(parameter, 'units') and parameter.units:
            self.setSuffix(" %s" % parameter.units)
```

Going further down into QtGUI's setValue method exceeds my python/programming skills... I have no idea how this works.

Its interesting that the number of decimals of the `QDoubleSpinBox` is set to 2. (by default?) This would explain the rounding to 0. But on the other hand, this value remains 2 when values with higher precision are entered in the GUI and these are displayed and passed correctly.

It's actually not a huge problem, since the values can still be set in the GUI. But it would be handy...

cjermain, 2020-01-17:
Thanks for reporting this @moritzj29. This appears to be related to #225. I will look at this problem as time allows, but certainly welcome a PR from others.

moritzj29, 2020-10-20:
should be fixed in #278, so closing this issue

closed 2020-10-20 by moritzj29