pymeasure/pymeasure #235: Issue with file access in the "using the plotter " example

badarirao, 2019-11-09:
Hi,
I have just begun to try the graphical interface part of the package. I see that when I run the [1st example](https://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html) in the "using the plotter" tutorial, the program runs fine in spyder (windows 10 machine). 
However, after the program has finished running, when I try to open the "random.csv" file, I get the following notice.
[image]

I have just started with this package, and not sure where exactly to look for the source of the problem. It feels like the file has not been properly closed by the code. I even tried to reduce the timeout in worker.join to 60 seconds, but the problem persists.
The only workaround I have found is that I have to restart the kernel in sypder, or close it, only then the file can be accessed properly.
Is this a bug or am I supposed to do something?

Also, this is the first example I have checked. Similar problem could be occurring in other examples as well!

cjermain, 2020-01-17:
@badarirao, thanks for raising this issue. When you are running the example are you doing so in an interactive terminal within Spyder?

For background, a `Recorder` object keeps a handle on the file as the process is running -- this is spawned by the `Worker` object. It runs in the background and appends data to the relevant files for the `Results` object. It's likely that this thread is still going in the background if you are using a interactive interpreter.
https://pymeasure.readthedocs.io/en/latest/tutorial/procedure.html#running-procedures

badarirao, 2020-01-17:
@cjermain, Yes I am running the example within Spyder.
Is there a way to stop the thread in Spyder without having to restart the kernel?

cjermain, 2020-01-18:
@badarirao, I've made a change in #245 that I suspect will fix the issue. Can you please test this change to see if it resolves the problem?

badarirao, 2020-01-19:
@cjermain, Hi, I tried the example after the change, however the problem persists.
After running the program, worker.is_alive() returns False, but opening the file still gives the same message.
I also tried invoking worker.recorder.stop() separately, however, it gave an error: "AttributeError: 'NoneType' object has no attribute 'join'", pointing to error in logging\handlers.py, line 1503, in stop self._thread.join()

Also, I verified that running the program from command prompt does not give any problem. However, python gave a warning: "QObject::~QObject: Timers cannot be stopped from another thread" Does this have anything to do with the above problem?

cjermain, 2020-01-22:
Thanks for checking @badarirao. Sorry to hear that it was not that simple.

Can you post the full warning that you saw regarding the `QObject` destructor?

badarirao, 2020-01-22:
@cjermain, Hi, I have pasted the screenshot of the log and warning messages seen when running the example from windows-command prompt. There were no other warnings.
[image]

bilderbuchi, 2020-10-20:
@CasperSchippers can you take a look at this, maybe you know what's going wrong?

CasperSchippers, 2020-10-22:
I don't know what the problem, but I will dig into it.

CasperSchippers, 2020-10-23:
It seems that the `close()` method of the file-handler in the recorder is not called. Using `worker.recorder.handlers[0].close()` the file is manually closed and thereby unlocked.
I will try to find out where if and where it should be called.

bilderbuchi, 2020-12-31:
Closed by #370.

closed 2020-12-31 by bilderbuchi