pymeasure/pymeasure #237: Instrument.control unreliable response problem

hududed, 2019-12-12:
Hello

I am trying to develop a new class for the instrument Lighthouse Photonics Sprout laser. The instrument reads and writes ASCII commands e.g. `OPMODE?` to check the operating mode status and `OPMODE=ON` to set the laser on.

```
from pymeasure.instruments import Instrument, RangeException
class Sprout(Instrument):

    op_mode = Instrument.control(
        "OPMODE?", "OPMODE=%s",
        """ A string property that controls the operation mode of
        the laser. Strings allowed 'ON','OFF','IDLE','CALIBRATE'.
        Remember to open shutter for laser on.
        It takes 3-5 minutes for laser to turn on.
        e.g. Laser on `OPMODE=ON`, Laser off `OPMODE=OFF`.
        """
    )
    power = Instrument.control(
        "POWER?", "POWER SET=%.2g",
        """ A float property that controls the laser power output.
        e.g. `my_instr.power=1.00` sets laser to 1.00 W.
        Remember to open shutter for laser on.
        It takes 3-5 minutes for laser to turn on.
        e.g. Laser on `OPMODE=ON`, Laser off `OPMODE=OFF`.
        """
    )

    def __init__(self, adapter, **kwargs):
        super(Sprout, self).__init__(
            adapter, "SproutG12W 532 nm laser", **kwargs)
        self.timout = 3000
```

I initiate the class and set baud rates, read and write terminations. The terminations shouldn't matter but '\r' was recommended by the company since "All replies to commands are terminated with '\r' only."

```
laser = Sprout('COM1')
laser.adapter.connection.baud_rate = 19200
laser.adapter.connection.read_termination = '\r'
laser.adapter.connection.write_termination = '\r'
```
I switch between two operating status `laser.ask('OPMODE?') and `laser.ask('POWER?') which should answer 'OPMODE=OFF' or 'POWER=0.00' respectively. I have written these two variables as an `Instrument.control` parameter (see instrument file ESP300.py below). However, I only get the correct response after sending the ask command multiple times. Note that even the basic write command `laser.write('POWER=0.00')` doesn't work, but `laser.power=0.00' works.

```
laser.power=4
laser.power
Out[18]: 'OPMODE=ON'
laser.power
Out[19]: 'OPMODE=ON'
laser.power
Out[20]: 'OPMODE=ON'
laser.power
Out[21]: 0.0
laser.power
Out[22]: 0.0
laser.power
Out[23]: 0.0
laser.power
Out[24]: 'POWER=4.00'
```

How can I best debug this problem? Please advise.

Cheers, Hud

bilderbuchi, 2020-06-06:
@hududed do you still have this problem? With the code you give above (which looks reasonable/correct to me), it is puzzling why you get the 'OPMODE=ON' reply when asking for  `laser.power`! I suspect that this could be a specific quirk of the instrument in question (check the manual). Same for the delay until you get the correct response, this could be the intended instrument behaviour. You could maybe work around this by implementing [check_errors()](https://pymeasure.readthedocs.io/en/latest/api/instruments/instruments.html?highlight=check_errors#pymeasure.instruments.Instrument.check_errors) and using that in your laser.power definition.

hududed, 2020-06-08:
Hi @bilderbuchi 
I haven't had an issue so far with this although I dont quite remember what I did to change this.
So I guess I'll close this!

Cheers

closed 2020-06-08 by hududed