pymeasure/pymeasure #246: The best way to implement COM devices

paulgoulain, 2020-02-04:
Hello,

I'm interested in using pymeasure for a experiment, but one of my device I want to use is controlled using a the Component Object Model. My first move was to create a new adapter that inherit the adapter class and uses win32com.client library. But when I want to communicate with my instrument, I have to use a syntax which is like that :

COMobject.name_of_the_method()

and it returns the result of the method.

What would be the right way to implement such an adapter ?

bests,

Paul

bilderbuchi, 2020-02-21:
Interesting question! I also have a device that communicates via a compiled DLL, would be curious if that can be made to talk via pymeasure...

tinix84, 2020-03-10:
I had done something similar already for a lecroy oscilloscope https://github.com/tinix84/pymeasure/tree/master/pymeasure/instruments/lecroy
I used this adapter
https://github.com/tinix84/pymeasure/blob/master/pymeasure/adapters/activedso.py

moritzj29, 2021-03-22:
just if anyone is still wondering how DLLs can be used: I did this in #417 by creating a separate adapter. I think this makes sense, if the DLL provides access to more than one instrument at the time.