pymeasure/pymeasure #248: Unit support

bilderbuchi, 2020-02-25:
I think it could be quite useful to add unit support, using pint, similar to how [InstrumentKit](https://instrumentkit.readthedocs.io/en/latest/devguide/util_fns.html#unitful-property) does it (currently transitioning from quantities to pint): Accept and return unitful values (pint `Qantity`), but also deal with plain floats optionally (then assuming a unit).

Any opinions on this?

maederan201, 2021-12-22:
#499 (Keysight N7776C) would be an instrument which would benefit from this implementation and could be used for testing as well.

bmoneke, 2022-06-17:
I'm working on an _adapter_ to make instrumentKit instruments appear as if they were from pymeasure.
The largest difficulty, I see, is, that the instrumentKit instruments always return a pint `Quantity` (number and unit) and not numbers alone. Right now, the pymeasure GUI wouldn't know how to handle them.

What's about making pymeasure able to handle pint `Quantity`?
I would add it as an option. That way instruments may use pint or not.

I did not use pymeasure's GUI and cannot estimate the amount of work.

I think it is good to be unit aware, but would not enforce it (but recommend it) for new instruments.

bilderbuchi, 2022-06-20:
Great to hear! I'm assuming you mean "adapter" in the general sense, not in `Adapter` subclass? (I couldn't make sense of the latter interpretation)

> always return a pint Quantity (number and unit) and not numbers alone. Right now, the pymeasure GUI wouldn't know how to handle them.

Two points: 
* Pint has [wrappers](https://pint.readthedocs.io/en/stable/wrapping.html) to bridge between unit-aware and unitless code sections.
* Pint Quantities have [`m_as()`](https://pint.readthedocs.io/en/stable/developers_reference.html?highlight=m_as#pint.Quantity.m_as) to convert a unitful quantity to a plain float.

Maybe you can use either of those to adapt between pymeasure and IK?

> What's about making pymeasure able to handle pint Quantity?
I would add it as an option. That way instruments may use pint or not.

Indeed, that's what this issue is about! We should first find a good, working design before implementing it. 
Some thoughts about potential approaches that I had in the back of my head were to
* Add a `unit` attribute to the property factories to indicate which unit the property is using. Now that we have dynamic properties, we can even change that at runtime! Very useful e.g. when you can change the unit in the instrument.
* Have a global switch (`pymeasure.USE_UNITFUL_QUANTITIES`?) to disable the behaviour by default (at least in a transition period), or find another good way to on-ramp users while eliminating/limiting breakage of existing code.
* If `USE_UNITFUL_QUANTITIES` is True _and_ a property has a `unit` defined, interaction (get/set) happens with pint Quantities. 
* I think it's possible for properties to always _accept_ float, and then assume they are in the property's `unit`.

> I think it is good to be unit aware, but would not enforce it (but recommend it) for new instruments.

Agreed!

bilderbuchi, 2022-06-20:
Note that the IK compatibility layer is actually issue https://github.com/pymeasure/pymeasure/issues/94, we should keep those topics separated. We _could_ make sure we have unit support _before_ introducing the compatibility layer, of course, if that makes it easier/more reasonable. I could e.g. see requiring that pymeasure unit support is activated if you want to use IK drivers, this could reduce the size of our implementation if we don't need the plain/unitful translation.

bmoneke, 2022-06-20:
I mentioned the IK conversion / translation ("adapter") here as a motivation only.
Therefore, I talked about the requirements of the pymeasure measurement part (not the instruments themselves) to adopt such a conversion.

If the measurement part of pymeasure did handle pint Quantities, the conversion is quite simple (just init, read, write and the context stuff have to be translated), because the properties do not have to be converted. Stripping all quantities is a lot of work and probably error prone.

1. Therefore, I'd like to see pymeasure measurements handle Quantities as a first step.
2. The next step (or concurrently) are pint compatible instruments.

I guess, that both parts (the instruments themselves, and everything related to a measurement without the instruments proper) are independent from point of view of the design (as long as the measurement part supports the types of the instruments' units, that is pint and plain numbers in this case). Therefore, we can do it in two steps, as long as the unit library (pint?) has been chosen.
Making the measurement part unit aware, does not change any code for the users, because we should keep the possibility to understand plain numbers as it is now. Only the transition of the instruments themselves might break code.

> If `USE_UNITFUL_QUANTITIES` is True and a property has a unit defined, interaction (get/set) happens with pint Quantities.

Having an option to deactivate units, might be good, because someone might not want to use units (simple program). Maybe a missing pint library switches to that mode? It's only necessary to change the getter, as the setter is always able to handle numbers (see answer below)

> I think it's possible for properties to always accept float, and then assume they are in the property's unit.

Yes, that should be done. I already improved IK's `assume_units` method, which takes the input (number or string or quantity) and returns a quantity, to be used by the driver.

bilderbuchi, 2022-06-20:
> as long as the unit library (pint?) has been chosen.

Yes, pint, it's the one that makes most sense.

Re measurement/GUI stuff, @CasperSchippers is the main contact person/most familiar with that. He's currently in the final stretch of his PhD, though, so might be busy.

bilderbuchi, 2022-06-28:
Without being too familiar with the csv part of pymeasure, and not having considered backwards compatibility etc:

> How to indicate the desired unit for a measurement quantity (COLUMNS)? 

In the header is best (`Voltage (V)`), or whatever is most frictionless with pyqtgraph (seems like it is already). One can argue if units are specified with parentheses `()` or brackets `[]`, and study the (SI, BIPM) standards; I think parentheses are fine.
Accept unit strings that pint can parse (concerning positive/negative powers like ` J.kg-1.K-1` vs. `J/(kg K)`)
Unitless quantities could either be omitted, `(1)` or `(-)`, I've seen them all.

> What to do, if we get a unitful quantity, but the header does not specify a unit? 

Use the units of the first received quantity, I would say? Before discarding anything, you can always convert to_base_units() and then take the magnitude? But maybe I don't fully understand the point you're trying to make?

bmoneke, 2022-06-28:
I'd let pint make all the parsing to be reliable and deterministic.
According to si, as far as I remember, parenthesis are fine, brackets are not. Pyqtgraph does use parenthesis, but does no parsing. Pymeasure does the parsing (which I copied to the csv parser).

In my implementation, I used no unit for indicating dimension less.

My point is, that if I have 50 dimension less numbers. Afterwards I have one unitful quantity, which makes the whole column unitful, such that those 50 values seem to be in that units, as well.

I see your point of always using the base units, which ensures, that quantities of the same dimensionality (e.g. length) are comparable, but not voltage and length. 
However, our point of introducing quantities is, to make sure, we store the full information.

Two more questions:

- should a mere number (not a dimension less quantity) be accepted as a valid input? That would allow to use a unitful header for an instrument without quantities (ensuring backwards compatibility)
- Should returned strings be converted to a pint Quantity, if possible? Maybe a 'str' unit as indicator for no conversion?

bilderbuchi, 2022-08-03:
> My point is, that if I have 50 dimension less numbers. Afterwards I have one unitful quantity, which makes the whole column unitful, such that those 50 values seem to be in that units, as well.

In which scenario does this appear though? Are there sensible procedures that first produce e.g. 5.3, 5.4, 5.45, then suddenly switch to unitful 5.6 V? 
The only thing I could think of are "invalid" readings (timeout, unavailable, improper setting/mode), but we should account for these in a different way (None/null?) anyway.
Maybe it makes sense to come up with a concrete case first, then maybe the proper course of action (and test case) become clearer?
From my current understanding, instruments/procedures that switch between unitful and unitless for one data column should be an error, but I suspect I don't have the full picture.

> should a mere number (not a dimensionless quantity) be accepted as a valid input even for a unitful quantity? That would allow to use a unitful header for an instrument without quantities (ensuring backwards compatibility). Here again, we have the question on how strict we want to impose the usage of quantities. IK is very strict (you MUST use quantities), I'd say it's in the responsibility of every user to use quantities or not. We just offer the possibility?

The pint wrapping functionality has "strict mode" (https://pint.readthedocs.io/en/stable/wrapping.html#strict-mode) which makes this configurable. We could do this equivalently (i.e., if there are specified units, but data comes in unitless, assume it's in the specified unit). 
I think in terms of a gradual transition to unitful usage, this setting should be non-strict by default. I think we should _enable_ unitful usage, but not force it. 
In the future we could always deprecate unitless usage (with a warning) and then switch the default, if we deem this sensible.

> Should returned strings be converted to a pint Quantity, if possible? Maybe a str unit as indicator for no conversion?

I'm not sure about this. We typically leave it to the instrument drivers to strip postfixes and/or scale in the instrument properties, and then do the casting to numerical values [here](https://github.com/pymeasure/pymeasure/blob/8cfdc9ae3afd9f949b358b10549574ad75de0847/pymeasure/adapters/adapter.py#L75). Converting strings could allow us to straight-up accept, e.g., `0.5 V` as an incoming value (and be automatically correct if the units are changed). 
However, I don't know how many instruments produce pint-compatible strings (which is the situation where doing converstion would give a benefit). Probably, a couple of get_process arguments could be simplified. Maybe we could even simply account for the changed logic in Adapter.values and other places. 
Btw, you could already try out this behaviour right now (I think), if you define a property with `cast=ureg.Quantity` :D

bmoneke, 2022-08-04:
Let's make a table for different cases. I mark open questions with a question mark
- header declares units ('1' indicates dimensionless. Pint recognizes '1' as dimensionless, but not '-')
  - compatible unit received: store it
  - incompatible unit received: drop it (nan)
  - number received: store it (assuming that it is in the right unit). E.g. a unit-unaware instrument, but I want to declare the unit in the header
  - string received: what to do?
    - Reading it with pint (conversion to strings is one recommended way to serialize pint quantities). The header declares a unit, therefore we expect one, even in strings. (my favorite)
    - dropping it, as it is not a number or compatible unit, although it might contain one... 
- header does not declare anything
  - number received: store it (obviously)
  - dimensionless quantity received: store it
  - dimensionful quantity received: what to do?
    - if we store it without further action, we might mix units (one time the device sends 1m stored as 1, another time 10cm stored as ten): drop it therefore?
    - declare that type of unit as default for that column and store it. Now we have (in the example above) always meters and store 10cm as 0.1. I think this is the best use case.
  - string received: store it as is (like right now)?

Did I miss any case? We have declared unit or not and we have number, quantity and string. So I guess we're complete in this picture.

bilderbuchi, 2022-08-04:
> Let's make a table for different cases. I mark open questions with a question mark
> 
>     * header declares units ('1' indicates dimensionless. Pint recognizes '1' as dimensionless, but not '-')
>       
>       * compatible unit received: store it
you mean "quantity", not "unit" here and below, right? 

And with "it" you mean the magnitude in the header's unit (e.g. using `Quantity.m_as(header_unit)`) (as it only makes sense to store floats in e.g. a csv column and the header defines the unit)

>       * incompatible unit received: drop it (nan)
I'd rather warn or error, not silently drop weird readings (unless explicitly told to e.g. with a setting/argument). Otherwise that could lead to weird hunting for missing/wrongly staggered data. I suspect the m_as above will fail if it's "incompatible" already.

>       * number received: store it (assuming that it is in the right unit). E.g. a unit-unaware instrument, but I want to declare the unit in the header
Yes
>       * string received: what to do?
>         
>         * Reading it with pint (conversion to strings is one recommended way to serialize pint quantities). The header declares a unit, therefore we expect one, even in strings. (my favorite)
Yeah, try to parse the string with pint (pass on the string as-is if that fails), and use `m_as(header_unit)` to get the correct magnitude.

What to do with an unparseable string, I don't know. But as I said, I don't know a situation where an instrument property would return some unitful quantity, then suddenly switch to plain floats and/or some unparseable string (with the exception of instrument errors)

> * dropping it, as it is not a number or compatible unit, although it might contain one...
>
>     * header does not declare anything
>
>       * number received: store it (obviously)
>       * [...]
yes, use the first received unit, imho. 
Alternatively, we could use `Quantity.to_reduced_units()`, but I think this is less useful as it could lead to weird units/scales, and we can presume that if the instruments sends a unitful quantity then the unit is a sensible one.

>       * string received: store it as is (like right now)?
yes. maybe extend this from "string" to "non-numeric"
> 
> 
> Did I miss any case? We have declared unit or not and we have number, quantity and string. So I guess we're complete in this picture.

Do we want an explicit way to declare column headers for non-numeric variables (strings, booleans), e.g. an instrument mode, output_enabled or somesuch? Or do we already account for this well in the case header-does-not-declare-unit + non-numeric value?

bmoneke, 2022-08-04:
> you mean "quantity", not "unit" here and below, right?
> 
> And with "it" you mean the magnitude in the header's unit (e.g. using `Quantity.m_as(header_unit)`) (as it only makes sense to store floats in e.g. a csv column and the header defines the unit)

yes and yes.

> I'd rather warn or error, not silently drop weird readings (unless explicitly told to e.g. with a setting/argument). Otherwise that could lead to weird hunting for missing/wrongly staggered data. I suspect the m_as above will fail if it's "incompatible" already.

you're right. It'll do a log entry.

> you mean "quantity", not "unit" here and below, right?
>
> And with "it" you mean the magnitude in the header's unit (e.g. using `Quantity.m_as(header_unit)`) (as it only makes sense to store floats in [...]

I don't like the reduced units thing which might mix different base units.

> Do we want an explicit way to declare column headers for non-numeric variables (strings, booleans), e.g. an instrument mode, output_enabled or somesuch? Or do we already account for this well in the case header-does-not-declare-unit + non-numeric value?

We could mark it with (-) or just use the "not defined" version.

I edit the above table to reflect the decided on parts.

bmoneke, 2022-08-04:
> yes, use the first received unit, imho.
> Alternatively, we could use `Quantity.to_reduced_units()`, but I think this is less useful as it could lead to weird units/scales, and we can presume that if the instruments sends a unitful quantity then the unit is a sensible one.

This raises one problem though: The file header is written in the beginning. In the end, the file does not contain the information, whether this column is stored in cm, dm, m or km.
We could write the units in the end, in case some column unit got declared dynamically.

bilderbuchi, 2022-08-05:
> This raises one problem though: The file header is written in the beginning. In the end, the file does not contain the information, whether this column is stored in cm, dm, m or km. We could write [...]

You're right! Ways out of that:
* Delay writing the file header until we have received data and the first row is ready to be written. Then we can adjust the header entry if needed.
* Seek back and overwrite the header in the file if we hit the code path that discovers "new" unit information. This only works in case of file storage (so you could not stream that elsewhere) and feels a bit hacky, so I'd prefer the former option.

I think the updated table is alright!

bmoneke, 2022-08-05:
It's difficult to wait with the writing, as the Results object just writes the header and then a complex working between different parts begins.
Maybe we could introduce a footer? But right now, a new measurement into the same file (if desired) appends the mere data without a new header, such that a footer might interrupt the data. On the other hand, you could (right now) write two totally different data streams into the same file, which is not very useful.

The case "unitful quantity for not-declared header units" is quite difficult. In the end, to_base_units might be the best idea, because the unit is predictable, if you know the input and does not depend on the order of magnitude of the first value (mm, m, km...).
If you know, what you measured (length or voltage...), you know the base unit and are able to interpret the data.

In any case, it'll create a log entry.

bilderbuchi, 2022-08-05:
I don't think a footer is a good idea, tbh.

> The case "unitful quantity for not-declared header units" is quite difficult.

We could always disallow that case: If we receive a unitful quantity and the header does not identify the unit, fail.

At the same time, I guess we will often encounter it in the transition period from unitless to unitful data. So, the approach with `to_base_units` sounds like the better/more user-friendly bet, as it gives knowable, correct, data. 
If the resulting base unit is inconvenient (for example, the instrument returns electronvolt or Joule), the user can easily define the desired unit in the header to resolve that.

bmoneke, 2022-08-05:
Exactly. I implemented the last case.

bilderbuchi, 2022-08-06:
Most of this is implemented by #642, the follow-on issue concerning unitful instrument properties is #666. :partying_face:

closed 2022-08-06 by bilderbuchi