pymeasure/pymeasure #249: Unit test extension: protocol tests

bilderbuchi, 2020-02-25:
InstrumentKit has this very neat way of testing if the instrument implementation follows and expected protocol and, given a defined query-response set, behaves correctl: https://instrumentkit.readthedocs.io/en/latest/devguide/testing.html#expected-protocols
@cjermain I think this could help reduce PR review effort and improve the test suite if such tests can be used to confirm correct instrument behaviour. What do you think? 

@scasagrande is it ok for you if we implement an equivalent functionality here? Do you have any input/pointers what to watch out for or improve?
I'm also hoping if IK and pymeasure gain more API/usage alignment (see also #248) a future interoperability between the projects will be easier to accomplish.

cjermain, 2020-02-27:
@bilderbuchi, I think this a great idea. I've been wanting to move our `Instruments` to an automated testing suite, but have not had the bandwidth to innovate on it. This would be a great addition on a course toward a stable 1.0 release.

scasagrande, 2020-05-06:
There's lots of better ways one can test the drivers if I were to do it from scratch. I'd still follow the same general style as what I have, but with the following:

- include a captured copy of input/output in an accompanying structured format file, and automatically test against it if the file is present with no/minimal boilerplate. Sometimes when developing the tests I was making guesses on the behaviour of the instrument based on experimentation or just the documentation. Sometimes these don't capture the full picture or are incorrect. This would ideally use a tool to iterate through the driver, dynamically call all the defined members with a real copy of the instrument connected and record all the traffic.
- rely on `hypothesis` for basic fuzzing coverage to help catch edge cases without having to implement each case manually and individually
- make a companion tool to generate test boilerplate / stubs

bilderbuchi, 2020-05-06:
Thanks for the insight, those points sound good!

That query-response record would be useful. I think iterating through all available functions is not too complicated, but the devil will be in the detail, for example 
* concerning functions that are not applicable in the present situation, for example because it needs some specific setup, and error handling in that situation (we don't want to record the error response as "known-good" :D)
* how to deal with noisy floating point responses
* how to deal with updating that file meaningfully when others update/regenerate the file because they extend/fix/modify the driver (especially connected to the previous point).

How do you see Hypothesis coming in here? Are you thinking mainly about checking the input/output validation (that's where I would see some benefit of Hypothesis throwing weird values at IK)? Or something running with the instrument connected?

For test generation: I know in PyCharm there is a function to generate tests, I just haven't used it, yet. Possibly if we have the automated iteration tool from above, that one's output could actually already be the tests, no? 

As an extension to that, do you think it feasible to also run the tests against an actual connected instrument if available (instead of the reference file)?

scasagrande, 2020-05-06:
- add a decorator to signify the requirements, which the scanner can then automatically skip over any its can't satisfy
- if the driver spits out unitful quantities you can just include an acceptable tolerance range for executing tests against real hardware
- it should be deterministic so diffs would be straight forward
- hypothesis check data validation, IK eg: the property factories, and should then be extended to any driver that bypasses them
- if you have access to the actual hardware, yes
- as long as the framework was designed to let you sub in fake instrument connections with real ones (like IK) then yes

bilderbuchi, 2022-08-02:
Finally, this is done! :tada: (#634)

PR #634 (2022-08-02): Add feature: Protocol tests without hardware present
This will be useful to test pymeasure instrument implementations without having the hardware present (see #627).

@bmoneke I have now gone through my notes for now and opened this PR so we can iterate further and knock out the remaining `TODO` items together. If you're up to it we can to PRs against this branch again, I think this worked well for collaborative design/implementation.

* [x] ~~What I still have noted down to do is "Debug/investigate and comment on what ProtocolAdapter.read() is doing with those indices" because it was not clea [... cut]
closed 2022-08-02 by bilderbuchi