pymeasure/pymeasure #250: Dynamic validator limits

bilderbuchi, 2020-02-25:
I have a power supply with voltage/current limits that are changeable depending on a mode setting. I want to use a strict_range validator to check against those limits. What would be the best way to do this?
I had previously tried to do this, but did not find a way to get around the fact that the properties are attributes of the class, and I did not see how to make the values that are passed upon construction dynamically changeable. 

Did I miss something? Do we need a `dynamic_validator` that takes a (class) function or somesuch to track the state of the limits?

bilderbuchi, 2020-03-03:
See also https://github.com/ralph-group/pymeasure/pull/243/files/f903411e8e99a02c193ea4bf41d28da8eb890b97#diff-638e256c0011ed60431a146ca1a44939R333-R334 for another interesting use of validators

msmttchr, 2020-11-16:
Hi, I have question that may be related to this issue and I would like your comments, if possible.
Let's say I am implementing a family of devices which differs for their supported frequency range and I have a command that set/read the frequency parameter. The most obvious way to implement it is to have a generic class implementing an Instrument.control for frequency property with dummy frequency parameters for validator range.
I would then inherit from generic class and have a subclass for each instrument in the family which only specifies the frequency limits.
Being the Instrument.control a property and being the property a class attribute, I found no easy way to change the validator limit, except redefining the property all together for each subclass which clearly violates the DRY principle and it is very tedious. Any easy way to achieve what I need ?

bilderbuchi, 2020-11-17:
I have a suspicion, which I have been meaning to try out for quite some time, that this is in fact quite easy if you use a mutable class attribute, e.g. a tuple with lower, upper limits. Then you can define the class attribute as you propose in the base class, and all subclasses just need to redefine the _class attribute_ in their implementations.
Then I think it even is possible to dynamically change that with a method, because the class attribute gets then overruled by an _instance_ attribute, and the property _should_ still find that. Still have to check/test that, but haven't found the time yet.

See [here](https://github.com/pymeasure/pymeasure/pull/243/files/f903411e8e99a02c193ea4bf41d28da8eb890b97#diff-ed0a3f46fc16d6535c06bbba333a60a47c8a1b1699297f8808733b5dcbef3f6fR101-R117) for the `CLIST_VALUES`.

bilderbuchi, 2020-11-17:
I agree that redefining the property should be avoided for the reasons you give.

msmttchr, 2020-11-17:
I had a look at the example of CLIST_VALUES, but I think the property does not get changed.
I did also a simple code as follow where A=Instrument class, property_test=Instrument.control, B is equivalent to a generic instrument and C and D are actual implementation of the generic instrument B.
```
class A(object):
    @staticmethod
    def property_test(values):
        """ Test property """
        print ("Defining property with", values)
        def fget(self):
            print ("fget: Params range is", values, self)
            return 0

        def fset(self, value):
            print ("fset: Params range is", values, self)
        return property(fget, fset)

class B(A):
    param_range=(1,2)
    # Class attribute
    my_property = A.property_test(param_range)
    def __init__(self):
        self.param_range=(8,9)

class C(B):
    param_range=(3,4)

class D(B):
    param_range=(5,6)

C().my_property
D().my_property

```
The output is as follow:
```
Defining property with (1, 2)
fget: Params range is (1, 2) <__main__.C object at 0x00837D70>
fget: Params range is (1, 2) <__main__.D object at 0x00837D70>
```
This is interesting since:
1) The property my_property is defined during class B definition and not during C or D class instantiation.
2) The param_range is taken into account only during class B definition and any redefinition of param_range in C and D is not taken into account. Actually when defining my_property python is not aware (he does not care) of C or D definition happening afterwards.

Probably, it is needed some dedicated mechanism to handle limits change when inheritance is used.

bilderbuchi, 2020-11-17:
Yeah, properties are tricky because they are class-bound. It's even worse when you want to dynamically work with them (i.e. generate a set of properties depending on a configfile or something - you quickly end up with wrapper classes etc. But I digress...

Here, you have to use _mutable_ fields, and don't redefine/overwrite them, but update instead, then you can make this work. To wit:
```
class Ins(object):
      @staticmethod
      def property_test(values):
          """ Test property """
          print ("Defining property with", values)
          def fget(self):
              print ("fget: Params range is", values, self)
              return 0

          def fset(self, value):
              print ("fset: Params range is", values, self)

          return property(fget, fset)

class B(Ins):
    param_range=[1, 2]  # this has to be _mutable_!
    # Class attribute
    my_property = Ins.property_test(param_range)
    def __init__(self):
        # Don't replace the field, _update_ it
        self.param_range[0:2] = [8, 9]

class C(B):
    def __init__(self):
        super().__init__()
        # Don't replace the field, _update_ it
        self.param_range[0:2] = [3, 4]

class D(B):
    def __init__(self):
        super().__init__()
        # Don't replace the field, _update_ it
        self.param_range[0:2] = [5, 6]

print(B().my_property)
print(C().my_property)
print(D().my_property)
```
```
Defining property with [1, 2]
fget: Params range is [8, 9] <__main__.B object at 0x7ff5e4dc0e10>
0
fget: Params range is [3, 4] <__main__.C object at 0x7ff5e4dc0e10>
0
fget: Params range is [5, 6] <__main__.D object at 0x7ff5e4dc0e10>
0
```
:partying_face:

msmttchr, 2020-11-17:
Thank you very much, really appreciated. Looks promising, tomorrow I will give it a try

bilderbuchi, 2020-11-18:
Great! Probably using a dict like `limits = {'min': 2, 'max': 5}` will be easier to use and give more understandable code than a tuple.

msmttchr, 2020-11-18:
I think I understand the solution you proposed, but there is a drawback as follow: basically, we are using a reference to a single mutable object to define the limits. When we change the mutable object, since all the references (class wise and instance wise) are pointing to that same mutable, this is changed for all of them and not just for a particular class or instance.
I have modified the code as follow and show that after the instantiation of C also B param_limit is changing and after the instantiation of D, both B and C param_limit are changing
```
class Ins(object):
      @staticmethod
      def property_test(values):
          """ Test property """
          print ("Defining property with", values)
          def fget(self):
              print ("fget: Params range is", values, self)
              return 0

          def fset(self, value):
              print ("fset: Params range is", values, self)

          return property(fget, fset)

class B(Ins):
    param_range=[1, 2]  # this has to be _mutable_!
    # Class attribute
    my_property = Ins.property_test(param_range)
    def __init__(self):
        # Don't replace the field, _update_ it
        self.param_range[0:2] = [8, 9]

class C(B):
    def __init__(self):
        super().__init__()
        # Don't replace the field, _update_ it
        self.param_range[0:2] = [3, 4]

class D(B):
    def __init__(self):
        super().__init__()
        # Don't replace the field, _update_ it
        self.param_range[0:2] = [5, 6]
b = B()
print ("B instantiated")
b.my_property
c = C()
print ("C instantiated")
b.my_property
c.my_property
d = D()
print ("D instantiated")
b.my_property
c.my_property
d.my_property
```
Output as follow:
```
Defining property with [1, 2]
B instantiated
fget: Params range is [8, 9] <__main__.B object at 0x00777C90>
C instantiated
fget: Params range is [3, 4] <__main__.B object at 0x00777C90>
fget: Params range is [3, 4] <__main__.C object at 0x00777E10>
D instantiated
fget: Params range is [5, 6] <__main__.B object at 0x00777C90>
fget: Params range is [5, 6] <__main__.C object at 0x00777E10>
fget: Params range is [5, 6] <__main__.D object at 0x00777DB0>
```
So, I think the proposed solution has really nasty side effects. Do you agree ?

bilderbuchi, 2020-11-18:
Ah yes, of course 😞 . It seems further research is needed...

bilderbuchi, 2020-11-18:
@CasperSchippers since the `CLIST_VALUES` approach is something you used, this discussion is probably relevant for you, too. I think that is also affected by the issues in here?

bilderbuchi, 2020-11-18:
Hah!
```
class Ins(object):
    @staticmethod
    def property_test(values):
        """ Test property """
        print("Defining property using ", values)

        def fget(self):
            print("fget: Params range is", getattr(self, values), self)
            return 0

        def fset(self, value):
            print("fset: Params range is", getattr(self, values), self)

        return property(fget, fset)

class B(Ins):
    param_range = [8, 9]
    my_property = Ins.property_test('param_range')

class C(B):
    param_range = [3, 4]

class D(B):
    param_range = [5, 6]
    def change_limits(self):
        self.param_range = [11, 22]

b = B()
print("B instantiated")
b.my_property
c = C()
print("C instantiated")
b.my_property
c.my_property
d = D()
print("D instantiated")
b.my_property
c.my_property
d.my_property
d.change_limits()
b.my_property
c.my_property
d.my_property
```
gives
```
Defining property using  param_range
B instantiated
fget: Params range is [8, 9] <__main__.B object at 0x0000025D401DCBC8>
C instantiated
fget: Params range is [8, 9] <__main__.B object at 0x0000025D401DCBC8>
fget: Params range is [3, 4] <__main__.C object at 0x0000025D401DCB88>
D instantiated
fget: Params range is [8, 9] <__main__.B object at 0x0000025D401DCBC8>
fget: Params range is [3, 4] <__main__.C object at 0x0000025D401DCB88>
fget: Params range is [5, 6] <__main__.D object at 0x0000025D401DCC48>
fget: Params range is [8, 9] <__main__.B object at 0x0000025D401DCBC8>
fget: Params range is [3, 4] <__main__.C object at 0x0000025D401DCB88>
fget: Params range is [11, 22] <__main__.D object at 0x0000025D401DCC48>
```

CasperSchippers, 2020-11-20:
It is indeed an interesting problem. I think for this you do indeed need to redefine the property factories a bit. I played around with it a bit a came up with the implementation in #371.

> (There was a PR/comment around that recently but I can't find it right now)

Do you mean this comment: https://github.com/pymeasure/pymeasure/pull/308#discussion_r503728136?

bilderbuchi, 2021-02-26:
When this is done, look at the agilent 33250A AWG from #380 and Keithley 2260B series from #395 for candidates for using this feature.

LongnoseRob, 2021-08-12:
I came across this open issue and remembered something that I read in docuumentation form R&S:
It is possible to query the min/max limits of a parameter with the following syntax:
**PARAMETER?** --> actual value
**PARAMETER? MIN** --> minum value
**PARAMETER? MAX** --> maximum value

Quick expample with `pyvisa-shell`
```
GPIB0::20::INSTR has been opened.
You can talk to the device using "write", "read" or "query".
The default end of message is added to each message.
(open) query SOUR:FREQ:CW?
Response: 4.351000e+08

(open) query SOUR:FREQ:CW? MAX
Response: 1.000000e+09

(open) query SOUR:FREQ:CW? MIN
Response: 5.000000e+06
```

This is actually specified in the SCPI-99 standard, $7.2.1.2  (page 35 of https://www.ivifoundation.org/docs/scpi-99.pdf)

So it would be technically possible to read the limits for certain parameters of an instrument and use these limits available dynamic range-check.
But I  have no good idea at the moment how to do this exept to modify 
``Instrument.control(get_command,get_command_min, get_command_max, set_command, docs,... 
``

bilderbuchi, 2021-08-12:
yeah, this kind of additional logic is probably best implemented as a normal Python property, instead of with the pymeasure property factories. Keep in mind though, that doing this basically triples the amount of communication you have with the instrument, which will probably be at least inconvenient, if not prohibitive. Considering that the limits will probably not change by themselves, only on user action, I think caching the value in the Instrument instance is the more promising approach.

bilderbuchi, 2022-01-23:
Closed by #376. :party:

closed 2022-01-23 by bilderbuchi