pymeasure/pymeasure #251: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed. with keithley 2400

jasonimm, 2020-02-28:
Hi all,

I connected my computer with Keithley 2400 and tried to communicate using this Pymeasure. when I executed my code. it looks connected very well because it communicate in some part. 
`from pymeasure.instruments.keithley import Keithley2400
import numpy as np
#import nidaqmx
from time import sleep
import time
    
def discharging(I_para, V_lim):
#    with nidaqmx.Task() as task:
#        task.ai_channels.add_ai_thrmcpl_chan("Dev2/ai0")
    # Connect and configure the instrument
    keithley = Keithley2400("ASRL1::INSTR")
    
    keithley.apply_current()                # Sets up to source current
    keithley.compliance_voltage = 10        # Sets the compliance voltage to 10 V
    keithley.source_current = I_para        # e.g., -0.024
    keithley.enable_source()
    # Loop through each current point, measure and record the voltage
    
    i = 0
    tic = time.time()  ` 

this is part of the my code. in definition of discharging, setting the compliance voltage and enabling source is well worked. however when I try to apply the current or make source current, the timeout error is popped out. 
What is problem in communication between pc and keithley2400 ???

bilderbuchi, 2021-02-16:
@jasonimm have you tried this with the latest pymeasure release 0.9?

bilderbuchi, 2021-02-16:
Oh, I misread the year; no update for a long time so I'll close this, feel free to reopen if this is still relevant.

closed 2021-02-16 by bilderbuchi