pymeasure/pymeasure #260: iv_keithley.py

danjohare, 2020-03-11:
Did not work for me. I noticed compliance had a typo.  
The graph displays, the file is started but no data populated. 
I do not see any signs of life on the Keithley front panel. It is
a 2400 and I set GPIB to 30 in hardware and code. 

Any new versions that might not have issues ? I would love to use it!

bilderbuchi, 2020-03-13:
Hi! I am not sure how current that example is. Do you have a Keithley2000 available and connected, too?
You're right about that typo, nicely spotted.
I tried to reproduce with a Keithley2400 I have around, but I'm hitting #168 upon pressing Queue before anything meaningful happens.

PYmeas, 2020-08-19:
Hello, I am trying to implement an IV procedure and plot using managedwindow while it works fine with random number i am having an attribute error when i connect toa Keithley 2400. using IV code on standalone without procedure works.
the error displayed is:
8/19/2020 03:38:50 PM : Worker caught an error on <IVProcedure(status=Running,parameters_are_set=True)> (ERROR)
Traceback (most recent call last):
  File "C:\Users\PycharmProjects\automated measurements\venv\lib\site-packages\pymeasure\experiment\workers.py", line 166, in run
    self.procedure.execute()
  File "D:/Data/IV Procedure.py", line 41, in execute
    self.sourcemeter.source_current = current  # Sets the source current to min current mA
AttributeError: 'IVProcedure' object has no attribute 'sourcemeter'

if any suggestion?

``` ruby
import logging
log = logging.getLogger(__name__)
log.addHandler(logging.NullHandler())
import sys
from datetime import datetime as dt, timedelta
from time import sleep
import pandas as pd
import numpy as np
from pymeasure.instruments.keithley import Keithley2400
from pymeasure.log import console_log
from pymeasure.display.Qt import QtGui
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import Procedure, Results, Worker
from pymeasure.experiment import IntegerParameter, FloatParameter, Parameter

class IVProcedure(Procedure):
    # averages = IntegerParameter('Averages', default=50)
    min_current = FloatParameter('Minimum Current', units='A', default=-0.01)
    max_current = FloatParameter('Maximum Current', units='A', default=0.01)
    data_points = IntegerParameter('Data points', default=50)
    mode = IntegerParameter('2 or 4 probes measurement', default=2)

    DATA_COLUMNS = ['Time', 'Current (A)', 'Voltage (V)']

    def startup(self):
        log.info("Connecting and configuring the instrument")
        self.sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
        self.sourcemeter.enable_source()
        self.sourcemeter.reset()
        self.sourcemeter.use_front_terminals()
        self.sourcemeter.wires = self.mode  # type 4 for 4 terminals measurement
        self.sourcemeter.apply_current()
        self.sourcemeter.source_current = self.min_current
        # sourcemeter.measure_resistance()

    def execute(self):
        currents = np.linspace(self.min_current, self.max_current, num=self.data_points)
        for current in currents:
            log.info("Setting the current to %g A" % current)
            self.timecode = dt.now()
            self.sourcemeter.source_current = current  # Sets the source current to min current mA
            sleep(0.1)
            self.sourcemeter.measure_voltage()
            sleep(0.1)
            data = {
                'Time': self.timecode,
                'Current (A)': current,
                'Voltage (V)': self.sourcemeter.voltage,
            }
            self.emit('results', data)
            log.debug("Emitting results: %s" % data)
            sleep(0.1)
            if self.should_stop():
                log.info("User aborted the procedure")
                break

    def shutdown(self):
        self.sourcemeter.shutdown()
        log.info("Finished measuring")

class MainWindow(ManagedWindow):

    def __init__(self):
        super(MainWindow, self).__init__(
            procedure_class=IVProcedure,
            inputs=['min_current', 'max_current', 'data_points', 'mode'],
            displays=['min_current', 'max_current', 'data_points', 'mode'],
            x_axis='Voltage (V)',
            y_axis='Current (A)',
            # sequencer=True,  # Added line
            # sequencer_inputs=['min_current', 'max_current', 'data_points', 'mode'],  # Added line
            # sequence_file="gui_sequencer_example_sequence.txt",  # Added line, optional if load a sequence file
        )
        self.setWindowTitle('IV procedure')
        self.count = 0

    def queue(self, *, procedure=None):
        self.count = self.count + 1
        if procedure is None:
            procedure = self.make_procedure()
        filename = '.\\test\\' + 'data_' + str(dt.now().strftime("%m%d%Y"))+'_' + str(self.count) + '.csv'
        log.info("Constructing the Results with a data file: %s" % filename)
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)
        self.manager.queue(experiment)

if __name__ == "__main__":
    app = QtGui.QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec_())

```

physicsninja, 2020-08-25:
@PYmeas Your issue indicates that the ```IVProcedure``` instance run by the worker isn't getting the attribute ```sourcemeter``` assigned. This is weird because is does look like you set it in your startup method. When you look at the logs, does the "Connecting and configuring the instrument" string get logged?

PYmeas, 2020-08-26:
Thanks for the feedback @physicsninja . No i dont see the "Connecting and configuring the instrument" string logged it s as if the `startup `method is shortcut. here is the exact log on the managedwindow when running the code as it is:
08/26/2020 12:37:19 PM : ManagedWindow connected to logging (INFO)
08/26/2020 12:37:22 PM : Constructing the Results with a data file: .\test\test2\data_08262020_1.csv (INFO)
08/26/2020 12:37:22 PM : Worker thread started (INFO)
08/26/2020 12:37:22 PM : Worker started running an instance of 'IVProcedure' (INFO)
08/26/2020 12:37:22 PM : Setting the current to -0.01 A (INFO)
08/26/2020 12:37:23 PM : Worker caught an error on <IVProcedure(status=Running,parameters_are_set=True)> (ERROR)
Traceback (most recent call last):
  File "C:\PycharmProjects\automated measurements\venv\lib\site-packages\pymeasure\experiment\workers.py", line 166, in run
    self.procedure.execute()
  File "D://coding/IV Procedure.py", line 49, in execute
    self.sourcemeter.start_buffer()
AttributeError: 'IVProcedure' object has no attribute 'sourcemeter'

**but on the IDE comsole i get this traceback slightly different:**
Exception in thread Thread-12:
Traceback (most recent call last):
  File "C:\Users\Anaconda3\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\PycharmProjects\automated measurements\venv\lib\site-packages\pymeasure\experiment\workers.py", line 172, in run
    self.shutdown()
  File "C:\Users\PycharmProjects\automated measurements\venv\lib\site-packages\pymeasure\experiment\workers.py", line 118, in shutdown
    self.procedure.shutdown()
  File "D:/coding/IV Procedure.py", line 93, in shutdown
    self.sourcemeter.shutdown()
AttributeError: 'IVProcedure' object has no attribute 'sourcemeter'

**at last if i modify the `startup `method as below the Keithley initializes but there is a conflict in during the data acquisiton:**
```ruby
    def __init__(self, **kwargs):
        super().__init__(**kwargs)
        self.sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
        self.sourcemeter.enable_source()

    def startup(self):
        log.info("Connecting and configuring the instrument")
        # self.sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
        # self.sourcemeter.enable_source()
        self.sourcemeter.reset()
        self.sourcemeter.use_front_terminals()
        self.sourcemeter.wires = self.mode  # type 4 for 4 terminals measurement
        self.sourcemeter.apply_current()
        self.sourcemeter.source_current = min_current
```
08/26/2020 01:18:54 PM : ManagedWindow connected to logging (INFO)
08/26/2020 01:18:54 PM : Initializing Keithley 2400 SourceMeter. (INFO)
08/26/2020 01:18:59 PM : Initializing Keithley 2400 SourceMeter. (INFO)
08/26/2020 01:18:59 PM : Constructing the Results with a data file: .\test\test2\data_08262020_1.csv (INFO)
08/26/2020 01:18:59 PM : Worker thread started (INFO)
08/26/2020 01:18:59 PM : Worker started running an instance of 'IVProcedure' (INFO)
08/26/2020 01:18:59 PM : Setting the current to -0.01 A (INFO)
08/26/2020 01:18:59 PM : Keithley 2400 SourceMeter is measuring voltage. (INFO)
08/26/2020 01:18:59 PM : Keithley 2400 reported error: -221, Settings conflict (INFO)
08/26/2020 01:19:00 PM : Setting the current to -0.00959184 A (INFO)
08/26/2020 01:19:00 PM : Keithley 2400 SourceMeter is measuring voltage. (INFO)
08/26/2020 01:19:00 PM : Keithley 2400 reported error: -221, Settings conflict (INFO)
08/26/2020 01:19:00 PM : Setting the current to -0.00918367 A (INFO)
08/26/2020 01:19:01 PM : Keithley 2400 SourceMeter is measuring voltage. (INFO)
08/26/2020 01:19:01 PM : Keithley 2400 reported error: -221, Settings conflict (INFO)
08/26/2020 01:19:01 PM : User aborted the procedure (INFO)
08/26/2020 01:19:01 PM : Shutting down Keithley 2400 SourceMeter. (INFO)
08/26/2020 01:19:02 PM : Finished measuring (INFO)
08/26/2020 01:19:02 PM : Monitor caught stop command (INFO)

the file gets some data but inconsistent.

PYmeas, 2020-09-01:
@physicsninja 

I managed to get measurements but the startup method is still not functional by putting the initialization commands in a def__ini__ method it works fine. I dont get it. the commands are the same and my understanding was that the startup method is meant to initilalize the instruments.
``` ruby
  # if use a def__ini __ the instrument initializes and get measurements
    def __init__(self):
        super().__init__()
        log.info("Connecting and configuring the instrument")
        self.sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
        self.sourcemeter.reset()
        self.sourcemeter.use_front_terminals()
        self.sourcemeter.wires = self.mode  # type 4 for 4 terminals measurement
        self.sourcemeter.apply_current()
        # self.sourcemeter.source_current = self.min_current
        # sourcemeter.measure_resistance()
        self.sourcemeter.enable_source()

   # i use the startup method the instrument doesnt initialize and i get an attribute error
    def startup(self):
        log.info("Connecting and configuring the instrument")
        self.sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
        self.sourcemeter.reset()
        self.sourcemeter.use_front_terminals()
        self.sourcemeter.wires = self.mode  # type 4 for 4 terminals measurement
        self.sourcemeter.apply_current()
        # self.sourcemeter.source_current = self.min_current
        # sourcemeter.measure_resistance()
        self.sourcemeter.enable_source()
```

the rest of the code is just the Managewindow class to plot so i dont think it would conflict?

bilderbuchi, 2021-03-03:
I just fixed that typo and a couple of linter warnings in iv_keithley.py, so this should work better now.
 
I don't see why `startup` would not be executed. Does the problem still occur? If so, can you supply a complete **minimal** working example, ideally with a FakeAdapter or FakeInstrument, so that we can troubleshoot?

bilderbuchi, 2022-01-09:
Didn't hear back from the reporter, so I'll close this. Feel free to reopen with a problem description if this this still occurs for you.

closed 2022-01-09 by bilderbuchi