pymeasure/pymeasure #261: Live Image Plotting Bugs and Improvements

jmittelstaedt, 2020-03-15:
This is an issue to track any bugs or other issues with the live image plotting introduced in PR #157.

One of the major findings so far was that after changing the image plotting behavior to only show one image at a time, after changing which image was shown the axes would begin zooming out uncontrollably. This is the same behavior seen in #54. For now, we allow multiple images to be shown on top of one another.

samcondon4, 2021-12-22:
The current version of image plotting continues to use csvs as the main file format that the `Results` and `Worker` classes write to. This is fine for small image files like the one created by default in the `image_gui.py` example, though when increasing the size of the image to a real standard image size like say 1024 x 768, it becomes incredibly slow.  With those image dimensions the `image_gui.py` example takes a full few minutes to write and display the image. If a user would ever like to use a real camera and log image data in an experiment this is completely intractable. 

I believe that a good first step to improving live image plotting performance is to add support for writing to real image file formats via the `Results` and `Worker` classes. My initial thought is that having FITS file support would be quite useful since FITS files support an arbitrary number of data channels and unlimited metadata in the header. It is also currently the most widely used image format in astrophysics/astronomy. 

What are people's thoughts on the usefulness/necessity of adding support for additional file formats in the `Results` and `Worker` classes and specifically for live image plotting, adding FITS support?

bilderbuchi, 2021-12-27:
> What are people's thoughts on the usefulness/necessity of adding support for additional file formats in the Results and Worker classes and specifically for live image plotting, adding FITS support?

Surely useful, see #136 for this. I imagine we'll have to come up with some optional dependency/plugin kind of structure, to enable different output formats while not dragging them into every installation if not needed. I have not spent any deep thought on that, though. Personally, my first candidates would be HDF5 or netCDF, but FITS is of course also a relevant format (although pymeasure is not strong on Camera interaction, for that other libs are much more mature I think). :-)

PkRPintu, 2025-05-08:
Hello, 

I am new to Pymeasure and would like to use it for live image scanning while grabbing data from various lock-ins. I have three questions if anyone can kindly help me. I really appreciate any help you can offer.

1. How can I add a color bar to the image scan display that changes dynamically depending on the Z axis values? (If the colour bar can be interactive, that would also be a great feature)
2. How to change the colormap from viridis to something else, probably a diverging colour map like ‘bwr’?
3. Also, there is a managed docked window for displaying multiple plots. Can we have managed docked image windows so that we can display multiple image plots at once? 

I think these are very handy features for the imagescan experiments.
Cheers

Translated with DeepL.com (free version)