pymeasure/pymeasure #262: Plain Vanilla Ethernet Adapter?

paulobear, 2020-03-17:
I am trying to communicate with a Fluke 1594A SuperThermometer. I have the documentation for the instrument which details the ASCII messages. I have created a driver for the instrument using the PyMeasure Instrument class. The vast majority of the messages fit into the Instrument.control(), Instrument.measurement() and Instrument.setting() paradigms.

I want to communicate with the instrument over TCP ethernet on port 5025 (assuming the instrument is configured correctly). Is there a simple PyMeasure ethernet adapter? The documentation doesn't say anything about the instrument being VXI-11 or VISA compatible. So not clear those adapters will work. Thought that perhaps the SerialAdapter would work, but that looks to be specific to RS232 and other serial comm type of ports.

Is there a PyMeasure best practice for direct ethernet connections?

Thanks, Paul E Ourada

bilderbuchi, 2020-03-17:
Have you tried with a [pyvisa-compatible](https://pyvisa.readthedocs.io/en/latest/introduction/names.html#visa-resource-syntax-and-examples) TCPIP resource name, and either the default adapter or specifically a VISA adapter?

paulobear, 2020-03-18:
Thanks bilderbuchi - I am currently locked out of the building, since we're on COVID-19 hiatus. I will give this a shot when I get back to the office.

paulobear, 2020-03-25:
UPDATE - I have had success with using the VISAAdapter from the point of view of constructing the adapter with a VISA TCP SOCKET resource ID. Of course, when I try to actually write anything, I get a connection refused error. I could, and may, create a socat object to capture the data. Or I may attempt to create an instrument simulator.

bilderbuchi, 2020-03-26:
Great! Feel free to leave this open until you are able to check, or close it already, depending on your confidence :D