pymeasure/pymeasure #264: Instrument Simulator Development?

paulobear, 2020-03-25:
Since I am locked out of the lab where all the equipment is, but am in need of developing device drivers, procedures and experiments, I thought that I might develop instrument simulators to support this work. I looked through the pymeasure code and see the Mock class, and FakeThis and FakeThat. But I don't see any examples of how to use them?

I imagine that a simulator would look pretty much like a driver, but instead attempt to implement the functionality in some fashion. I thought that perhaps the unit test framework might give some clue, but I'm not seeing it. Can someone lend me a clue?

NOTE: I do plan to donate these instrument drivers back, if the company allows. I suspect that they will.

bilderbuchi, 2020-03-26:
You're right, this is not heavily used right now. 
One idea/approach to extend unit test with a "simulated" instrument is #249. The idea is that you know, e.g. from instrument manual examples, what messages will be sent to the device (even including termination chars) and what it will then be returned (the "protocol"). That can then be used to implement unit tests that check if the Instrument driver has implemented the correct behaviour. InstrumentKit's wide array of unit tests largely work with this. It would be useful to have, but I don't have the time, so feel free to pick this up.

paulobear, 2020-03-26:
Thanks, bilderbuchi. Can you describe in short fashion how the pymeasure unit tests work? I see in the docs how to add new instruments and then invoke the unit test VMs, but it's not clear what those unit tests actually check and the mechanism.

bilderbuchi, 2020-03-26:
@cjermain is much more authorative on that. I can say that you can use pytest to run the tests, and that there are [nearly no tests](https://github.com/ralph-group/pymeasure/tree/master/tests/instruments) testing instrument implementations, what is there is mostly for surrounding functionality. This is a hole that #249 could help plug.

Satertek, 2020-03-26:
If you're utilizing VISA devices, be sure to look into pyvisa-sim.  I've updated our in-house tools to utilize this and so far it has gone well.  (We'll see when I can hook back up to real equipment)

This may be something can can be integrated into pymeasure.

bmoneke, 2022-08-02:
We're working on a test environment, which allows to code without the devices at hand. #634 #627

bilderbuchi, 2022-08-02:
> We're working on a test environment, which allows to code without the devices at hand. #634 #627

This has been merged now, so everybody can test their implementations without hardware being present (in time for the next Covid lockdowns, I guess :D) I'll close this

closed 2022-08-02 by bilderbuchi