pymeasure/pymeasure #270: Read all instrument responses in the buffer

NitramC, 2020-04-11:
The motion controller I'm working with (PI E-873.3QTU) returns multi-line responses "<axisID>=<value> LF", where LF stands for linefeed, and is used as the termination character. Seems like bad design to use the same character as both a separator and a termination character... but that's how it is. Here is an example:
axis1=10.0\n
axis2=20.0\n
axis3=30.0\n

How do I read back all the responses without knowing how many there are, and without calling the adapter directly? The only hack I could think of is repeatedly calling Instrument.read() in a try/except block and catching a timeout error. Even then, I wouldn't know the exact exception returned a priori.

Is it better to write an adapter class that replaces the intermediate termination characters with a different character (maybe semicolon)? e.g.
axis1=10.0;axis2=20.0;axis3=30.0\n

Thoughts?

NitramC, 2020-04-11:
I am going with the idea provided by @MatthieuDartiailh here: https://github.com/pyvisa/pyvisa/issues/504

So closed, unless someone has a better approach.

closed 2020-04-11 by NitramC

bilderbuchi, 2020-04-12:
Yeah that approach is quite elegant, I'm not sure we could provide something better. Glad that the instrument manufacturer at least left _some_ way to disambiguate between separators and message termination! 😅