pymeasure/pymeasure #271: Running Multiple Simultaneous Procedures

paulobear, 2020-04-28:
I have been working on an instrument calibration project which requires simultaneous communication with multiple measurement instruments. One is a temperature controlled bath, which is stepped from one temperature to another (say 40 C to -20 C in -10 C steps). Another is a precision thermometer which is used as the reference standard. A third is a suite of up to eight custom scientific instruments which report temperature. It's these last instruments which are being calibrated using the reference thermometer. At each step, the temperature is monitored to make sure it stabilizes, and then data is saved off to be used in the calibration (although I may change this to save all data, and merely mark data as being stable).

I have written procedures for the bath control and the reference thermometer control, and they work as desired, collecting the data in real time. At present, they both run in the same process on different worker threads.

I would like to be able to display all 10 temperature plots in the graph object of one ManagedWindow object. However, ManagedWindow only takes one Procedure object at a time. It does allow for staging multiple invocations of the same Procedure using different values of the Parameters. However, my use case requires running multiple procedures at the same time.

I have a vague idea of having a list of procedure classes, which can then all be instantiated and run simultaneously. Does this seem like a good approach?

bilderbuchi, 2020-05-14:
Phew, out of my depth on that question, but running multiple procedures could be useful. In my case, I would like to start a setup/experiment, observe/plot for a while, while also being able to interact with the instruments (e.g. via buttons/GUI fields), then kicking off a defined procedure. That could be done maybe with an "endless" procedure running in parallel. Or maybe better with a `procedure.startup()` that hands control back after it runs, to wait before executing.
Sorry, a little rambling, not very actionable input for you.

paulobear, 2020-05-14:
> Phew, out of my depth on that question, but running multiple procedures could be useful. In my case, I would like to start a setup/experiment, observe/plot for a while, while also being able to [...]

Hi bilderbuchi - I actually have created two classes based on Plotter and PlotterWindow called MultiPlotter and MultiPlotterWindow as an initial foray into this area. Please see the attached files. The file which actually uses the MultiPlotter* classes won't work for you, but will give you some idea about how to use the new MultiPlotter classes.

I've been thinking about this a bit in terms of ManagedWindow. It seems to me that this should be fairly doable if a Queue element were introduced that was a list of procedures, each of which have a zero time delay between them, as opposed to a consecutive execution. So maybe a delay attribute of the Queue element procedures, which might be 'delay=parallel' or 'delay=consecutive' or 'delay=60.0'.

What do you think?

Here are my new classes.

multi_plotter.py:
```
"""
 ********************************************************************
 ** isfcallab - UCAR/EOL/ISF Calibration Lab Software
 **
 ** 2020, Copyright University Corporation for Atmospheric Research
 **
 ** This program is free software; you can redistribute it and/or modify
 ** it under the terms of the GNU General Public License as published by
 ** the Free Software Foundation; either version 2 of the License, or
 ** (at your option) any later version.
 **
 ** This program is distributed in the hope that it will be useful,
 ** but WITHOUT ANY WARRANTY; without even the implied warranty of
 ** MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 ** GNU General Public License for more details.
 **
 ** The LICENSE.txt file accompanying this software contains
 ** a copy of the GNU General Public License. If it is not found,
 ** write to the Free Software Foundation, Inc.,
 ** 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA.
 **
 ********************************************************************

 multi_poltter.py contains classes and methods to take multiple Procedure
 subclasses and plot their individual outputs on a single pyqtgraph plot.
 It uses pymeasure.Plotter as the base class to set the gui up initially.

 :class: MultiPlotter
"""

import sys
import pyqtgraph as pg
import time

from pymeasure.display import Plotter
from pymeasure.display.windows import PlotterWindow
from pymeasure.experiment import Results
from pymeasure.display.Qt import QtGui
from pymeasure.display.curves import ResultsCurve

class MultiPlotter(Plotter):
    """

    """
    def __init__(self, results: list, refresh_time=0.1):
        super(MultiPlotter, self).__init__(results, refresh_time)
        self.results = results[0]
        self.results_list = results[1:]
        self.window = None

    def run(self):
        app = QtGui.QApplication(sys.argv)
        self.window = MultiPlotterWindow(self, self.refresh_time)
        self.setup_plot(self.window.plot)
        self.add_curves()
        app.aboutToQuit.connect(self.window.quit)
        self.window.show()
        time.sleep(0.1)
[... 107 lines cut ...]
    results = []
    time_prefix = datetime.now().strftime("%Y-%m-%d-%H:%M:%S")
    bath_filename = time_prefix + '-bath_temp_results.csv'
    bath_proc = MonitorBathTemp(simulate=True)
    results.append(Results(bath_proc, bath_filename))

    ref_therm_filename = time_prefix + '-ref_therm_temp_results.csv'
    ref_therm_proc = MonitorRefTherm(simulate=True)
    results.append(Results(ref_therm_proc, ref_therm_filename))

    bath_proc.simulator.subscribe(ref_therm_proc.simulator)

    bath_worker = Worker(results[0], scribe.queue, log_level=logging.DEBUG)
    bath_worker.start()
    ref_therm_worker = Worker(results[1], scribe.queue, log_level=logging.DEBUG)
    ref_therm_worker.start()
    # sleep(3600.) # uncomment this to use breakpoints in the procedure code...
    plotter = MultiPlotter(results, refresh_time=1.0)
    plotter.start()
    plotter.wait_for_close()
    bath_worker.join()
    ref_therm_worker.join()
    scribe.stop()

```

PYmeas, 2020-08-03:
I manage to run 2 procedures in parallel on 2 different windows but I d also like to observe the execution of various procedures on the same window. I added an integer (devices) in the parameter of each procedure and put an if statement in the queue to choose the corresponding procedure but i get the AttributeError: 'NoneType' object has no attribute 'devices' error. I did shift up procedure call before filename in the queue so i should get the parameters updated (as proposed in #302) but the device parameter input seems not to operate. whenever i get a display it s only procedure one being executed independently of devices parameter value. #280 seems to be working on a similar issue , with his approach i get the first procedure running, the second appear in the queue but is not executed.

```ruby
class MainWindow(ManagedWindow):

    def __init__(self):

        super(MainWindow, self).__init__(
            procedure_class=RandomProcedure_1,
            inputs=['devices', 'mode', 'iterations', 'delay'],
            displays=['devices', 'mode','iterations', 'delay'],
            x_axis='Iteration',
            y_axis='Random Number',
            sequencer=True,  # Added line
            sequencer_inputs=['iterations', 'delay', 'seed', 'mode'],  # Added line
            inputs_in_scrollarea=True
            # sequence_file="gui_sequencer_example_sequence.txt",  # Added line, optional
        )
        self.setWindowTitle('TEST')
        self.count = 0

    def queue(self, *, procedure=None):
        self.count = self.count + 1
        if procedure.devices == 1:
            procedure = RandomProcedure_1() 
        elif procedure.devices == 2:
           procedure = RandomProcedure_2()
        filename = '.\\test\\'test'+ str(self.count) + '.csv'  
        log.info("Constructing the Results with a data file: %s" % filename)
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)
        self.manager.queue(experiment)

```