pymeasure/pymeasure #273: StoppableThread join behaviour weird

bilderbuchi, 2020-05-14:
I noticed that the `join` method of `thread.StoppableThread` does not match the `threading.Thread.join` behaviour, especially concerning blocking. 
I realised that, if I have a StoppableThread running with some cleanup procedure (e.g. to cleanly shutdown instruments), and then I call stop() and join(), it immediately joins/terminates the thread without blocking until the cleanup has run its course.

The reason is that in https://github.com/ralph-group/pymeasure/commit/43217ab34a650e0730c66e64b5732b5f2ca0d231#diff-2133365464b62f18898d8808f6eddaa2R52 by @cjermain, there was a change in StoppableThread .join to to call threading.Thread.join(0) with a mandatory `0` timeout, which joins and times out immediately, even when the user supplied no `timeout` argument to the join.
I also don't fully grasp why we `_should_stop.wait` for the Event to set, and by default wait for 0 seconds.
Also, we set the _should_stop flag, but then give the thread by default no opportunity to actually complete its operation before joining immediately - I don't really get the purpose behind that.

I think we should change the default timeout of `StoppableThread.join(timeout=None)`, and use that timeout in the `super().join()` call. If the user want to immediately join, they can supply `timeout=0`.

In this context I think this test in test_thread.py does not do what we think it does, 
```
def test_thread_joining():
    t = StoppableThread()
    t.start()
    t.join()
    assert t.should_stop() is True
```

Here is a test that demonstrates the behaviour, comment/uncomment the Failing and Better sections
```
def test_thread_joining_waits():
    """Confirm that StoppableThread.join waits for the thread to finish when 
    joining without arguments.
    """
    t = StoppableThread()
    t.run = lambda: sleep(0.5)  # some operation that takes some time
    t.start()
    assert t.is_alive()
    # Failing
    t.join()  # this should IMO, but does not wait for the thread to finish
    assert t.should_stop() is True  # but we set the Event correctly
    # Better
    # If I force the call of the threading.Thread.join, this works as expected.
    # super(StoppableThread, t).join()

    assert not t.is_alive() # Is the thread dead after joining without giving a timeout?
```

msmttchr, 2021-06-20:
My understanding of `StoppableThread` class.

According to the comments in the code, the `StoppableThread` exists to avoid that a `join` to a thread becomes insensitive to keyboard interrupt.
Actually a `StoppableThread` implements also a thread-safe communication mechanism to:

1. Tell the thread to stop (call to method `stop`)
2. Inform other thread that the thread is in terminated state (call to method `should_stop`)

Since the mechanism for inter-thread communication is based on `Event` class, the mentioned bug https://bugs.python.org/issue35935 force to use a workaround with `InterruptableEvent` class.

The other conventions are:

- When the `Event` is set, then the thread is terminated or it will eventually terminate
- A `StoppableThread` should call method `stop` upon completion.
- If some other thread calls the `stop` method, we should assume that the `StoppableThread` will eventually check for it and terminates.

Apparently, the join behavior is as follow:

1. Wait for `Event` to be set within a timeout (possibly infinite)
2. If the timeout expires, without the `Event` being set, the set the `Event`
3. Call a `Thread.join` with immediate timeout

If the default timeout is changed to `None` and the `StoppableThread`, for some reason, never terminates, then the join call will be stuck in step 1 above and the following `join` in step 3 will never be called.

So, I guess that changing the default timeout to `None` will make the `join` more similar to default threading `join`.
On the other hand, the `join(0)` should not be changed, since in that point the `Event` is set and by convention the thread is completed (you could argue then on the need to call the `join` at all)

On another point, I am not entirely sure about the need of the `InterruptableEvent` `_should_stop` attribute,  I would change it to a simple boolean flag initialized to `False`.
I see no race condition on this variable since all the thread either read it or set it to `True`. I would change the join implementation to do a sequence of join with small timeout (e.g. 0.1 s) until the timeout is reached or forever and eventually I set the flag to `True` and return.

bilderbuchi, 2021-07-23:
I agree with all you said (except for the last paragraph). 
I think what I argued back then (and now) is that, if all we want to add is interruptability, we should keep the behaviour of our `StoppableThread` as close as possible to `Thread`. That IMO includes the user-observable behaviour of `join()` with different timeouts. 
I think you agree above that just changing the function signature to `join(self, timeout=None)` would achieve the matching behaviour that I strive for.  I agree that in that case the `super().join(0)` should stay, as in effect the `_should_stop` event takes over the role of `Thread.join`, with the benefit of interruptability.

I disagree with your last paragraph in that I think it's nice that there is some separation of concerns between Thread-tasks and interruptability, and it's already present and maybe used by others. 
I'm assuming in your proposed revised join implementation, you would always check for that boolean flag after every small join has timed out, correct?

bilderbuchi, 2021-07-23:
I think an improved test method related to our instrument-related usage would be something like the following run method:
```
def run(self):
    try:
        while not self.should_stop():
            print('working')
            sleep(0.1)
    except Exception:
        print('Was interrupted!')
    finally:
        print('Cleaning up')
        sleep(0.1)
        print('Finished cleaning up')
        raise
```
and then use this to put a `StoppableThread` through its paces with different parameter configurations, and check that the cleanup always finishes. 
Can we meaningfully test in the test suite that its join() method is interruptable by `KeyboardInterrupt` and still completes the cleanup? Otherwise, we have to do it manually, once. ( with human-compatible sleep durations :D)

msmttchr, 2021-08-01:
> I agree with all you said (except for the last paragraph).
> I think what I argued back then (and now) is that, if all we want to add is interruptability, we should keep the behaviour of our [...]

My idea was to implement something like this:
```
class StoppableThread(Thread):
    """ Base class for Threads which require the ability
    to be stopped by a thread-safe method call
    """

    def __init__(self):
        super().__init__()
        self._should_stop = False

    def join(self, timeout=None):
        """ Wrapper to threading.join API to make threads responsive to KeyboardInterrupt.

        See also :meth:`threading.join`
        """

        try:
            super().join(timeout)
        except (KeyboardInterrupt, SystemExit):
            self.stop()

        # Make sure thread is in stop state
        self.stop()
        if self.is_alive():
            # Give chance to terminate cleanly
            super().join(0.1)
            
        return None

    def stop(self):
        self._should_stop = True

    def should_stop(self):
        return self._should_stop

    def __repr__(self):
        return "<%s(should_stop=%s)>" % (
            self.__class__.__name__, self.should_stop())
```
Apparently only main thread is receiving `KeyboardInterrupt` signal and `join` in my version of python (3.8.10) is responsive to `KeyboardInterrupt`.

As you can see, the class interface does not change and code is much more simple than original implementation.

msmttchr, 2021-08-02:
```
class StoppableThread(Thread):
    """ Base class for Threads which require the ability
    to be stopped by a thread-safe method call
    """

    def __init__(self):
        super().__init__()
        self._should_stop = False

    def join(self, timeout=None):
        """ Wrapper to threading.join API to make threads responsive to KeyboardInterrupt.

        See also :meth:`threading.join`
        """

        def is_timeout(start, timeout):
            if timeout is None:
                return False
            else:
                return (time() < (start+timeout))

        start = time()
        try:
            while not is_timeout(start, timeout) and self.is_alive():
                super().join(0.1)
        except (KeyboardInterrupt, SystemExit):
            self.stop()

        # Make sure thread is in stop state
        self.stop()
        if self.is_alive():
            # Give chance to terminate cleanly
            super().join(0.1)
            
        return None

    def stop(self):
        self._should_stop = True

    def should_stop(self):
        return self._should_stop

    def __repr__(self):
        return "<%s(should_stop=%s)>" % (
            self.__class__.__name__, self.should_stop())

```
Improved version tested also in Windows