pymeasure/pymeasure #279: Adding new instruments?

reliable-bam, 2020-05-30:
I'm very new to python, but need to develop some code for an RF test bench in a hurry. I've done SCPI / GPIB command level control in other environments before, and I was hoping there was a quick way to convert / import drivers from other formats rather than banging out the code myself. Is there a sticky instruction or other generic script to add new instruments? Specifically I need an Agilent power meter (E441X type), and Agilent datalogger mainframe 34980A

Thanks in advance!

bilderbuchi, 2020-05-31:
There is not automated import function. I think the best approach (and what I mostly end up doing) is to get the manual and implement the commands you need, and test those. If it's a straightforward SCPI device, the effort is mostly quite limited - define a class and implement a handful of control/measurement/setting properties, and some methods for stuff that is not really usable as a python *property*.

If you are stuck and don't know how to implement a certain command/method, check [the docs section on how to add instruments](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#writing-properties) for some best practice, and check the [available instruments](https://pymeasure.readthedocs.io/en/latest/api/instruments/index.html) for similar devices, to use as a reference for how to implement stuff.

The effort increases when devices are complex (with channels and complicated mode configurations), or the API surface/number of commands is large. 

Does that help?

reliable-bam, 2020-06-04:
It helps to know what's what. I made a basic script for addressing and measuring a Keysight E4417A with dual power heads, and it seems ok. I can go back and add more functionality before submitting it to the instrument library. I'm still quite new, and haven't gotten all the style, tricks, and grammar down yet.

import logging

log = logging.getLogger(__name__)
log.addHandler(logging.NullHandler())

from pymeasure.instruments import Instrument
from pymeasure.instruments.validators import truncated_range
from pyvisa.errors import VisaIOError

from pymeasure.adapters import VISAAdapter

PMTrigDel=0.0005  #trigger delay for usb power sensors
frequency = 6000 #constant for debug of power meter

class keysightE441X_PM(Instrument):
    """ Represents the Keysight E441X Power Meter
    interface for interacting with the instrument.

    .. code-block:: python

    """

    chA = Instrument.measurement("FETCH2:POW:AC? DEF,3,(@1)", "measure channel A, 3-dig resolution")

    chB = Instrument.measurement("FETCH2:POW:AC? DEF,3,(@2)", "measure channel A, 3-dig resolution")

    def measA(self):
        return self.chA

    def measB(self):
        return self.chB
       
       # preset = Instrument.write("SYST:PRES", "Preset the power meter")

       #freq = Instrument.write("FREQ %sMHz" % 'frequency', "Set the frequency in MHz")

       #chA_freq = Instrument.write("SENS1:FREQ %sMHz" % 'frequency', "Set power head freq in MHz")

       #chB_freq = Instrument.write("SENS2:FREQ %sMHz" % 'frequency', "Set power head freq in MHz")

       # chA_CF = Instrument.control(

       # chB_CF = Instrument.control(

       #units_dBm = Instrument.write("UNIT:POW DBM", "measure in dBm")

       #gain_meas = Instrument.measurement("MEAS1:DIFF:REL? DEF,3,(@2),(@1)", "(B - A), 3-dig resolution")
        
       #Instrument.control("INIT:CONT:All 1")  # Set it to free run model
           #errs=(pmeter.query("SYST:ERR?"))
           #if "No error" not in errs:
               #print("function setupPowerMeter errors - resourceID=%s  and frequency=%s " %(resource_ID,frequency))
               #print(errs)
           #pmeter.close()
           #time.sleep(1)

    def __init__(self, adapter, **kwargs):
        super(keysightE441X_PM, self).__init__(
            adapter, "Keysight E441X power meter", **kwargs
        )
        self.timeout = 30000
        # format: output ascii
        self.write("FORM ASC")
	
    def check_errors(self):
        """ Read all errors from the instrument."""
        while True:
                err = self.values(":SYST:ERR?")
                if int(err[0]) != 0:
                        errmsg = "Keysight E441X: %s: %s" % (err[0],err[1])
                        log.error(errmsg + '\n')
                else:
                        break

axs5489, 2020-06-11:
I'm quite new to pymeasure (& github apparently), but I've been writing my own instruments classes modeled somewhat after pymeasure Instruments. Sorry not sorry for tabs.

`from pymeasure.instrument import Instrument
from pymeasure.validators import strict_range, strict_discrete_set
import time

class AgilentE4418B(Instrument):
	def __init__(self, name, adapter, enableSCPI=True, **kwargs):
		super(PowerMeter, self).__init__(name, adapter, enableSCPI, **kwargs)
		self._num_channels = 1
	
	offset = Instrument.control("CALC:GAIN?", "CALC:GAIN %d", "Offset, in dB or W",
							strict_range, [-100, 100]
							)
	
	offset_enable = Instrument.control("CALC:GAIN:STATE?", "CALC:GAIN:STATE %s", "Offset, ON or OFF",
							strict_discrete_set, ["ON", "OFF"]
							)
	
	units = Instrument.control("UNIT:POW?", "UNIT:POW %s", "Units, in dB or W",
							strict_discrete_set, ["W", "DBM"]
							)
	
	power = Instrument.measurement("FETC?", "Power, in dB or W")

class AgilentE4419(PowerMeter):
	def __init__(self, name, adapter, enableSCPI=True, **kwargs):
		super(DualPowerMeter, self).__init__(name, adapter, enableSCPI, **kwargs)
		self._num_channels = 2
		
	power_1 = Instrument.measurement("FETC?", "Power, in dB or W")
	power_2 = Instrument.measurement("FETC?", "Power, in dB or W")
	power_ratio = Instrument.measurement("FETC?", "Power, in dB or W")
	power_rel = Instrument.measurement("FETC?", "Power, in dB or W")
	
	units_2 = Instrument.control("UNIT2:POW?", "UNIT2:POW %s", "Units, in dB or W",
							  strict_discrete_set, ["W", "DBM"]
							  )
	
	units_ratio = Instrument.control("UNIT:POW:RAT?", "UNIT:POW:RAT %s", "Units, in dB or W",
							  strict_discrete_set, ["PCT", "DBM"]
							  )`

bilderbuchi, 2020-06-11:
You can put code samples into blocks by putting 3 backticks (```) in the lines before and after. I fixed that already here.
Yeah, those look like pretty typical implementations. one remark: @reliable-bam I don't know why you have both a property and a measA() method, that seems redundant to me.

reliable-bam, 2020-06-12:
@bilderbuchi No reason, except just learning how to use classes, methods, and the like.

@axs5489 Appreciate the sample code. I definitely want to implement good instruments, and will replace what I have with yours!

closed 2020-06-16 by reliable-bam