pymeasure/pymeasure #280: Running scripts in sequence?

reliable-bam, 2020-06-05:
I need to call several different parametric sweeps and save data at different time intervals, and am struggling on the implementation with procedures / experiments. I have a DC voltage sweep derived from the Keithley IV process/GUI example, and have a separate script for an RF power sweep controlling 2 instruments (Keithley SMU and Agilent power meter). I would like to have a main control program that calls these sweeps at regular time intervals during an environmental test I'm setting up on an amplifier. Is there a way of calling these separate sweeps from a main program, and having them save their data, while the main program performs the timing and call functions? The exact instruments used are irrelevant. I just need to know how I need to structure the main and sub scripts so that the GUI for the top level controls and executes the lower level scripts with timing.

Thanks,
Brad

PYmeas, 2020-07-29:
@reliable-bam I am looking for something similar but rather write each procedure separately and then import each of them in a main script. The issue i face now is that when i import my pymeasure e.g experiment.py script it doesnt execute and no errors displayed while the other instructions execute. Doing the other way around i.e. the instructions into my pymeasure module works but i have no control of the sequence of execution? say i have my experiment script using pymeasure (experiment.py) and i import into a main script to such as for instance:
Print(`hello`)
import experiment
print( `bye')

i get the Hello and bye in output but no execution of exepriments.py which should launch the managewindow and no error but if anybody has a hint, I appreciate it.

PYmeas, 2020-08-03:
@reliable-bam using a main as below file i can call 2 experiments running in parallel but on 2 different windows, i could not manage to have what you defined in your queue() with your sequence of procedures. i only get the first one executed and plotted while the second one is ignored even by putting all procedure and managedwindow in same file. did you managed to have your sequence executed and displayed?

``` ruby
import experiment_1 as exp_1
import experiment_2 as exp_2

if __name__ == "__main__":
    app = QtGui.QApplication(sys.argv)
    window_1 = exp_1.MainWindow_1()
    window_1.show()
    window_2 = exp_2.MainWindow_2()
    window_2.show()
    sys.exit(app.exec_())
```