pymeasure/pymeasure #296: Question regarding prefered implementation of non-SCPI devices

Applepi, 2020-06-30:
What is the preferred approach to add support for instruments that do not use SCPI to interface with the adapters. My thinking would be to write an interpreter to convert the non-ascii commands to SCPI commands, the reason for this being that some devices such as thorlabs rotation stages and such require to use either DLL or to send direct serial commands using APT and they possible bytes for the commands do not fall within the readable ASCII table. 

I want to add support for GRBL serial motion controllers as well as thorlabs APT devices, and I just wanted to touch base and see what the preferred approach for these would be. 

I think for the GRBL side of things it should be easy as all the commands are ascii sent over serial, but the main issue would be for me to write some sort of Throlabs APT to SPCI interpreter.

bilderbuchi, 2020-07-01:
Some discussion around frame-based communication can be found in the comments of #239. We think it implementing a separate Adapter class for frame-based communication would be most useful, maybe based on pyVISA RegisterBasedResource, but I have not yet investigated this in detail due to lack of time.

moritzj29, 2021-03-25:
since we discussed in the PR mentioned above: there appears to be an implementation of Thorlabs APT devices in Python, based on the underlying serial protocol (instead of using the DLL):

https://gitlab.com/yaq/thorlabs-apt-protocol

maybe this is useful for you.

ksunden, 2021-03-25:
Hi, author of that `thorlabs-apt-protocol` package here. 

If you have any questions, feel free to ask. It is mostly just wrapping the messages in python functions to be readable/handle the struct packing/unpacking. It itself does _precisely zero_ state maintenance or even direct communication (though you can hand it a pyserial or other file like serial pipe and it will allow you to iterate over responses).

If you want to see an example of it being used, I have an implementation of a yaq (https://yaq.fyi) daemon for thorlabs motors [here](https://gitlab.com/yaq/yaqd-thorlabs/-/blob/master/yaqd_thorlabs/_thorlabs_apt_motor.py).

bilderbuchi, 2021-04-27:
Thanks, appreciate the offer and the link to your implementation, that is quite useful. Oh, the APT implementation is quite large! :-o

In any case, I think we need to keep in mind that we have some base layer that can be used by ~arbitrary protocols, not only APT - it's big but not the only thing out there.

bilderbuchi, 2022-05-14:
#621 add an instrument with a command_word-register-length kind of syntax, so this fits into this frame-based communication scope.