pymeasure/pymeasure #299: how to uptade a parameter defined in the IVProcedure in the class MainWindow(ManagedWindow) then read it as a string

PYmeas, 2020-07-19:
I am using the managedwindow to live plot IV from a keithley2420.  It works but for some reason i cannot pass a parameter defined in the IVprocedure class into the queue function. here i define a parameter **_mode_** that takes the nature of the measurement 2 or 4 wires. when i queue experiments i can change that parameter in the GUI and record a new set of data. i can check in the keithley that the 4w mode is on and in the data file i do see the right mode parameter. i am trying to get that parameter as a string for my filename so i typed in str(self.mode), but the mode is stuck to the default value defined in the parameter, it is not updated after typing in the new value in the GUI whereas the right data appears in the file?.  i use fake data to test the code below.

```ruby
import logging
log = logging.getLogger(__name__)
log.addHandler(logging.NullHandler())

import sys
import tempfile
from tempfile import NamedTemporaryFile
import random
import datetime
from datetime import datetime as dt, timedelta
import time
from time import sleep
import pandas as pd
from pymeasure.log import console_log
from pymeasure.display.Qt import QtGui
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import Procedure, Results
from pymeasure.experiment import IntegerParameter, FloatParameter, Parameter

class RandomProcedure(Procedure):
    iterations = IntegerParameter('Loop Iterations', default=10)
    delay = FloatParameter('Delay Time', units='s', default=0.2)
    seed = Parameter('Random Seed', default='20')
    mode = IntegerParameter('2 or 4 probes measurement', default=2)

    DATA_COLUMNS = ['Iteration', 'Timecode', 'Elapsed', 'Random Number']

    def startup(self):
        log.info("Setting the time of the random number generator")
        random.seed(self.seed)

    def execute(self):
        log.info("Starting the loop of %d iterations" % self.iterations)
        start = datetime.datetime.now()
        for i in range(self.iterations):
            timecode = dt.now()
            elapsed = (datetime.datetime.now() - start).total_seconds()
            data = {
                # 'Mode': self.mode,
                'Iteration': i,
                'Timecode': timecode,
                'Elapsed': elapsed,
                'Random Number': random.randint(1, 100)
            }
            self.emit('results', data)
            log.debug("Emitting results: %s" % data)
            sleep(self.delay)
            if self.should_stop():
                log.warning("Caught the stop flag in the procedure")
                break

class MainWindow(ManagedWindow, RandomProcedure):

    def __init__(self):
        super(MainWindow, self).__init__(
            procedure_class=RandomProcedure,
            inputs=['iterations', 'delay', 'seed', 'mode'],
            displays=['iterations', 'delay', 'seed', 'mode'],
            x_axis='Iteration',
            y_axis='Random Number',
            sequencer=True,  # Added line
            sequencer_inputs=['iterations', 'delay', 'seed', 'mode'],  # Added line
            # sequence_file="gui_sequencer_example_sequence.txt",  # Added line, optional
        )
        self.setWindowTitle('GUI Example')
        self.count = 0

    def queue(self, *, procedure=None):
        self.count = self.count + 1
        filename = 'D:\\test\\' + str(dt.now().strftime("%m%d%Y"))+'_' + 'mode_' + str(self.mode) + '_' + str(self.count) + '.csv'
        log.info("Constructing the Results with a data file: %s" % filename)
        if procedure is None:
            procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)
        self.manager.queue(experiment)
        # mode = self.mode

if __name__ == "__main__":
    app = QtGui.QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec_())

```

closed 2020-07-24 by PYmeas

Daivesd, 2020-08-01:
Hi! I may have a suggestion for your issue which is probably more a workaround than a proper solution ...
Since both the inputs and sequencer_inputs objects defined in the __init__ istance of your ManagedWindow class belong to qt textbox widgets, you could resort to the textbox method text() to recover the string in the textbox.
In a nutshell: filename = 'D:\\test\\' + str(dt.now().strftime("%m%d%Y"))+'_' + 'mode_' + **self.sequencer_inputs.mode.text()** + '_' + str(self.count) + '.csv' if the mode input is given as a sequencer input, or rather self.inputs.text() if the mode is specified in the ManagedWindow textbox widget
I hope this helps you out with your problem!

PYmeas, 2020-08-01:
Thanks for your time @Daivesd , tested it , it does work for the update of the filename but then the execute is stuck so i believe the variable is not updated in the queue or the format is wrong. I ve been proposed a working solution here [Trouble with scoping #302](https://github.com/ralph-group/pymeasure/issues/302). I am now able to run various experiments and save with different filenames and modes.

Daivesd, 2020-08-01:
Ok sorry I didn't realize this issue was already solved in another topic! Actually the 'real' solution is even more general and will work even when using sequencer in ManagedWindow as it sets the relevant parts of the filename in the procedure itself! 
thank you very much for pointing that topic out