pymeasure/pymeasure #300: Keithley 2400 does not respond to commands

ccory2004, 2020-07-23:
When using Keithley 2400 with Pymeasure over a Serial Adapter(RS-232), it will not print out its own identifier. However, it is evident that the connection has been applied to the sourcemeter, because the REM label, for remote control, is lit up on the screen. 

I feel I might be missing something obvious here, but I'm not sure what.

The code I am running:
`from pymeasure.adapters import SerialAdapter
from pymeasure.instruments.keithley import Keithley2400
adapter = SerialAdapter("COM3")
keithley = Keithley2400(adapter)
print(keithley.id)`

bilderbuchi, 2020-07-23:
Does it work if you don't construct a serialdapter separately, i.e. pass the port to the constructor as string?

ccory2004, 2020-07-23:
When I change line 4 to `keithley = Keithley2400(SerialAdapter("COM3"))` throws the error "AttributeError: 'SerialAdapter' object has no attribute 'connection'"
If I list resources with PyVISA, it gives me the resource "ASRL3::INSTR". Putting that into the script throws the error "NotImplementedError: Please upgrade PyVISA to version 1.8 or later."
Checking the version of PyVISA get me '1.10.1', which is greater than 1.8. Maybe similar to #222?

PYmeas, 2020-07-24:
I use "from pymeasure.instruments.keithley import Keithley2400" then "sourcemeter = Keithley2400("ASRLxx::INSTR")" with an RS232 I had a similar issue but that was due to the terminator on the Keithley that must be set to CR+LF terminator as most computer. you can also test with the NI Visa interface first to check if any proper communication the the REM tag alone is not enough. make sure you get the instrument ID.

bilderbuchi, 2020-07-24:
@ccory2004 whatis your pymeasure version? Is this up to date? Can you try with latest master to make sure that the problem has not been fixed in the meantime?

ccory2004, 2020-07-24:
@PYmeas changing the line to `keithley = Keithley2400("ASRL3::INSTR")` errors with "NotImplementedError: Please upgrade PyVISA to version 1.8 or later." I'm not sure what you mean by the CR+LF terminator. Is that a setting I have to change? I am able to connect to the Keithley 2400 with PyVISA and query the device, outputting the instrument ID. It is "KEITHLEY INSTRUMENTS INC.,MODEL 2400,1369030,C33   Mar 31 2015 09:32:39/A02  /V/K"

@bilderbuchi My PyMeasure version is '0.8.0', the latest version of PyMeasure

Thank you both for your help!

PYmeas, 2020-07-24:
CR+LF terminator must be set on the keithley instrument : configuration/
communication/ rs232/terminator.  First time i installed pymeasure few
weeks ago i could not get the instrument ID and that was only due to this
terminator setting. Now have no issue with rs232.

ccory2004, 2020-07-24:
Hm...it seems the terminator was already set to <CR+LF>, and I was able to get instrument ID by PyVISA through querying *IDN?, but not through PyMeasure by querying sourcemeter.id.

PYmeas, 2020-07-24:
These terminator must be same on computer and instrument. Usually on
computer it s cr +lf it may not be the case for you. just try the other
terminator on the instrument to check if one combination fit.

ccory2004, 2020-07-24:
@PYmeas It looks like Cr +LF is standard as the terminator on Windows systems? If that is the case, then both the Keithley and my computer are set to the same terminator. Nonetheless, I am still getting errors.

bilderbuchi, 2020-07-24:
Cr-lf is the standard *line* terminator for text files in Windows, yes, but this is only tangentially related to communication terminators. You can adjust the terminator setting in the keithley instrument settings (i.e. in the device in your table), and if this does not match the setting in the pymeasure instrument then comms can fail.

ccory2004, 2020-07-26:
@bilderbuchi I've tried all 4 terminators to no avail. The code I tested with was: 
```from pymeasure.adapters import SerialAdapter
from pymeasure.instruments.keithley import Keithley2400
adapter = SerialAdapter("COM3")
keithley = Keithley2400(adapter)
print(keithley.id)```

To alleviate any problems with potential misconfiguration, I installed a fresh copy of Windows, and then installed the necessary modules and applications according to instructions a Tektronix applications engineer sent me, available here: https://github.com/tektronix/keithley/tree/master/Instrument_Examples/Instructables/Get_Started_with_Instr_Control_Python.

No avail...running the code above produces no result, no error codes or any information displayed on the terminal.

PYmeas, 2020-07-27:
@ccory2004 I connected to a Keithley 2420. i ve tried the following options:
**1** i checked using NI Visa app if any connection issue (terminator issue mentioned previously). I do get my instrument ID.
KEITHLEY INSTRUMENTS INC.,MODEL 2420,1314433,C34 Sep 21 2016 15:30:00/A02  /H/L

**2** i tried your lines:
``` ruby
from pymeasure.adapters import SerialAdapter
from pymeasure.instruments.keithley import Keithley2400
adapter = SerialAdapter("COM6")
keithley = Keithley2400(adapter)
print(keithley.id)
```
_Output_:  the REM tag in on but i don't get the ID on the output.

**3** i tried using Pyvisa:
``` ruby
import pyvisa
rm = pyvisa.ResourceManager()
rm.list_resources()
('ASRL6::INSTR')
inst = rm.open_resource('ASRL6::INSTR')
print(inst.query("*IDN?"))
```
_Output_:  the REM tag in on and i get my instrument ID

**4**: I tried with Pymeasure but using VISAAdapter:
``` ruby
from pymeasure.instruments.keithley import Keithley2400
from pymeasure.adapters import VISAAdapter
adapter = VISAAdapter("ASRL6::INSTR")
keithley = Keithley2400(adapter)
print(keithley.id)
```
_Output_:  the REM tag in on and i get my instrument ID

There seems to be an issue with the Serialadapter. I am using a Keithley 2420 with an RS 232 for various data acquisition i just  declare in experiments
``` ruby
from pymeasure.instruments.keithley import Keithley2400
```
and
``` ruby
sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
```
the process has been made transparent for the user, you could easily change to GPIB or else with minimum lines at least for Keithley`s. Out of curiosity is there any advantage to go necessarily with Serialdapter syntax?

ccory2004, 2020-07-27:
@PYmeas The only reason I was using SerialAdapter is because it was recommended as a solution in #222. The VISAAdapter works great, and the 4th snippet of code works for me. 
I originally thought the VISAAdapter could only be used with GPIB-USB devices from NI, but it looks like RS-232 works just as well.
Thank you for your help!

closed 2020-07-27 by ccory2004