pymeasure/pymeasure #301: VI_WARN_CONFIG_NLOADED not understood

Mike-HubGit, 2020-07-24:
Starting from the very simple code :
```python
from pymeasure.instruments import Instrument
scope = Instrument('USB0::0x0699::0x03AC::C011932::INSTR','tektro')
```
I always get a warning :
```
/opt/miniconda3/lib/python3.7/site-packages/pyvisa/ctwrapper/functions.py:1227: VisaIOWarning: VI_WARN_CONFIG_NLOADED (1073676407): The specified configuration either does not exist or could not be loaded. VISA-specified defaults will be used.
  ret = library.viOpenDefaultRM(byref(session))
```

I run pymeasure un Mac. Someone knows ?

Best Regards,
  Mike

bilderbuchi, 2020-07-24:
Do you get the same when using pure pyvisa, without pymeasure Instrument in between?

Mike-HubGit, 2020-07-27:
I checked and yes it does. A code as simple as :
```python
import pyvisa
rm = pyvisa.ResourceManager()
```
drops this warning. Any idea why ?

bilderbuchi, 2020-08-02:
Thanks. Sorry, I have no idea what could cause this, but this makes me think this is not something caused by pymeasure. Can you take this up with [PyVISA](https://github.com/pyvisa/pyvisa/issues), please?

closed 2020-08-02 by bilderbuchi

Mike-HubGit, 2020-08-02:
I will do.