pymeasure/pymeasure #302: Trouble with scoping

physicsninja, 2020-07-26:
Hello,

I can save multiples files generated from queued experiments from Keithley 2420 but i want to use a label _type_mode_ in the filename that is updated in the execute() from the Randomprocedure class. here i use fake data for testing purpose I am getting a NameError: name 'type_mode' is not defined. I understood that variable exit only within a class but i thought by inheriting Managedwindow from Randomprocedure this could be possible? 

``` ruby
import logging
log = logging.getLogger(__name__)
log.addHandler(logging.NullHandler())

import sys
import tempfile
from tempfile import NamedTemporaryFile
import random
import datetime
from datetime import datetime as dt, timedelta
import time
from time import sleep
import pandas as pd
from pymeasure.log import console_log
from pymeasure.display.Qt import QtGui
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import Procedure, Results
from pymeasure.experiment import IntegerParameter, FloatParameter, Parameter

class RandomProcedure(Procedure):

    iterations = IntegerParameter('Loop Iterations', default=10)
    delay = FloatParameter('Delay Time', units='s', default=0.2)
    seed = Parameter('Random Seed', default='20')
    mode = IntegerParameter('Measurement mode : 2 or 4 probes', default=2)

    DATA_COLUMNS = ['Mode', 'Iteration', 'Timecode', 'Elapsed', 'Random Number']

    def startup(self):
        log.info("Setting the time of the random number generator")
        random.seed(self.seed)

    def execute(self):
        log.info("Starting the loop of %d iterations" % self.iterations)
        start = datetime.datetime.now()
        for i in range(self.iterations):
            timecode = dt.now()
            elapsed = (datetime.datetime.now() - start).total_seconds()
            data = {
                'Mode': self.mode,
                'Iteration': i,
                'Timecode': timecode,
                'Elapsed': elapsed,
                'Random Number': random.randint(1, 100)
            }
            self.emit('results', data)
            log.debug("Emitting results: %s" % data)
            sleep(self.delay)
            if self.should_stop():
                log.warning("Caught the stop flag in the procedure")
                break
        if self.mode == 2:   # it s also data[0, 0]
            type_mode = '2_wire_mode'
        elif self.mode == 4:
            type_mode = '4_wire_mode'
        else:
            type_mode = 'undefined'
        print(type_mode)  # i want type_mode to appear in the filename in the queue() function depending on the result here

class MainWindow(ManagedWindow, RandomProcedure):

    def __init__(self):
        super(MainWindow, self).__init__(
            procedure_class=RandomProcedure,
            inputs=['iterations', 'delay', 'seed', 'mode'],
            displays=['iterations', 'delay', 'seed', 'mode'],
            x_axis='Iteration',
            y_axis='Random Number',
            sequencer=True,  # Added line
            sequencer_inputs=['iterations', 'delay', 'seed', 'mode'],  # Added line
            # sequence_file="gui_sequencer_example_sequence.txt",  # Added line, optional
        )
        self.setWindowTitle('GUI Example')
        self.count = 0

    def queue(self, *, procedure=None):
        global type_mode
        self.count = self.count + 1
        filename = 'D:\\test\\' + str(dt.now().strftime("%m%d%Y")) + '_' + 'mode_' + str(type_mode) + '_' + str(self.count) + '.csv'
        #if put self.mode instead of type_mode it works but always mode 2 printed
        log.info("Constructing the Results with a data file: %s" % filename)
        if procedure is None:
            procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)
        self.manager.queue(experiment)

if __name__ == "__main__":
    app = QtGui.QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec_())

```

_Originally posted by @PYmeas in https://github.com/ralph-group/pymeasure/issues/75#issuecomment-663973332_

physicsninja, 2020-07-26:
I don't remember enough about how the code is structure to explain why I think that `type_mode` defined in the `execute` function of `RandomProcedure` will not escape the local scope of the `execute` function even with the global invocation earlier, but suffice it to say that I don't think it will. Someone can correct me if I am wrong. 

The reason you always get 2 when you invoke `self.mode` in the `queue` definition is that the instance of `MainWindow` referenced by `self` only knows about the skeleton of `RandomProcedure` (i.e. the defaults). The instance of `RandomProcedure` with the parameters of a given queued experiment are in the `procedure` instance that you generate later in the `queue` function. 

All this is to say, that if you redefine your `queue` function like:
```ruby
def queue(self, *, procedure=None):
    self.count = self.count + 1
    if procedure is None:
        procedure = self.make_procedure()
    # Now we have an instance of the procedure with whatever inputs have actually been set
    type_mode = 'undefined'  # let's just start with undefined so we don't need that else clause and don't upset linters
    if procedure.mode == 2:   # Now the mode will reflect the queued experiment and not just the default
        type_mode = '2_wire_mode'
    elif procedure.mode == 4:
        type_mode = '4_wire_mode'
     
    filename = 'D:\\test\\' + str(dt.now().strftime("%m%d%Y")) + '_' + 'mode_' + type_mode + '_' + str(self.count) + '.csv'
    log.info("Constructing the Results with a data file: %s" % filename)
    results = Results(procedure, filename)
    experiment = self.new_experiment(results)
    self.manager.queue(experiment)
```
You should get the behavior that you want. Basically just move your `type_mode` logic one level up into the `MainWindow` so that you can apply it to the `procedure`'s as they are queued by the user. If this doesn't work, please reopen this issue.

closed 2020-07-26 by physicsninja

physicsninja, 2020-07-26:
Alternatively you could also keep the `type_mode` logic in the procedure, and just make it a class attribute:

```ruby
class RandomProcedure(Procedure):

    iterations = IntegerParameter('Loop Iterations', default=10)
    delay = FloatParameter('Delay Time', units='s', default=0.2)
    seed = Parameter('Random Seed', default='20')
    mode = IntegerParameter('Measurement mode : 2 or 4 probes', default=2)

    DATA_COLUMNS = ['Mode', 'Iteration', 'Timecode', 'Elapsed', 'Random Number']

    def startup(self):
        log.info("Setting the time of the random number generator")
        random.seed(self.seed)

    def execute(self):
        log.info("Starting the loop of %d iterations" % self.iterations)
        start = datetime.datetime.now()
        for i in range(self.iterations):
            timecode = dt.now()
            elapsed = (datetime.datetime.now() - start).total_seconds()
            data = {
                'Mode': self.mode,
                'Iteration': i,
                'Timecode': timecode,
                'Elapsed': elapsed,
                'Random Number': random.randint(1, 100)
            }
            self.emit('results', data)
            log.debug("Emitting results: %s" % data)
            sleep(self.delay)
            if self.should_stop():
                log.warning("Caught the stop flag in the procedure")
                break
        self.type_mode = 'undefined'
        if self.mode == 2:   # it s also data[0, 0]
            self.type_mode = '2_wire_mode'
        elif self.mode == 4:
           self.type_mode = '4_wire_mode'
        print(self.type_mode)  # i want type_mode to appear in the filename in the queue() function depending on the result here
```
then you could call `procedure.type_mode` in the `queue` function
def queue(self, *, procedure=None):
    self.count = self.count + 1
    if procedure is None:
        procedure = self.make_procedure()
    # Now we have an instance of the procedure with whatever inputs have actually been set
    filename = 'D:\\test\\' + str(dt.now().strftime("%m%d%Y")) + '_' + 'mode_' + procedure.type_mode + '_' + str(self.count) + '.csv'
    log.info("Constructing the Results with a data file: %s" % filename)
    results = Results(procedure, filename)
    experiment = self.new_experiment(results)
    self.manager.queue(experiment)
```

PYmeas, 2020-07-26:
@physicsninja thanks a lot. option 1 is working fine. 
option 2 is actually the format i was searching but without considering the logic upshift of `procedure `. Nevertheless there is still an attribute error despite the declaration in `execute()` : error AttributeError: 'RandomProcedure' object has no attribute 'type_mode'.

physicsninja, 2020-07-26:
Oof, you're right, there would be that error in the second method I suggested. I didn't think it through. The `execute` method isn't invoked until the queued procedure is actually run, which doesn't happen until after it is queued, much later than when the filename needs to be produced

PYmeas, 2020-07-29:
@physicsninja having this procedure up and running in standalone. can i define various experiments using the same framework and then import each procedure as module in a main script? 
print('hello") # output hello
import experiment  # script using the pymeasure procedure for experiment  - this one doesnt execute?
import cleanfolder # a script to erase specific files after running the experiement (this executes)
print("bye") # output bye
I am asking because i try to import the previous code as a module but it doesn't execute and no error displayed?

PYmeas, 2020-07-30:
@physicsninja question solved