pymeasure/pymeasure #303: Mailing list?

qm2k, 2020-07-29:
Dear PyMeasure community! I could not find any mailing list where casual quesions on usage and contribution could be asked. Don't you think having one would be a good idea?

Meanwhile, here's my question:

I need to write a measurements automation code with the following requirements:
* solid 24/7 unattended operation
* reliable and early error detection
* easy addition of necessary drivers (we would need a lot more instruments than PyMeasure currently supports)
* compatibility with small headless Linux servers (think Raspberry Pi without X installed)

Would be nice to have:
* asyncio support
* unit support (cf #248)

Do you think PyMeasure in its current state would be a good start for this kind of project, as opposed to building a custom monolithic application from scratch?

paulgoulain, 2020-07-30:
Hi,

I don't have the knowledge to tell you if PyMeasure is the library you need for your project or not but out of curiosity, what would be the use case of asyncio in your project ?

qm2k, 2020-07-31:
> I don't have the knowledge to tell you if PyMeasure is the library you need for your project or not but out of curiosity, what would be the use case of asyncio in your project ?

To query multiple devices simultaneously. In the simpliest case, to control multiple identical setups and perform multiple identical experiments over different samples. Some equipment can be shared between setups so it is more complicated than just running several independent Python processes.

bilderbuchi, 2020-08-02:
Hi! Yes, we don't have a mailing list because as far as I know this has always been a low-traffic project. Maybe when [Github discussions](https://github.blog/2020-05-06-new-from-satellite-2020-github-codespaces-github-discussions-securing-code-in-private-repositories-and-more/#discussions) is generally available, we can easily add that aspect of communication here.

I would say it is in fact easy to add new instruments, see the [relevant section](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html) in the docs. You can also keep them in your project if you want and upstream them here when they are matured (there are too few maintainers with too little time, so things will take a while).

Note that if you don't find the GUI/procedure aspects appropriate for your needs, you can also use just the instrument driver portion of pymeasure (which already removes a lot of complexity), and create the interaction code yourself as you need. 

I think headless operation should not be a problem if you don't use the GUI parts, the one thing I don't know what happens at pymeasure install time - it will pull in pyqt and I don't know what _that_ does when there is no x server around.

Your applications sounds complex, so reliability & error detection will probably be mostly relevant for your code. I am not aware of memory leaks or somesuch in the pymeasure code.

As to asyncio, I am personally leaning more towards adding this via concurrent.futures instead of asyncio, because I want to keep the complexity low and not introduce async code and the associated event loops.
Note that PyVISA access is [thread-safe](https://pyvisa.readthedocs.io/en/latest/faq/faq.html?highlight=thread#is-pyvisa-thread-safe), so if your instruments are all accessed via PyVISA, you should be reasonably safe to talk to the instruments from different threads.

bilderbuchi, 2021-03-03:
We have a Github Discussions forum now, so I'll convert this :-)

closed 2021-03-03 by bilderbuchi