pymeasure/pymeasure #305: Keithley 2600 series driver

Daivesd, 2020-08-01:
Hi guys! First of all thank you very much for your great job developing this project, I find it not only very usefull for everyone working in a lab, but also a great opportunity to share know hows and bunch of knowledge 👍 
Now to the proper question: for my experimental setup I need to drive a Keithley sourcemeter belonging to the 2600 class; I was wondering if anybody was working on a functional driver for this instrument compatible with Pymeasure. 
I have written my own bunch of code to perform very basic measurements using pymeasure environment, but I guess it is too rudimental to be directly used in the project... 
If anybody has done something for this, I would totally love to collaborate to add this instrument to the list of instrument drivers in pymeasure package :D

bilderbuchi, 2020-08-02:
Great that you like it, thanks for the praise! :bow: 
I am not aware of anyone working on this. If it's a sourcemeter, maybe the implementation of the Keithley2400 or 2450 sourcemeters can be a template for your implementation? If the command set is identical, or a subset of your instrument's, it might even make sense to inherit from those and just add any needed tweaks/additional commands.

Daivesd, 2020-08-02:
Yes my starting point actually was the Keithley 2400 code ^^ the commands are kind of different, mainly because the 2600 series has two separate channels. 
I guess I'll keep on working to my version of the driver and try to pull it to the project asap :) Thanks!

bilderbuchi, 2020-08-02:
Check out [this instrument](https://github.com/ralph-group/pymeasure/blob/07cf309b8d113931525c8cc216cc704f2b7ff413/pymeasure/instruments/tektronix/afg3152c.py) for a way to implement channels in a device driver.

Daivesd, 2020-08-02:
Oh thank you very much! I actually went for a quick and dirty solution, creating two separate classes in which each channel command is hard-coded... not that pretty but still it's working for me :P I'll change the code to this more elegant way btw

Daivesd, 2020-09-21:
@bilderbuchi  Sorry to bother again... I'm rather new to github in general :\ I have written a working version for driving the keithley2600 and I have opened a pull request to merge my code. It seems to have passed some checks but I would like to know if I should perform some other steps to have that code merged in the project... it would be very useful to me in order to avoid some boring copy/paste procedures on other computers ... Thank you very  much!

bilderbuchi, 2020-09-22:
Thanks for your contribution, and the poke. Sorry for lingering so long - we are only 2 maintainers with very little spare time for this, so merging/reviewing takes a very long time unfortunately. :-(

In the meantime, have you already tried pip-installing from the [git repo directly](https://stackoverflow.com/questions/20101834/pip-install-from-git-repo-branch)? You should be able to install pymeasure with `pip install git+https://github.com/Daivesd/pymeasure.git@master` to get your current master state (or some other branch). 
You should also be able to install the [merge commit of your PR](https://stackoverflow.com/questions/13561618/pip-how-to-install-a-git-pull-request) in here with  `pip install git+https://github.com/ralph-group/pymeasure.git@refs/pull/307/merge` to get the version you are proposing to merge. 
This should maybe help avoid copy-paste work until we get out a new release.

Daivesd, 2020-09-22:
Thank you very much I didn't know that was something I could do!

ryan-cashen, 2021-07-22:
Any updates on the merging process? Looking to use a 2600 series device as well and would greatly appreciate using the work you've completed already.

bilderbuchi, 2021-07-23:
I'll take a look at #307 which adds that driver. If you immediately need to use it see [here](https://github.com/pymeasure/pymeasure/issues/305#issuecomment-696573274).

bilderbuchi, 2021-07-24:
Closed by #307

closed 2021-07-24 by bilderbuchi