pymeasure/pymeasure #312: Filename line edit in ManagedWindow

paulgoulain, 2020-08-28:
Hello,

I don't know if the option or the widget exists, but I couldn't find if the ManagedWindow has an option to add a line edit for the filename for the next precedure that will be queued. If it doesn't would you be interested in a pull request for this option ?

Paul

PYmeas, 2020-08-31:
@paulgoulain do you mean save each queued experiment under a different filename? i save each file by iterating a counter or date as below not sure if helps.

``` ruby
class MainWindow(ManagedWindow):

    def __init__(self):
        super(MainWindow, self).__init__(
            procedure_class=IVProcedure,
            inputs=['min_current', 'max_current', 'data_points', 'mode'],
            displays=['min_current', 'max_current', 'data_points', 'mode'],
            x_axis='Voltage (V)',
            y_axis='Current (A)',
            # sequencer=True,  # Added line
            # sequencer_inputs=['min_current', 'max_current', 'data_points', 'mode'],  # Added line
            # sequence_file="gui_sequencer_example_sequence.txt",  # Added line, optional if load a sequence file
        )
        self.setWindowTitle('IV procedure')
        self.count = 0

    def queue(self, *, procedure=None):
        self.count = self.count + 1
        if procedure is None:
            procedure = self.make_procedure()
        filename = '.\\test\\' + 'data_' + str(dt.now().strftime("%m%d%Y"))+'_' + str(self.count) + '.csv'
        log.info("Constructing the Results with a data file: %s" % filename)
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)
        self.manager.queue(experiment)
```

paulgoulain, 2020-08-31:
Hi,

Thanks for your input. I was thinking of a QLineEdit just above the Queue button (see screenshot)

[image]

For this exemple, I just added a QLabel and a QLineEdit in the window.py code of pymeasure. The content of the QLineEdit could be accessed in the queue function of the ManagedWindow (using self.filename_lineedit.text()). If this doesn't exist yet, I can provide a PR for this. Also the same can be done for the directory.

Paul

PYmeas, 2020-08-31:
@paulgoulain i see would be nice indeed especially for directory. Thanks

PYmeas, 2020-08-31:
@paulgoulain are you measuring IV with a Keitlhley ? if yes do you initialize in the startup method?

paulgoulain, 2020-08-31:
Yes, I use a Keithley 2401.

Here is my startup function.
```Python
def startup(self):
    log.info("Setting up instrument")
    self.keithley = Keithley2400("GPIB::8")
    self.keithley.apply_voltage()
    self.keithley.source_voltage_range = 10
    self.keithley.compliance_current = self.current_range
    self.keithley.source_voltage = 0
    self.keithley.enable_source()  
    self.keithley.measure_current() 
    sleep(2)
```

I just reused the IV exemple from pymeasure and adapted it for my use.

PYmeas, 2020-08-31:
@paulgoulain interesting i am doing same thing but i cannot initialize the instrument i use serial RS 232 i opened issue #315 for discussion.  i will test your lines in my codes. Thanks

PYmeas, 2020-09-01:
@paulgoulain It s really strange, i managed to get an IV but not by using the `startup `method but by initializing through a `def__ini__` method. it works but dont understand why. thanks for your input.
``` ruby
    # if use a def__ can have the instrument initializes and get measurements
    def __init__(self):
        super().__init__()
        log.info("Connecting and configuring the instrument")
        self.sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
        self.sourcemeter.reset()
        self.sourcemeter.use_front_terminals()
        self.sourcemeter.wires = self.mode  # type 4 for 4 terminals measurement
        self.sourcemeter.apply_current()
        # self.sourcemeter.source_current = self.min_current
        # sourcemeter.measure_resistance()
        self.sourcemeter.enable_source()

   # this method doesnt initialize the instrument and i get an attribute error
    def startup(self):
        log.info("Connecting and configuring the instrument")
        self.sourcemeter = Keithley2400("ASRL6::INSTR")  # using RS 232 COM6
        self.sourcemeter.reset()
        self.sourcemeter.use_front_terminals()
        self.sourcemeter.wires = self.mode  # type 4 for 4 terminals measurement
        self.sourcemeter.apply_current()
        # self.sourcemeter.source_current = self.min_current
        # sourcemeter.measure_resistance()
        self.sourcemeter.enable_source()
```

paulgoulain, 2020-09-01:
@PYmeas Strange indeed. Don't you have a GPIB connector ? It would be insightful if you can try with it and see if the startup function works or not.

PYmeas, 2020-09-02:
@paulgoulain no just on RS232 for now but for sure it s not a communication issue. on the iteration example tutorial it works fine. i will try to figure this out.

PYmeas, 2020-09-08:
@paulgoulain i solved the issue the parenthesis of the startup method were missing in the worker.py file. thanks