pymeasure/pymeasure #319: Keithley 2400

jskaggs99, 2020-09-18:
Can we include high impedance settings in the Keithley 2400 function library? 

```
def set_high_impedance(self):
        self.inst.write(":OUTP:SMOD HIMP")
        print('Set high impedance mode')
```

Something like this does not exist in the documentation currently, and would help out a lot of people. Thank you!

paulgoulain, 2020-09-25:
Hi, I'm preparing a PR for some functionalities that are currently not implemented in the Keithley 2400 driver. I can add this function.

Paul

jskaggs99, 2020-09-25:
That's awesome. Thank you so much

paulgoulain, 2020-09-25:
Looking at the manual, I think I would implement it using the Instrument.control function. It would look like that :

```python
    output_off_state = Instrument.control(
        ":OUTP:SMOD?", ":OUTP:SMOD %s",
        """ Some explanations """,
        validator=strict_discrete_set,
        values=['HIMP', 'NORM', 'ZERO', 'GUAR'],
        map_values=False)
```

Does it look ok for you ?

Paul

jskaggs99, 2020-09-25:
That looks great Paul

paulgoulain, 2020-09-25:
Great, I'll add it to my futur PR.

bilderbuchi, 2020-10-14:
Closed by #326, thank you!