pymeasure/pymeasure #320: Controlling equipment with pymeasure

byzantic, 2020-09-22:
Hi, I'm not sure if this has been covered before ..

I'm investigating whether PyMeasure is suitable for my needs. The Instrument handling seems great, and I made a quick stab at adding support for a new instrument, which has so far been successful.

However, what is the recommended method for controlling experimental equipment? For example, say I have some experimental hardware connected to my machine, and I can control it using a python interface that I have developed. I need to be able to set experimental variables, and start and stop the experiment. What is the best way to do that?

One way would be to just use my interface in scripts.

But would it be better to wrap my custom interface in an 'Instrument' wrapper? If so, how should I do that, and do I need to also abstract the adapter layer?

bilderbuchi, 2020-09-22:
I normally write a script and collect/connect all instruments/equipment (but I seldomly use the pymeasure GUI portions). Where possible, I try to use pymeasure Instruments for all of them, as the common API and handling (i.e. mainly property-based interaction) simplifies the use and implementation of the control logic. 
However, nothing stops you from just using your python interface as-is, but you may end up special-casing often - this depends on the specifics of your setup. 

Yeah, if you create an Instrument implementation for your python interface, you will also need an adapter. If whatever your thing uses to communicate does not correspond to one of the existing adapters, you'll probably need to implement a small shim or something, implementing at least read() and write() methods that pymeasure uses internally to talk to the instrument hardware. Again, depending on the specifics of your interface, maybe you could push the communication specifics into the adapter, i.e. implement a protocol to talk to your interface (as opposed to, e.g.  a serial connection).

Was that comprehensible?

byzantic, 2020-09-23:
Thanks for the reply .. OK, I think I understand, I would need to look at the Adapter class.

I have two types of things I need to connect.

The first is a motor controller that communicates over USB, but already has a python adapter layer supplied. I think in that case, I just need to provide an adapter with a connection parameter that corresponds to the connection string currently used with the library.

The second is custom hardware on an FPGA. I currently have a python layer that talks to a C++ shared lib that implements the low level communication to the device. My python layer provides some descriptor interfaces to the FPGA registers. In this case, would it be easier to just provide an instrument that doesn't use an adapter? It can't be used on any other machine, so the connectivity part of instruments is not really relevant.

I would _prefer_ to implement the Instrument interface, because it makes things easier later on, but I might have to do some hacky scripting first ..

bilderbuchi, 2020-09-23:
The thing is, pymeasure is based on a command-like interface - you send commands to your hardware (via an adapter over serial, usb, gpib, visa,...), and get results back. A Pymeasure Instrument is basically an adapter layer for this (as you already seem to have?) 
If you implement an adapter that interfaces such commands with your hardware or existing interface, that should work, but I am not sure of the cost/benefit (if you already have a "python layer"). An Instrument without an adapter doesn't make sense (it would always choose one), however the adapter could be pretty minimal (i.e. subclass Adapter and implement read() and write() as appropriate for your hardware/python layer, as I hinted above.

You can check #246 #296 for something a little outside that scope (frame-/registermap-based communication).

bilderbuchi, 2020-09-23:
btw, you can also check the [instrument list](https://instrumentkit.readthedocs.io/en/latest/apiref/index.html) of the awesome https://github.com/Galvant/InstrumentKit if at least your motor controller already exists.

byzantic, 2020-09-24:
Thanks for the replies Christoph. I have been playing around a bit, and I have come to the following conclusions.

The `Instrument` part of `PyMeasure` is a library for connecting to (mainly) SCPI instruments. It seems to work well, and I can connect to several instruments from different manufacturers; I had to implement simple test instrument classes for these, which don't do very much, but they show the principle. You need something like this library in order to support all the various pieces of T&M kit out there. The Adapter layer has to deal with all the ASCII based communication protocols.

Once an instrument is instantiated, though, it is accessed through a specific interface; the methods return values in the results domain, such as integers, floats, etc.

My hardware interface does this already, so once I have instantiated it, I can treat it like any other instrument. I would also expect that I can use instruments from `InstrumentKit` as well! (Thanks for the pointer - it looks very interesting)

The experiment layer (Procedure / Result / ManagedWindow) is, I think, independent of the Instrument layer, so it can be used with any of the techniques. Actually, I rather like the GUI, and I think it will serve my needs for the short term. There are some things that don't quite fit my workflow, and some facilities that I would like in addition. If I come up with anything that looks useful, I'll set up a fork and a PR.

I'll close this issue now, as I think it has been answered. Thanks for the help!

closed 2020-09-24 by byzantic

bilderbuchi, 2020-09-24:
I agree with your conclusions above.
Regarding the experiment layer as you call it: I'm not sure how tightly coupled this is to the Instrument layer. It might be that you will run into troubles if the instruments you use are not pymeasure Instruments, but I think you will notice quickly. If you keep the instantiation similar (first argument is the "port"/connection specifier) that should work I think.