pymeasure/pymeasure #321: Can't pickle <class 'sqlalchemy.orm.session.Session'>: it's not the same object as sqlalchemy.orm.session.Session

nimrodmar, 2020-09-23:
Each time I try to execute airsenal_run_prediction I get the following error. 
I've tried executing it with --num_thread=1 and it still didn't work.

Would appreciate any assistance. 
```
airsenal_run_prediction --weeks_ahead 3

INFO:numexpr.utils:Note: NumExpr detected 12 cores but "NUMEXPR_MAX_THREADS" not set, so enforcing safe limit of 8.
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
INFO:pystan:COMPILING THE C++ CODE FOR MODEL anon_model_8c4b3dae224ed40cd55af0e4da27a5f1 NOW.
Traceback (most recent call last):
  File "/Users/nimrodm/opt/anaconda3/bin/airsenal_run_prediction", line 8, in <module>
    sys.exit(main())
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/site-packages/airsenal/scripts/fill_predictedscore_table.py", line 142, in main
    tag = make_predictedscore_table(
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/site-packages/airsenal/scripts/fill_predictedscore_table.py", line 107, in make_predictedscore_table
    calc_all_predicted_points(gw_range, season, tag, session)
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/site-packages/airsenal/scripts/fill_predictedscore_table.py", line 89, in calc_all_predicted_points
    processor.start()
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/multiprocessing/context.py", line 283, in _Popen
    return Popen(process_obj)
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 32, in __init__
    super().__init__(process_obj)
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/multiprocessing/popen_spawn_posix.py", line 47, in _launch
    reduction.dump(process_obj, fp)
  File "/Users/nimrodm/opt/anaconda3/lib/python3.8/multiprocessing/reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
_pickle.PicklingError: Can't pickle <class 'sqlalchemy.orm.session.Session'>: it's not the same object as sqlalchemy.orm.session.Session

```

bilderbuchi, 2020-09-23:
Did you accidentally report this in the wrong repository? This seems to have nothing to do with pymeasure?

nimrodmar, 2020-09-23:
> Did you accidentally report this in the wrong repository? This seems to have nothing to do with pymeasure?

You are right! Im sorry, feel free to delete this...

closed 2020-09-23 by bilderbuchi