pymeasure/pymeasure #323: A string parameter to store free comments

Bruyant, 2020-09-29:
Hi,

I'm looking to define a string parameter to enteer a comment in the gui and store it in the csv file.

Is it a feature which alredy exist ? 

If not, i should be able to ad it as a new string parameter class. What do you think of this idea ?

Thanks for the great work
Regards,

physicsninja, 2020-10-01:
If you are ok with the string comment being stored in the header along with all the other input parameters then yes, totally.

The default `Parameter` class is what you are looking for. If you look at this [example](https://github.com/ralph-group/pymeasure/blob/master/examples/Basic/gui.py), the parameter called `seed` is basically what you would need to do. You would just define a new parameter as normal

`comment = Parameter('Comment', default = 'No Comment')`

and add it to the inputs. If you want to use the string you entered to do something in the measurement loop you could (like how `seed` sets the random seed in the  example) otherwise the content of that parameter will just show up in the header of the results file. Please let me know if you still have questions or if this doesn't work for you.

Bruyant, 2020-10-03:
Hi,

Thanks, it worked like a charm, I use it to track experiments parameters which are not computer controlled.

The only improvement I could dream of is to change the qt widget to allow longer comments to be entered, but I will try to do it by overriding the Parameter Class.

Nicolas BRUYANT
Ingénieur Instrumentation - Accueil des visiteurs

	
CNRS - LNCMI
143, avenue de Rangueil
31400 TOULOUSE
FRANCE

Tel : +33 (0) 6 22 84 24 73
Fax : +33 (0) 562 17 28 16 
Bureau 16 
Mail : nicolas.bruyant@lncmi.cnrs.fr
Lab wiki: http://pythia/pmff/

closed 2020-10-15 by Bruyant