pymeasure/pymeasure #328: VI_ERROR_TMO with Oxford Instruments Intelligent Temperature Controller 503

jklynch, 2020-10-08:
We are getting a VI_ERROR_TMO while trying to write the temperature setpoint of an Oxford Instruments 503 Intelligent Temperature Controller.

`pyvisa.ResourceManager()` identifies the device:

```
Resource: 'ASRL5::INSTR'
IDN:OXFORD INSTRUMENTS:MERCURY ITC:144850017:2.2.6.9 
```

Instantiating ITC503 seems to work (if clear_buffer=False is specified):

```python
from pymeasure.instrument.oxfordinstruments import ITC503
itc = ITC503("ASRL5::INSTR", clear_buffer=False) # times out if clear_buffer=True

itc.control_mode = "RU"
itc.heater_gas_mode = "AUTO"
itc.auto_pid = True

print(itc.temperature_setpoint)
```

At this point VI_ERROR_TMO (-1073807339) occurs. The same error occurs when instantiating ITC503 with `clear_buffer=True`.

Any advice would be appreciated.

physicsninja, 2020-10-08:
I have not interacted with this device before, but my first debugging thought is if you do something like:

```
rm = pyvisa.ResourceManager()
itc = rm.open('ASRL3::INSTR')
print(itc.query('R0\n'))
```
Do you get a timeout or what you expect? I am concerned that there is a serial termination problem given the buffer problems as well

bilderbuchi, 2020-10-13:
In addition to what @physicsninja suggests (although you have to terminate with `\r`, not `\n` according to the manual I found), do the other commands (control_mode, heater_gas_mode, auto_pid) work correctly/as expected? 
I.e. can you check if making these settings changes them on the device as expected, or can you even query them correctly?

Possibly the default serial interface settings are incorrect for this device, what happens if you include `stopbits=2` when instantiating the instrument?

The [command](https://github.com/ralph-group/pymeasure/blob/9705b081970254d01b54816a49945bb0db93da82/pymeasure/instruments/oxfordinstruments/itc503.py#L101) for temperature_setpoint seems correctly defined as far as I can tell. 

What happens when you leave out the previous three commands and read the temperature_setpoint right after instantiating?

bilderbuchi, 2020-10-13:
Also, I'd like to pull in @CasperSchippers, who currently has a PR (#308) open on updates for this instrument, maybe he has some wisdom to share.

CasperSchippers, 2020-10-16:
I have only interacted with this device using GPIB, so not using a serial connection. It sounds like a communication issue (indeed something like the termination or stop-bits). If you try connecting with pyvisa (as suggested by @physicsninja), I think you would also need to specify the termination-character (`itc.read_termination = '\r'`).

jklynch, 2020-10-16:
Thanks for these very helpful responses!

CasperSchippers, 2021-02-08:
@jklynch, did you succeed in getting the instrument to work correctly? If so, we can close this issue.

jklynch, 2021-02-08:
Yes, you can close the issue. Ultimately we used a different library but I feel good about using pymeasurwe. Thank you for your help.

closed 2021-02-08 by jklynch