pymeasure/pymeasure #331: Parse returned strings on Adapter level

bilderbuchi, 2020-10-14:
If a device returns strings that don't fit into the simple pattern that Adapter.values can deal with, one has to parse after values() by defining get_process for every property. If the reply structure is consistent for an Instrument, this could probably use some improvement. 

One alternative way to handle this would be to use a modified `extract_value` as the `cast` callable and make it return a float - but this only replaces the use of `get_process=` with `cast=`, so it's not simpler per se.

If one wanted to apply the same processing/parsing to all gets of an Instrument without specifying this for every control/measurement individually, I could see optionally implementing some generic method at the Adapter level. If that is defined for a given Instrument (e.g. via kwarg to the Instrument constructor that gets passed to the Adapter), it gets called within Adapter.values right after receiving the string. This can be used to prepare the reply for the usual separator/cast application, via regex or whatever you want to use, and you only have to define/use this once for the whole Instrument. Of course this depends on the instrument's replies all having the same structure.

_Originally posted by @bilderbuchi in https://github.com/ralph-group/pymeasure/pull/327#issuecomment-707998416_

dkriegner, 2020-10-14:
Although its of course possible that a specific device has a syntax which allows for the use of a general get_process function I think this would be a very academic case. In the case of #327 this get_process functions are only applied to certain properties. Namely those which return numerical values which requires different functions for every data type. 

One significant simplification would be possible if `get_process` would get `**kwargs` forwarded as its the case for the `values` method. In this case one could use a `cast` (or similar) argument also in `get_process`. Otherwise one can of course write somehow lengthy lambda expressions in get_process which use a general get_process and perform the type cast. But this is also not very readable.

bilderbuchi, 2020-10-14:
Hm, I was under the impression that your device always replies with something like `Name = somevalue`, and you wanted to avoid defining a `get_process` (or other kwarg) for _every_ control/measurement just to get rid of the `Name = ` part and get numbers back. 
If you handled that at Adapter level, you could avoid adding something to every control/measurement, and the default `values()` behaviour makes nice floats out of `somevalue`.

Or are you arguing for a possibility for more fine-grained control of the `values()` behaviour, for individual controls/measurements, beyond the `call` and `separator` kwargs that don't catch many reply structures? 
I also have a device where basically an appended unit in the reply string that I don't use makes it impossible that `values()` spits out numbers, but I handle this with individual lambdas for `get_process=`.

bilderbuchi, 2020-10-14:
Forwarding `**kwargs` to `get_process` would probably complicate the implementation of all existing `get_processes`, no? E.g. when you also have `cast` or `separator` present. 

I think it could be more feasible to make `Adapter.values` understand/consume additional kwargs (like a `preprocess_reply` Callable that it then executes if present). What do you think?

dkriegner, 2020-10-14:
Its not true that I want to do the same for every reply from the instrument. As I wrote https://github.com/ralph-group/pymeasure/pull/327#issuecomment-708618674 the communication with the instrument can be like this (> indicate sent strings, < shows the reply)

a query of a numerical value
\> 'getv 1\r\n'
< 'voltage = 33.000000 V\r\nOK\r\n'

a query of a string reply:
\> 'getser 1\r\n'
< 'SERIALNUMBER-XYZ\r\nOK\r\n'

My main concern was that I could not benefit from the already existing `cast` argument to `Adapter.values` because I need to `get_process` the reply from the device before it can be cast. **I think you are right that a `preprocess_reply` callable would fix this** issue. I guess this would need to go after this [line](https://github.com/ralph-group/pymeasure/blob/master/pymeasure/adapters/adapter.py#L71). But if the device in PR #327 is the first one needing this I guess it should be implemented locally in this instrument.

bilderbuchi, 2020-10-15:
> I think you are right that a preprocess_reply callable would fix this issue. I guess this would need to go after this line.

Indeed, I agree!

What I am not really sure about yet is the benefit - afaict everything that you want to do in `preprocess_value` (name TBD), you could also do in `get_process`, so essentially you just move the processing logic around, with the exception of being able to make use of the automatic `values` casting when you preprocess to isolate a float. 
So, in your control/measurement definitions, instead of, e.g.,  
`(...,get_process=lambda i: float(i.split('A')[0])`, you would have  
`(...,preprocess_value=lambda i: i.split('A')[0])`, and instead of,  
`(...,get_process=lambda i: int(i.split('A')[0])`, you would have  
`(...,preprocess_value=lambda i: i.split('A')[0], cast=int)`.

Maybe the benefit lies in more clarity concerning the message processing steps/order? 
Or it's in the fact that we _could_ add Adapter-level preprocessing (with possibility to override in control/measurement defs) in case instrument replies are mostly of the same structure (to not repeat nearly-identical code). I am not sure how often that occurs.

dkriegner, 2020-10-15:
I agree with what you wrote. For my particular case the current logic of the steps makes no sense (or the preprocess step would be needed additionally). If this is a singular case you should keep all your code in place unchanged and as you write my case can be handled by slightly more complicated `get_process`. The more general question would be to assess how many other uses of get_process in your code base suffer from similar issues. Since values is an adapter level function I can in principle add a configurable preprocess function myself for the adapter in #327. I do know already that for the second device I am implementing I will need that again.

bilderbuchi, 2020-10-15:
> to assess how many other uses of get_process in your code base suffer from similar issues.

Yeah that would be useful.

I think if we implement this, for maximum flexibility I lean towards having 
* an optional callable `preprocess_value` callable that, if available, runs [near the beginning](https://github.com/ralph-group/pymeasure/blob/f14fdbf51065c128770e9e0e1f726500cd082cfb/pymeasure/adapters/adapter.py#L71) of `Adapter.values()`,
* The callable can be specified at the Adapter level (e.g. via kwarg for the Adapter), and then runs for all invocations of `values()`, i.e. via the property getter. Best practice will probably be to use a regex (maybe compiled for speed) to extract the desired values, and if there is no match then to pass through the string unchanged. This way one can do e.g. unit stripping, but leave string-only replies alone.
* A callable can also be specified in the call of Instrument.control/measurement via kwarg, and will then replace an existing Adapter-level definition.

bilderbuchi, 2020-10-15:
@CasperSchippers  @cjermain thoughts on this proposal?

dkriegner, 2020-10-15:
I did a quick search on the use of [`get_process`](https://github.com/ralph-group/pymeasure/search?p=1&q=get_process) in your repository. 

From all occurrences I manually counted the following (I did not count equal uses in the same instrument repeatedly):

- pure numerical processing/array operations 2 (1 of which is in your unit tests)
- pure string processing 6
- casting and string processing 9 (1 of which is in your unit tests)
- only casting 1
- string, cast, and numerical processing 1

Even if you would count differently I think its clear the the majority of use cases includes string processing which could likely be done in a `preprocess` callable in `Adapter.values()`. Not sure if any existing use-case would benefit from an adapter level function (equally applied to all return values), but in PR #327 this would indeed be conceivable.

bilderbuchi, 2020-10-16:
Thanks for this, that is quite useful!

CasperSchippers, 2020-10-16:
I think it would be a useful addition; at the moment i doubt we can estimate how many cases will benefit from it, but as long as it is optional to use and doesn't break/change the way of working if you don't use it, if even only a few cases benefit from it, it will be worth it.

dkriegner, 2020-10-21:
Since I felt there is some tendency to accept a contribution addressing this I started to implement following these [guidelines](https://github.com/ralph-group/pymeasure/issues/331#issuecomment-709015956). 

In principle this can be done with minimal changes in [`Adapter` ](https://github.com/ralph-group/pymeasure/blob/e4c603981ea96a937140124f5db4789f9b4d3286/pymeasure/adapters/adapter.py#L29). However, the problem is how to safely forward the `preprocess_reply` argument in the Adapter constructor. Several Adapters already use the `kwargs` argument in their `__init__` as means to forward settings to underlying libraries (eg. [VisaAdapter](https://github.com/ralph-group/pymeasure/blob/e4c603981ea96a937140124f5db4789f9b4d3286/pymeasure/adapters/visa.py#L46) or [SerialAdapter](https://github.com/ralph-group/pymeasure/blob/e4c603981ea96a937140124f5db4789f9b4d3286/pymeasure/adapters/serial.py#L41)). So one would need to implement some logic to pass through the `preprocess_value` argument or just ignore all other keyword arguments in `Adapter.__init__`. In this case the changes to the existing adapters should be minimal. It would only be important to filter out `preprocess_value` from the kwargs for further use in derived Adapters.

See [here for a possible patch](https://gist.github.com/dkriegner/21c9864cba6536db9bcb5bc7be2ae672)

If you do not consider implementing this premature and agree that the patch looks reasonable I would start a pull request and check the other Adapters.

bilderbuchi, 2020-10-22:
This looks like a good beginning to me!
I think you could add `preprocess_reply` (I like that name better than `preprocess_value`!) to the Adapter init, maybe like this: 
`__init__(self, preprocess_reply=None, **kwargs)`, then you keep `**kwargs` "clean" for forwarding to sub-adapters.
Furthermore, the `results = self.preprocess_value(results)` clause needs to be in an `elif` checking if self.preprocess_value is not None.

Later one, you should be able to use `FakeAdapter` to implement some tests checking the behaviour. Also, some documentation of the feature would be great, although I am not yet sure where it fits best, probably with `Instrument`.

dkriegner, 2020-10-22:
I agree that `__init__(self, preprocess_reply=None, **kwargs)` would be nicer since the property shows up in the call signature.
The problem is that I do not need the `kwargs` clean in `Adapter.__init__()` but in the derived classes like `VisaAdapter.__init__()`. To make this happen with your code suggestion I think `Adapter.__init__()` would somehow need to return the clean kwargs to the calling constructor... Or am I misunderstanding the python behavior?

The other alternative is to rely on the [safeKeywords](https://github.com/ralph-group/pymeasure/blob/d37bc5b6ca33660be9334fe73e56b640b62c3363/pymeasure/adapters/visa.py#L58)-cleaning in the derived classes, but as issue #334 shows this has some limitations and likely is also not implemented in all Adapters.

Thats why I decided for the [`kwargs.pop` use in line 14 of the mentioned patch](https://gist.github.com/dkriegner/21c9864cba6536db9bcb5bc7be2ae672#file-issue331-patch-L14)

bilderbuchi, 2020-10-22:
You're right, you could also leave out **kwargs from Adapter.__init__, and then call `super().__init__(preprocess_reply=preprocess_reply)` in the derived inits (which then of course also need the `preprocess_reply=None` parameter)

dkriegner, 2020-10-22:
What you propose would work, but I would find it awkward that every derived adapter needs to explicitly mention all kwargs of the base-class. I will soon initiate a pull request and we can continue line by line discussion better there.

bilderbuchi, 2020-10-22:
Well, the derived adapter is derived/inherited, so it better have all the parameters of the super class, or am I misunderstanding?
Yeah, sure, I agree. maybe wait until #327 is merged, though.

dkriegner, 2020-10-22:
I was not very clear in my statement. What I meant is that with your proposal (now implemented in the [gist](https://gist.github.com/dkriegner/21c9864cba6536db9bcb5bc7be2ae672)) one duplicates the default value of preprocess_reply into every call signature of derived Adapter classes. I would consider that harder to maintain, but this solution for sure has the advantage that the feature is more visible (which maybe is most important for a project like yours).

bilderbuchi, 2020-10-22:
Yeah, maybe that's preference, but I'm personally not a fan of kwarg-parsing-heavy APIs, e.g. like matplotlib favours - it becomes hard to find out about the APIs options if most parameters are stuffed into **kwargs and are not visible in the signature. OTOH, matplotlib often has a huge number of options, but that's not a problem we have here.

bilderbuchi, 2020-10-26:
Closed by #345

closed 2020-10-26 by bilderbuchi