pymeasure/pymeasure #332: Homebuilt instruments.

Sacroth, 2020-10-15:
Hi, this may seem like a simple / novice question. 

We have a lot of home built serial equipment and therefore adding it to your instrument list is not worth while. 

My question is, what is the most efficient way to integrate a home built instrument with pymeasure locally so it works seamlessly with the managed window. 

is it as simple as
instrument =  = SerialAdapter("COM3") 

and then using instrument.write() etc? 

or is there a more seamless way of doing this? 

Cheers

bilderbuchi, 2020-10-15:
You can implement a [pymeasure Instrument](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#instrument-file) which implements all functionality you need, and keep the definition in your code. Then, when [implementing your procedure](https://pymeasure.readthedocs.io/en/latest/tutorial/procedure.html), instead of `from pymeasure.instruments.something import SomeInstrument` you import and use your homebuilt Instrument class.

Does that answer the question?

Sacroth, 2020-10-15:
Yes this helps a lot. I initially just tried to use the PySerial adapter and connect via its com port. This worked to write to the instrument` as intended, but when using read it does not. 
When using instrument.read(8) (I expect only 8 bytes) I get the error read() expects 1 argument not 2. This was how I set up the device (not using the instrument library you mentioned) 
....
From pymeasure import PySerial 
instrument = PySerial(“address”) 

instrument.write(“L/r”) 
instrument.read(8) 
...
But 
instrument.write(“C900\r”) will move the instrument to the 900 position on its stepper motor. 

Does adding it to the library allow you to set timeouts etc? I presume that is why the read is not working. Using pyserial directly and setting a timeout = 0.05 the issue is resolved. Is this possible in pymeasure? Seems it’s using pyserial under the hood. 

Thanks again @bilderbuchi for your time.

bilderbuchi, 2020-10-15:
What it enables you to do is something like (untested code!)
```
class SacrothMotor(pymeasure.instruments.Instrument):

    def __init__(self, resourceName, **kwargs):
        super().__init__(
            resourceName,
            "SacrothMotor",
            read_termination='\r',
            write_termination='\r',
            # timeout should also work
            **kwargs
        )
    position = pymeasure.instruments.Instrument.control(
        ":C?", ":C%i",
        """Control the stepper motor position""",
    )
```
and then use it like
```
motor = SacrothMotor('COM9')
print(motor.position)
motor.position = 900
print(motor.position)
```
Docs are at https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#writing-properties

bilderbuchi, 2021-02-16:
This issue has more of a discussion/usage question character, so I will move it to our new discussion forum, which is a better fit than the issue tracker.

closed 2021-02-16 by bilderbuchi