pymeasure/pymeasure #333: Managed plot slow down.

Sacroth, 2020-10-15:
Hi, sorry for two questions in one day. 

When using the managed plot (something required to implement this in our lab instead of labview) 

I have noticed a sudden slowdown in the plotting when the data sets get above ~1000 points. Is this due to the cursor being active? I have noticed similar behavior using plotly with large data sets. 

If there is a way to have the plot be a bare-bones x,y plot (turning a lot of these graphics processes off) I would love to know, looking at the pyqtgraph documentation did not seem to help with how it is implemented in pymeasure. 

I am currently running win10, python 3.7.
using this code.     
```
 
import sys
sys.modules['cloudpickle'] = None

import logging
log = logging.getLogger(__name__)
log.addHandler(logging.NullHandler())

import sys
import tempfile
import random
from time import sleep
from pymeasure.log import console_log
from pymeasure.display.Qt import QtGui
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import Procedure, Results
from pymeasure.experiment import IntegerParameter, FloatParameter, Parameter

class RandomProcedure(Procedure):

    start = IntegerParameter('start')
    stop = IntegerParameter('stop')
    step = IntegerParameter('step')
    delay = FloatParameter('Delay Time', units='s', default=5)
    seed = Parameter('Random Seed', default='12345')

    DATA_COLUMNS = ['Wavelength', 'Intensity']

    def startup(self):
        log.info("Setting the seed of the random number generator")
        random.seed(self.seed)

    def execute(self):
        log.info("Starting the loop of %d iterations" % self.start)
        for i in range(self.start,self.stop,self.step):
            data = {
                'Wavelength': i,
                'Intensity': random.random()
            }
            self.emit('results', data)
            log.debug("Emitting results: %s" % data)
            sleep(self.delay)
            if self.should_stop():
                log.warning("Caught the stop flag in the procedure")
                break

class MainWindow(ManagedWindow):

    def __init__(self):
        super(MainWindow, self).__init__(
            procedure_class=RandomProcedure,
            inputs=['start','stop','step', 'delay', 'seed'],
            displays=['start','stop','step', 'delay', 'seed'],
            x_axis='Wavelength',
            y_axis='Intensity'
        )
        self.setWindowTitle('GUI Example')

    def queue(self):
        filename = tempfile.mktemp()

        procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)

if __name__ == "__main__":
    app = QtGui.QApplication(sys.argv)
    
    window = MainWindow()
    window.show()
    sys.exit(app.exec_())    
```

When the range is set such as 0:1000:1 the plotting becomes quite slow. 

It is also worth noting my progress bars in queue dont update. 

Cheers.

jmittelstaedt, 2020-10-16:
I can't speak to the plotting slowdown too much, but the reason the progress bars aren't updating is that you need to tell pymeasure what the progress of your `Procedure` is in the `execute` method. You do that by including a line somewhere in the main loop like `self.emit("progress", i/max_iterations)`, I usually cast `i/max_iterations` to an `int` but I'm not sure if that's actually necessary.

bilderbuchi, 2021-02-16:
This issue has more of a discussion/usage question character, so I will move it to our new discussion forum, which is a better fit than the issue tracker.

closed 2021-02-16 by bilderbuchi