pymeasure/pymeasure #334: Allow passing serial parameters to PyVISA through PyVisaAdapter?

alterscape, 2020-10-16:
I'm trying to control an instrument through PyMeasure which uses a rather unorthodox serial setup: 57600 baud, 7 data bits, 1 stop bit, odd parity. All these settings are kwargs for PyVisa's `get_instrument` / `open_resource` but are not permitted by safeKeywords in visa.py.  I ended up hacking my local copy of the library to validate that adding them to safeKeywords works.

Is there any compelling reason not to allow these keyword args, since they make it possible to configure more instruments?

bilderbuchi, 2020-10-16:
Hm, you raise an interesting point. I'd have to dig for why the safe keywords are around. Right now I can't see why we should not extend the list, but [Chesterton's fence](https://en.wikipedia.org/wiki/Wikipedia%3AChesterton%27s_fence) and all that...

In the meantime, to define fixed serial params for your instrument without modifying pymeasure, the following pattern can be useful in your specific Instrument's init-- you create a `SerialAdapter` and pass _that_ to the `Instrument` init:
```
    def __init__(self, adapter, name='My unorthodox instrument',
                 baudrate=57600, bytesize=7, parity='O', stopbits=1, **kwargs):
        """Docstring
        """
        adapter = SerialAdapter(adapter, **kwargs)
        super().__init__(adapter, name, **kwargs)
```

dkriegner, 2020-10-21:
I am preparing a [pull request](https://github.com/dkriegner/pymeasure/tree/ibeam) (which waits for #327 to be finished) which is also suffering from this issue. 
My feeling was that the safe keywords are indeed only keywords which **every** pyvisa resource supports. this of course comes with some limitations.

My workaround was to set the communication parameters after opening the resource. see [here](https://github.com/dkriegner/pymeasure/blob/13bd9b2cf217294ea36c8d287c2ee083b8e626b7/pymeasure/instruments/toptica/ibeamsmart.py#L123)

bilderbuchi, 2020-10-27:
Also, [`baud_rate`](https://github.com/ralph-group/pymeasure/pull/352/files#diff-fc67f8adf8c97a2f75456448361816bffbb5c18d82cf31f709c6620638426ccbR47) is not getting through.

bilderbuchi, 2021-02-16:
> My feeling was that the safe keywords are indeed only keywords which every pyvisa resource supports

Indeed, every [MessageBasedResource](https://pyvisa.readthedocs.io/en/latest/api/resources.html#pyvisa.resources.MessageBasedResource), but yes. If we don't pop invalid ones off, PyVISA will raise a ValueError when [opening the resource](https://pyvisa.readthedocs.io/en/latest/_modules/pyvisa/highlevel.html#ResourceManager.open_resource), but the set of valid kwargs may change depending on the resource_name (or type of connection used). 
I suspect (based on my usage) that a large number of connections are probably to SerialInstruments, but we want to retain the generic nature of the VISAAdapter as much as possible. I don't currently know how to best implement this, beside the pattern I suggested above, let me think about it.

bilderbuchi, 2021-03-04:
Another case reported by @msmttchr (#412):
```
>>> from pymeasure.instruments.hp import hp34401A
>>> i=hp34401A.HP34401A('ASRL5::INSTR', baud_rate=9600, data_bits=7, parity=2)
>>> i.adapter.connection.data_bits
8
>>> i.adapter.connection.parity
<Parity.none: 0>
>>> i.id
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\michele sardo\AppData\Local\Programs\Python\Python39\lib\site-packages\pymeasure\instruments\instrument.py", line 81, in id
    return self.adapter.ask("*IDN?").strip()
  File "C:\Users\michele sardo\AppData\Local\Programs\Python\Python39\lib\site-packages\pymeasure\adapters\visa.py", line 112, in ask
    return self.connection.query(command)
  File "C:\Users\michele sardo\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\michele sardo\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc5 in position 1: ordinal not in range(128)
>>> i.adapter.connection.data_bits=7
>>> i.adapter.connection.parity=2
>>> i.id
'HEWLETT-PACKARD,34401A,0,11-5-3'
```

The instrument is expecting `baud_rate=9600`, `data_bits=7` and `parity=2` (even)., but as You can see from the log above, the parameters are not passed to pyvisa `open_resource` call, because they are filter by `safeKeywords `list. Indeed also `baud_rate` is not passed, but the default happens to be 9600.

bilderbuchi, 2021-03-04:
One potential approach would be to 
* enlarge `safeKeywords` to encompass **all** valid pyVISA kwargs
* wrap `self.manager.open_resource` in a try-catch
* catch any occurring ValueErrors
* parse the affected arg name out of the exception, pop it off of `kwargs`
* And try `self.manager.open_resource` again, until it works
I think this could be viable because in pyVISA, the presumably slow `open` call happens *after* the kwarg check, but that needs to be profiled/tested first - we don't want to incur 1s delay for every kwarg that needs to be popped off.

msmttchr, 2021-03-04:
@bilderbuchi I am not sure I understand the benefit of this filtering service done in the VISAAdapter, since:
- Users can pass keywords which are silently ignored, but what is then the purpose of having a generic parameters as **kwargs instead of a specific list of, possibly initialized, parameters ?
- The ignored parameters are not used any where else.
- There are other API, e.g. `ask_values(self, command, **kwargs)` which does not offer such filtering service.
- As I user, I prefer to have an exception raised, if the parameters are not valid instead of discarding them.

Could You please shed some light on what is the purpose of it ?

bilderbuchi, 2021-03-08:
As I understand it (and mind you, this was present way before I joined the project), `safeKeywords` is here to preserve the generic nature of our instruments - you give it an appropriate port/adapter designation and don't have to fiddle with settings beyond that (in a well-implemented instrument). 
For the VISAAdapter the case is a little more involved because 
a) it handles a number of different protocols (with a variable set of appropriate/admissible keywords)
b) you don't know what the appropriate set is before you call `open_resource`, because that is determined within PyVISA.

So, for example, say you have a power supply with a number of different interfaces: serial over RS-232, USB, TCPI/IP (e.g. [one of these](https://www.aimtti.com/product-category/dc-power-supplies/aim-qlseries)).
Then, for the serial connection, you may have to specify e.g. `baud_rate` and `parity`. However, if you connect over TCP/IP, these settings are irrelevant and the kwargs invalid. If you plug it in via USB, you may have a different set, still. Which interface is used is determined by the VISA resource name you pass to your instrument on creation.

So, if you want your users to be able to supply a resource designator or adapter, and have the rest configured as per the device manual, you can define all the appropriate kwargs, and filter them appropriately -- historically via `safeKeywords`, and for the more flexible VISAAdapter via the proposed approach above. You can also of course do the selection of the correct `Adapter` subclass in the instrument init, based on the passed resource string. 
Ideally, you define the kwarg values as defaults in the signature of your Instrument, so that the user is free to override in case the value is not hard-coded within the instrument.

dphaas, 2021-04-22:
I just got caught by this "safeKeywords" feature on a serial power supply that is set to 115200 baud.  I was trying to baud_rate=115200 and of course it was filtered out.  I'd vote for at least allow baud_rate.

It seems like a fundamental feature someone was trying to provide in "resources.py" list_resources().  One can't send a *IDN? to s serial instrument if the host and instrument port settings don't match.  I'd suggest making list_resources() a bit more robust with a hint about serial port setting for ASRL resources.  Something like:
```
finally:
                if 'ASRL' in res.resource_name:
                    print(n, ":", instr, ":", idn,
                        f', HOST PORT: baud={res.baud_rate}',
                        f'bits={res.data_bits}, parity={res.parity}, '
                        f'stop={res.stop_bits}, flow={res.flow_control}')
                else:
                    print(n, ":", instr, ":", idn)
                res.close()
```

samcondon4, 2021-11-30:
> One potential approach would be to
> 
> * enlarge VISAAdapter's `safeKeywords` to encompass **all** valid pyVISA resource kwargs
> * wrap `self.manager.open_resource` in a try-catch
> * catch occurring ValueErrors (see https://pyvisa.readthedocs.io/en/latest/_modules/pyvisa/highlevel.html#ResourceManager.open_resource, "...is not a valid attribute for type..")
> * parse the affected arg name out of the exception, pop it off of our `kwargs`
> * And try `self.manager.open_resource` again, until it works

@bilderbuchi I think the suggested approach here could potentially be problematic for things like catching typos. If a developer is creating an Instrument class and needs to specify a set of kwargs for `open_resource()` and mistypes one of the kwarg names, it would just be silently popped off and the Instrument would be instantiated anyway. This could make debugging the simple typo mistake much more difficult as the normally raised `ValueError` would pass silently. I tend to think that a `kwargs` argument should remain generic, or an exception should be raised when an invalid parameter is passed rather than silently popping. In the latter approach, the `safeKeywords` list would need to be expanded to encompass all of the possible pyvisa resource kwargs which seems like quite a pain given the wide variety of protocols/kwargs implemented in a visa resource. I think this is also in line with @msmttchr 's thinking from the above discussion.  What is wrong with scrapping `safeKeywords` and implementing a simple try catch block here that takes a raised open_resource() ValueError and changes the error message to something more useful for Instrument developers?

bilderbuchi, 2021-11-30:
The typo thing is a good point.

> What is wrong with scrapping safeKeywords and implementing a simple try catch block here that takes a raised open_resource() ValueError and changes the error message to something more useful for Instrument developers?

AIUI, [this point](https://github.com/pymeasure/pymeasure/issues/334#issuecomment-792627705), i.e. the instrument implementation defines a `baud_rate`, but the user wants to simply connect over TCP/IP, which PyVISA selects based on the resource/adapter string. They would get an error message concerning an invalid kwarg `baud_rate`, which they would not get if they connected over serial, where they would *need* the `baud_rate` argument that the instrument developer helpfully already set for them.

It's a hairy problem caused by the generic/multiadapter/multitransport nature of pymeasure adapters (which otherwise is A Good Thing). I haven't checked recently, maybe PyVISA has changed in the meantime concerning its kwarg handling for different protocols?

bilderbuchi, 2021-11-30:
Unfortunately, I don't have multi-transport (or any device) at home to play with and find a good way to deal with this. :-/

msmttchr, 2021-12-02:
What about considering the possibility that the advantages of this `SafeKeyword` mechanism (if any) are way lower compared to the user's inconveniences ? Based on python consenting adult principle, if I pass a keyword argument we need to assume that I know what I am doing. If the parameter I passed is wrong, I am happy to get an error. If I want to use only the resource name and it works I am happy with it. If I need to manage multiple interface on the same instrument, I know what interface I am specifying and it's my choice/responsibility to pass keywords arguments overriding possible default values defined by  instrument's developer.
Some instrument's have programmable baud rate and currently if instrument developer needs to define a default value need to add a sequence like this:
```
        if isinstance(self.adapter, VISAAdapter):
            self.adapter.connection.baud_rate = 2400
```
In this case, if the user changed the default baud rate in the instrument, then it needs to some code like this:
```
i = FictionalInstrument("COM1")
# Change default baud rate
i.adapter.connection.baud_rate = 9600
```
i.e. I am forced to use internal implementation structures and it's way less elegant than:
```
i = FictionalInstrument("COM1", baud_rate = 9600)
```
Simple and user friendly. What do you think?

LongnoseRob, 2021-12-02:
@msmttchr I am with you here.
I think with good documentation on the specifics of the instruments using the simple 
`i = FictionalInstrument("COM1", baud_rate = 9600)` should be less an issue.

dphaas, 2021-12-02:
@msmttchr Well said and I agree.  I'd like to make an instrument contribution for Instek programmable power supply but this serial baud_rate isssue is a bit of a stumbling block for me.  Instek provides both a serial and USB interface but the USB interface has a custom driver which is not signed.  This forces the use of the serial interface for security purposes.  The instrument supports baud rate up to 115.2k.  9600 seems like teletype speed.

bilderbuchi, 2021-12-02:
Largely agree I think, Michele. I'm looking for default behaviour that is convenient for users while preserving the possibility of using multiple connections. 

Maybe I'm confused, but considering your example

> Some instrument's have programmable baud rate and currently if instrument developer needs to define a default value need to add a sequence like this:

>        if isinstance(self.adapter, VISAAdapter):
>            self.adapter.connection.baud_rate = 2400

> In this case, if the user changed the default baud rate in the instrument, then it needs to some code like this:

>     i = FictionalInstrument("COM1")
>     # Change default baud rate
>     i.adapter.connection.baud_rate = 9600

> i.e. I am forced to use internal implementation structures and it's way less elegant than:

>     i = FictionalInstrument("COM1", baud_rate = 9600)

Isn't that exactly the problem we have to solve here? Giving kwargs with defaults that are easily overridden by the user is the most desirable pattern, I agree. 
However, how do you pass that on to the super() call (`super().__init__(adapter, <what here>)` ) so that the VISAAdapter does not choke e.g. on a default-set baud rate when the user passes a TCPI/IP or USB resource string (where Pyvisa-internally `baud_rate` is invalid), and still sets the correct `baud_rate` for serial connections? 
```
i = FictionalInstrument("TCPIP::192.168.0.2::INSTR")
```
The user did not pass any kwargs, so the "consent" does not come into play?

bilderbuchi, 2021-12-02:
I want to avoid that instrument contributors have to handle the connection details in the `__init__` of their specific instruments, this does not scale well. We need to pass all collected kwargs on to `super().__init__` and sort it out then.

bilderbuchi, 2021-12-02:
_Let's find a concise definition of the problem and what we want to achieve. This is how I see it currently, feedback is appreciated! We can tune this and then determine an implementation that supports it._

How do we enable *all* these use cases?

## User stories
Let's say we have an instrument, requiring connection settings that are not the defaults for the respective connection, and featuring multiple possibilities/transports to connect. 
A general assumption (not repeated) for these stories is that the user does not want to encounter error messages if they haven't made a user mistake.

1. As a user, I want to simply connect to an instrument using default settings
```
i1 = FictionalInstrument("COM1") # correctly sets a default baud_rate of 2400
```
2. As a user, I want to easily override default settings to fit my needs:
```
i2 = FictionalInstrument("COM1", baud_rate=115200) # lightning-fast speed for @dphaas
```
3. As a user, I want to be free to choose which connection to use (e.g. serial-over RS-232, USB, TCP/IP) should an instrument support more than one
```
i3 = FictionalInstrument("TCPIP::192.168.0.2::INSTR") 
# here, baud_rate is invalid for pyvisa, but is a default kwarg anyway, no consent given or possible.
```
4. As a user, I want to be able to supply a self-generated `Adapter` instance (e.g. to enable serial connection sharing over RS-485/-422)
```
ser = SerialAdapter("COM1", baudrate = 9600) # note: not baud_rate, as this goes to pyserial
i4 = FictionalInstrument(ser)
```
5. As an instrument contributor, I want to define default connection settings for my instrument with a minimum of boiler-plate. These settings should be easily user-overridable if appropriate without having to change the code/implementation of my instrument.
---
This is what I I hope this makes it clearer w

I think we need some handling in `Instrument.__init__` or further on. So far, this was (badly) handled by `safeKeywords` in `VISAAdapter`. I'm fine with replacing it, it does not work correctly! I'm eager to hear the counterproposal that make these use cases possible.

bilderbuchi, 2021-12-02:
Note that if we pass a string as `adapter` (I suspect the majority of use cases), we automatically use a `VISAAdapter`. Therefore, we can optimize mainly for that use case - if the user passes a self-generated `Adapter`, any kwargs except name, includeSCPI  are being disregarded.

msmttchr, 2021-12-04:
@bilderbuchi 
1, 2, 4, 6: remove safeKeyword and you're done
3: User knows which interface is using: wrong parameter? error thrown (no babysitting)
5: parameter passed by user ? use it else provide implementor default

bilderbuchi, 2021-12-05:
@msmttchr thank you for the PR starting this!

> 3: User knows which interface is using: wrong parameter? error thrown (no babysitting)

I have to admit, I don't understand -- a user that uses an instrument with default settings like this `FictionalInstrument("TCPIP::192.168.0.2::INSTR")` should expect that errors unrelated to that (unexpected argument `baud_rate`) are raised? This is caused by a `baud_rate` default set in the instrument definition, not in any user-created code.
I don't think it's "babysitting" to avoid users having to deal with error they cannot do anything against.

I have now put the user stories into a set of tests (you were faster with the PR, though) -- see #537.
Before your change, 4/6 of the tests failed (keyword not in safeKeywords not passing through, incorrect kwargs don't raise). 
With your change, another set of 4/6 tests are failing (inappropriate keywords being passed on to pyvisa resource classes).

bilderbuchi, 2021-12-05:
I found it important to demonstrate the possibility of kwargs which are invalid for a serial connection, hence the set gpib attribute and test for that. We could also add some for USB connections (which are maybe more frequent nowadays), but the principle is demonstrated as it is.

bilderbuchi, 2021-12-05:
I think I found a way to implement a flexible behaviour:
* From the entries of `self.manager._resource_classes` collect a set of all possible attributes.
* Compare our kwargs against that set - 
-> this catches any typos like `baudrate` (and we raise a ValueError)

Then, we can 
* try-catch self.manager.open_resource() with all kwargs, 
* parse any returned ValueErrors for the name of the offending arg, and pop that off kwargs
* and retry.
-> This ensures that any kwargs not valid for the class defined by the resource_name are removed
(Time how much overhead that creates, I suspect not much)

Alternatively, we could determine the appropriate class before (like pyisa does in highlevel.py) and remove the kwargs before calling open_resource(). I suspect this will be not much faster, because the kwarg checking in pyvisa happens before it establishes a connection, so should be fast.

msmttchr, 2021-12-05:
> @msmttchr thank you for the PR starting this!
>
>
> I have to admit, I don't understand -- a user that uses an instrument with default settings like this [...]
No, I didn't mean this. I meant:
`FictionalInstrument("TCPIP::192.168.0.2::INSTR", baud_rate=9600)` should throw an error.
Sorry for the confusion.
On the other hand, the instrument implementation may provide appropriate defaults, but it should deal with different interfaces properly.
I think, we need to define a way for instrument implementors to express this in a concise and clear way.
In any case, this is a marginal case up to know.

Possibly the misunderstanding comes from the term user which I normally refer to pymeasure users while, I think, you refer to pymeasure contributors.

bilderbuchi, 2021-12-05:
> No, I didn't mean this. I meant

Ha, now I get it :relieved: . Indeed, I agree, this should not be a usecase to keep in mind to support/enable.

> On the other hand, the instrument implementation may provide appropriate defaults, but it should deal with different interfaces properly.
I think, we need to define a way for instrument implementors to express this in a concise and clear way.

I agree. Part of the motivation of writing the tests was to define a good design pattern, to then add it to the docs, too.

For me: 
Users: people using pymeasure-provided instruments and gui, without changes to pymeasure code
Contributors: great people writing and upstreaming instrument implementations, fixes, enhancements. Are also users, of course.
:-)

bilderbuchi, 2021-12-11:
Update: safeKeywords has been removed in #536, thank you @msmttchr! 
There is a follow-up PR #539 in progress that implements an (I believe) quite flexible solution to the various needs for communication configuration.

bilderbuchi, 2021-12-22:
We now have working and more flexible solution in place to deal with connection configuration when using the VISAAdapter (i.e. the majority of use cases). Already existing instrument drivers have been updated where appropriate. 

Please see https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#defining-default-connection-settings for instructions on how to implement an instrument driver with custom connection settings, and https://pymeasure.readthedocs.io/en/latest/tutorial/connecting.html#modifying-connection-settings for how to override those settings as a user.

closed 2021-12-22 by bilderbuchi