pymeasure/pymeasure #353: ManagedWindow not working on Win10 with Spyder installed

david-hopper, 2020-10-27:
Hi there,

I've been checking out your project as a potential framework to incorporate in my lab and was running some tests and came across an error.

If I create a new conda environment and install only the required dependencies to run pymeasure, the examples all work fine.

Now, if I install the IDE Spyder in that same environment, all of the gui examples that rely on `ManagedWindow` (gui.py, image_gui.py, etc.) throw the same error when a measurement is queued. Removing Spyder from the environment then allows the examples to run error free. The example `script_plotter.py` runs without error though.

Error:
```
(pymeasure) C:\pymeasure\examples\Basic>python gui.py
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\dhop\Anaconda3\envs\pymeasure\lib\site-packages\zmq\sugar\socket.py", line 451, in send_multipart
    memoryview(msg)
TypeError: memoryview: a bytes-like object is required, not 'int'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\dhop\Anaconda3\envs\pymeasure\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\dhop\Anaconda3\envs\pymeasure\lib\site-packages\pymeasure\experiment\workers.py", line 161, in run
    self.update_status(Procedure.RUNNING)
  File "C:\Users\dhop\Anaconda3\envs\pymeasure\lib\site-packages\pymeasure\experiment\workers.py", line 115, in update_status
    self.emit('status', status)
  File "C:\Users\dhop\Anaconda3\envs\pymeasure\lib\site-packages\pymeasure\experiment\workers.py", line 95, in emit
    self.publisher.send_serialized((topic, record), serialize=cloudpickle.dumps)
  File "C:\Users\dhop\Anaconda3\envs\pymeasure\lib\site-packages\zmq\sugar\socket.py", line 536, in send_serialized
    return self.send_multipart(frames, flags=flags, copy=copy, **kwargs)
  File "C:\Users\dhop\Anaconda3\envs\pymeasure\lib\site-packages\zmq\sugar\socket.py", line 456, in send_multipart
    raise TypeError(
TypeError: Frame 0 (128) does not support the buffer interface.
```
Looking through the required packages for Spyder, it looks like it requires `pyzmq` and since the error seems to originate in a similar package `zmq` I'm wondering if the error is related to that?

My system settings are
- Win10
- Conda 4.8.4 (all packages installed through conda)
- Python 3.8.6
- pymeasure-0.8
- spyder-4.1.5

bilderbuchi, 2020-10-27:
Hi! Can you check if it is the presence of cloudpickle that triggers the bug? Maybe Spyder drags that in? If so, I think this is a duplicate of #168 which we hope to solve for the next release.

david-hopper, 2020-10-27:
It looks like when cloudpickle and pyzmq are both installed this error is thrown. Spyder does indeed bring in both, and I could recreate this by manually installing both packages into the environment. If just one or the other is installed the code runs error free.

bilderbuchi, 2020-10-27:
OK, that's interesting, thank you. I'll close this in favour of  #168, let's track this over there.

closed 2020-10-27 by bilderbuchi