pymeasure/pymeasure #356: Keithley2400 and Keithley2450 classes: incompatible functions

pyMatJ, 2020-11-05:
Hello,

Recently added functions in the Keithley2400 class ( [https://github.com/ralph-group/pymeasure/pull/326](https://github.com/ralph-group/pymeasure/pull/326) ) are actually written in the command syntax that match the 2450 series. The syntax for the 2400 class is very close. (code snippets at the end).

What is the preferred method to fix this, i.e. correct the syntax in the keithley2400 driver and add them in the keithley2450 one ?
I can. Should I prepare a PR with the modifications/corrections, or is there another preferred way ?

Thank you in advance,

Mathieu

----------------------------- code snippets ------------------
The functions in question are all from the `Filters` section. Basically, they include a ` :FUNC: ` (`:CURR:` or `:VOLT:` ) which acutally belongs to the 2450 series.
Current implementation in 2400 (which instead should go in 2450) example:
```python
 current_filter_count = Instrument.control(
        ":SENS:CURR:AVER:COUNT?", ":SENS:CURR:AVER:COUNT %d",
        """ A integer property that controls the number of readings that are 
        acquired and stored in the filter buffer for the averaging""",
        validator=truncated_range,
        values=[1, 2500],
        cast=int)
```
Correct implementation in 2400 example:
``` python
current_filter_count = Instrument.control(
        ":SENS:AVER:COUNT?", ":SENS:AVER:COUNT %d",
        """ A integer property that controls the number of readings that are 
        acquired and stored in the filter buffer for the averaging""",
        validator=truncated_range,
        values=[1, 2500],
        cast=int)
```

bilderbuchi, 2020-11-05:
Hi! Yes, a PR would be great! Can you also check the other properties added in that PR, maybe they also need to be modified. From what I can see, it seems these properties are generic in the 2400, while the 2450 has got separate voltage/current ones, correct? 
also, pinging @paulgoulain who contributed #326 - do you have any further insight here?

paulgoulain, 2020-11-05:
Hi. Sadly no, I apparently made a boo-boo here. I didn't double checked well enough the changes. I'll check with Mathieu if the other functions I implemented in this PR are also compatible with our 2401.

pyMatJ, 2020-11-06:
I checked with @paulgoulain and the other functions were redacted with the 2450 syntax as well. I fixed all of them and submitted the PR https://github.com/ralph-group/pymeasure/pull/357

closed 2020-11-06 by bilderbuchi