pymeasure/pymeasure #362: config method in VISAAdapter class (adapter/visa.py) is not supported since pyvisa v1.10

msmttchr, 2020-11-12:
The config method probably needs to be updated to be compatible with latest version of pyvisa.
The values_format has been removed and it is no longer supported.
The change in pyvisa where support has been removed says "* pyvisa: drop Python 2 compat stuff" (28 aug 2018).

bilderbuchi, 2020-11-14:
Thanks for the heads-up, we should fix this before the next release!
A couple of other things we should fix at the same time:
* I found out that we don't see these deprecation warnings in time since we don't have test coverage of the VisaAdapter. We should implement a test that exercises that part of the codebase, maybe using pyvisa-sim that we already pull in for another test, or with some other faked instrument? That should be implemented first to check if it correctly surfaces the warnings in CI.
* We also need to fix the call to ` query_values(command)`, probably by using [`query_ascii_values(command)`](https://github.com/pyvisa/pyvisa/blob/master/CHANGES#L37-L48). 
* I'm not sure if our `.binary_values` should use `read_binary_values` instead of ` read_raw` -- we should take a look at that.

cjermain, 2021-02-07:
Thanks for raising these. I believe they are fully addressed in #391. Please let us know if not.

closed 2021-02-07 by cjermain

msmttchr, 2021-02-07:
Looks good to me. Thanks