pymeasure/pymeasure #364: Add documentation about Channel implementations

bilderbuchi, 2020-11-16:
For instruments having channel functionality, it would be good to outline a preferred approach for implementing a Channel class to do that. Much of the content is already in here in various comments, the explanation in https://github.com/pymeasure/pymeasure/pull/307 should suffice probably.

moritzj29, 2021-03-11:
Hi, since this was an issue for me as well when I wrote my first instrument driver, I'll give it a shot and try to write some documentation. At least as a basis for discussion.

## Instruments providing multiple channels
Some instruments provide multiple measurement channels. Each channel has the same functionality, e.g. the same controls. Since Python is object orientated, this is best implemented in providing in an additional "channel" class. The instrument holds instances of this class for every available channel. 

```
class Extreme5000(Instrument):

    def __init__(self, resourceName, **kwargs):
        # code to initialize
        self.ch1 = Extreme5000_Channel(self, 'ch1')
        self.ch2 = Extreme5000_Channel(self, 'ch2')  

class Extreme5000_Channel():
    # Controls, Settings, Measurements on Channel level
    # Instrument.control(...)
    # Instrument.setting(...)
    # Instrument.measurement(...)

    # provide methods used by Instrument.control/setting/measuerment
    # use commuincation methods of parent instrument
    # add channel specific parts to commands
    def __init__(self, parentInstrument, channel, **kwargs):
        self._parentInstrument = parentInstrument  # is weakref recommended here??
        self.channel = channel  # ToDo: validate channel specification

    def values(self, command, **kwargs):
        return self._parentInstrument.values(
            "ch%d:%s" % (self.channel, command), **kwargs)

    def ask(self, command):
        self._parentInstrument.ask("ch%d:%s" % (self.channel, command))

    def write(self, command):
        self._parentInstrument.write("ch%d:%s" % (self.channel, command))

    def read(self):
        self._parentInstrument.read()
```

To address a specific channel in the communication with the instrument, usually some command prefix or postfix is necessary. The communication methods (`write`,`read`,...) are implemented in the Channel class to automatically add the channel to the command.

### Example Instruments:
- Tektronix AFG 3000 https://github.com/pymeasure/pymeasure/blob/07cf309b8d113931525c8cc216cc704f2b7ff413/pymeasure/instruments/tektronix/afg3152c.py#L127-L144
- Agilent 4156
- Agilent B1500 (channels added dynamically depending on instrument configuration)
- ...

## Multifunction Instruments (??)
Some instruments provide multiple functionalities within the same instrument, e.g. an oscilloscope and a function generator. Both instruments use the same commuincation channel but apart from that, have nothing in common. If universal communication protocols like VISA or serial are used, the sub-instruments can easily be implemented as separate instruments (different instrument classes). This is the preferred way.
As soon as the there are also settings relevant for both instruments or if the communication relies on a common interface, e.g. a DLL instance, a common instrument instance is necessary. This should be implemented similarly to an instrument with multiple channels. Only this time the channels are instances of different classes.
```
class ExtremeMultiTool(Instrument):

    def __init__(self, resourceName, **kwargs):
        # code to initialize
        # settings, controls for the entire instrument
        self.oscilloscope = ExtremeOscilloscope(self)
        self.fgenerator = ExtremeFGenerator(self)  

class ExtremeOscilloscope():
    # Controls, Settings, Measurements
    # Instrument.control(...)
    # Instrument.setting(...)
    # Instrument.measurement(...)

    # provide methods used by Instrument.control/setting/measuerment
    # use commuincation methods of parent instrument
    # add channel specific parts to commands
    def __init__(self, parentInstrument, **kwargs):
        self._parentInstrument = parentInstrument

    def values(self, command, **kwargs):
        return self._parentInstrument.values(
            "osc:%s" % (command), **kwargs)

    def ask(self, command):
        self._parentInstrument.ask("osc:%s" % (command))

    def write(self, command):
        self._parentInstrument.write("osc:%s" % (command))

    def read(self):
        self._parentInstrument.read()

class ExtremeFGenerator():
    # as ExtremeOscilloscope, modify specifier
```
### Examples
- NI VirtualBench
- ...

## ? Instruments with custom adapter ?
I think this makes also sense to add to the documentation. Instruments that do not support an universal communication protocol but e.g. a DLL. Example is again NI Virtual Bench, also newer stuff from Thorlabs (I think I haven't created a PR for that yet).
- Class for "Adapter" holding the DLL instance
- Instrument class taking adapter and instrument specifier/address to create an instrument instance
- why separate Adapter and Instrument? Theoretically, one can use multiple instruments but only one instance of the DLL is allowed at a time.

---
When writing the B1500 driver, I used `weakref` to hold the parentInstrument instance in the Channel class. Apparently, this should be better for garbage collection. This is basically a cyclic reference since both classes hold instances of each other. Do you agree that `weakref` should be used in those cases? I don't see it for example for the Tektronix insturment.

This is still WIP, but please add ideas and comments.

byzantic, 2021-04-27:
I came across an issue with the proposed method for providing instrument channels, or sub-instruments. The technique outlined above if fine if commands are _always_ preceded by the same prefix, but this is not always the case. Many PSUs for example will use several different prefixes, such as
`SOUR<N>:` and `OUTP:<N>`

Hence I found it necessary to declare the channel parametrisation in the instrument controls themselves. Unfortunately, this tends to clash with the way controls are declared in the `Instrument` package. For example, if I were to use a `%s` channel parameter in a control that also required a `%s` argument, there is confusion. I wanted to use the more modern `{key}` formatting parameters with the `format` method, but this has two problems:
* 1. The `Instrument` methods would need to be changed to understand the format specifiers
* 2. It requires two stages of format interpolation, or partial interpolation, which is not possible with `format` (though it is possible - see [this stack exchange](https://stackoverflow.com/questions/11283961/partial-string-formatting))

So the technique below uses a horrible double stage of string interpolation. The first in the `Instrument.control` is unchanged. The second uses the `format` method in the communication routines that specialise the channel, then hand off to the base instrument routines.

I'm not very happy about it, but it does work. Anyone got any better ideas?

```python
class Extreme5001_Channel():

    def __init__(self, parentInstrument, name, **kwargs):
        self._parentInstrument = parentInstrument 
        self._name                    = name             

    def ask(self, command):
        s = command.format(name=self._name)
        return self._parentInstrument.ask(s)

    voltage_range = Instrument.measurement(
             'MEAS:VOLT:RANG? "{name}"',
             'MEAS:VOLT:RANG "{name}", %f',
             """Reads voltage""",)
```

moritzj29, 2021-04-28:
This is indeed a more difficult case...
If I understand correctly, your channel specification appears at different locations within the SCPI commands?

My first idea was to set up the controls individually for each channel, so setting the channel name in the `get_command` and `set_command` already. But this is even more complicated than your approach, since it needs to get the get and set methods back from the property returned by `Instrument.control`:

```
class Extreme5000(Instrument):

    def __init__(self,resourceName, name,**kwargs):
        super().__init__(
            resourceName,
            "Extreme 5000",
            **kwargs
        )
        self._name = name

   control1=property(
      lambda self: Instrument.control("TEST:ABC:{}".format(self._name),"TEST:ABC:{} %s".format(self._name),"Docstring").__get__(self), 
      lambda self, val: Instrument.control("TEST:ABC:{}".format(self._name),"TEST:ABC:{} %s".format(self._name),"Docstring").__set__(self,val)
      ) 
```
This seems to work, but is everything else but elegant since the `Instrument.control` is needed twice. It would be easier if the plain `fget` and `fset` methods would be available, so one would be more flexible in setting up the property:

```
   control1=property(
      lambda self: Instrument.fget(self,"TEST:ABC:{}".format(self._name),"Docstring"), 
      lambda self, val: Instrument.fset(self,val,"TEST:ABC:{} %s".format(self._name))
      ) 
```
But still much more to code than with your double string formatting approach... So, a lot of text for nothing :D

neal-kepler, 2021-08-06:
When I encountered the problem @byzantic raised I just implemented the outlier methods separately as special cases using @property because there were only  few commands not using the common preterm (i.e. there were  many source<n>, but only a few output<n>). 

```
class Channel():
    """ Implementation of a BN765 fast pulse generator channel.
     """

    BOOLS = {True: 1, False: 0}

    load_impedance = Instrument.control(
        'LOAD:IMP?', 'LOAD:IMP %.4E',
        """A mixed parameter that sets the load impedance. Any float between
        [50, 1e8] are allowed as well as 'MIN' and 'MAX'.""",
    )

    period = Instrument.control(
        "PERiod?", "PERiod %.4E",
        """ A floating point property that controls the period of the output1
        waveform function in seconds, ranging from 3e-5 s to 8 s. Can be set
        and overwrites the frequency for *all* waveforms. If the period is
        shorter than the pulse width + the edge time, the edge time and pulse
        width will be adjusted accordingly. """,
        validator=strict_range,
        values=[1.25e-9, 8],
    )

    frequency = Instrument.control(
        "FREQuency?", "FREQuency %g",
        """ A floating point property that controls the frequency of the output1
        waveform function in Hz, ranging from 0.125 Hz to 1.25e8 Hz. Can be set
        and overwrites the frequency for *all* waveforms. If the period is
        shorter than the pulse width + the edge time, the edge time and pulse
        width will be adjusted accordingly. """,
        validator=strict_range,
        values=[.125, 1.25e8],
    )

    burst_ncycles = Instrument.control(
        "BURS:NCYC?", "BURS:NCYC %d",
        """ An integer property that sets the number of cycles to be output
        when a burst is triggered. Valid values are 1 to 4294967295. This can be
        set. """,
        validator=strict_range,
        values=[1, 4294967295],
    )

    init_delay = Instrument.control(
        "INITDELay?", "INITDELay %g",
        """ This parameter is the group delay from the Trigger IN signal that all
         the pulses associated to the output N have in common.""",
        validator=strict_range,
        values=[0, 1e6],
    )

    invert = Instrument.control(
        "INV?", "INV %s",
        """ A boolean property that sets inversion on ('ON') or off ('OFF') 
        the output 1 of the function generator. Can be set. """,
        validator=strict_discrete_set,
        values=BOOLS,
        map_values=True
    )

    amplitude = Instrument.control(
        "VOLT:LEV:IMM:AMPL?", "VOLT:LEV:IMM:AMPL %g",
        """ This control reads and sets the amplitude of the source 1 output. 
        Range is 0.01-5 V""",
        validator=strict_range,
        values=[0, 5],
    )
[... 79 lines cut ...]
    phase = Instrument.control(
        'phase?', 'phase %.4E',
        """Float paramer which controls the phase of the pulse for the given channel.
        Alternative way to set the delay."""
    )

    def __init__(self, instrument, pulse_number):
        self.instrument = instrument
        self.pulse_number = pulse_number

    def values(self, command, **kwargs):
        """ Reads a set of values from the instrument through the adapter,
        passing on any key-word arguments.
        """
        return self.instrument.values("pulse%d:%s" % (
             self.pulse_number, command), **kwargs)

    def ask(self, command):
        self.instrument.ask("pulse%d:%s" % (
            self.pulse_number, command))

    def write(self, command):
        self.instrument.write("pulse%d:%s" % (
            self.pulse_number, command))
```

neal-kepler, 2021-08-06:
@moritzj29 With regards to weakref, that seems like the correct way to do it, but since python 3.4 and [pep 442](https://www.python.org/dev/peps/pep-0442/) cyclic references don't break garbage collection.

closed 2022-11-19 by bmoneke