pymeasure/pymeasure #366: Fixed Current Measurement range in B1500A instrument not working

HansMalo, 2020-11-18:
Hey guys, 
awsome software. I use it for my B1500A. There is just one issue: when trying to use a fixed current range (not limited auto, or auto) the software does not recognize the command even this should be taken into account by the procedure. in the current_ranges class  there is a look for the "-" in front of the range number selector, but the values are not included in the dictionary. I did not find the result in this case since the building of the ranges dictionary is very complicated.

bilderbuchi, 2020-11-19:
Thanks for the praise. I'll ping the original implementor of this instrument, @moritzj29. This is easily the most complex instrument in our codebase, he's probably much more efficient in finding out what's going wrong.

moritzj29, 2020-11-19:
Hello,

it's great to hear that somebody else is really using this B1500 library 🥳 
You're totally right, the fixed ranges could not be selected due to an error in the code: Accidentally, I did not add the negative indices as possible selections to the corresponding SMUs. Therefore these were not allowed.

I addressed this in PR #369. In my setup I can select fixed ranges now. I did not do thorough testing yet, since I'm not in the lab until next week. 

Hopefully, one of the maintainers can merge this soon.

Have a nice evening!

bilderbuchi, 2020-11-19:
Closed by #369, please test this as thoroughly as you can now that this is merged. :-)

closed 2020-11-19 by bilderbuchi

HansMalo, 2020-11-19:
Hello,  Maybe I have the opportunity tomorrow.Thanks for the fix.