pymeasure/pymeasure #381: having a communication error with instrument Keithley2000

wolfg-2020, 2020-12-14:
I'm new to pymeasure and pyvisa. I've been trying to communicate a multimeter Keithely 2000 via a Keysight GPIB adapter or RS232 to USB cable, but both failed.
I have installed pyvisa (1.11.3) and pymeasure (0.8.0) as well.
==> when I use the RS232 to USB cable, 
I use below commands:
import pymeasure
from pymeasure.instruments.keithley import Keithley2000
dmm = Keithley2000("COM::3")
[image]

and have this error message:
raise NotImplementedError("Please upgrade PyVISA to version 1.8 or later.")
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

==> when I use GPIB adapter, I wrote below commands:
import pymeasure
from pymeasure.instruments.keithley import Keithley2000
dmm = Keithley2000("GPIB::11")

and have a similar error message:
aise NotImplementedError("Please upgrade PyVISA to version 1.8 or later.")
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

I don't have an NI Labview installed on my computer, thus an NI-VISA not present. Does anybody know if this is the problem that I'm not able to connect an instrument via PyVISA or Pymeasure ? Thanks for any suggestion~~

bilderbuchi, 2020-12-15:
I suspect this is because of a deprecation in pyvisa - they  [deprecated](https://pyvisa.readthedocs.io/en/latest/faq/faq.html#error-visa-module-has-no-attribute-resourcemanager)/renamed the `visa` package (that you import) to `pyvisa`. As I understand it, this was then removed in pyvisa 1.11, so our check in pymeasure 0.8 fails for the wrong reason. We have since fixed this in our master, but this is not yet released.

You can confirm my suspicion by running `import visa` and `visa.__version__` and post the result. If this fails, you can try downgrading to pyvisa 1.10 for now, which should avoid that problem.

bilderbuchi, 2020-12-15:
In any case, then you probably also need to install a VISA implementation (`pyvisa-py` is a Python-only implementation) for pyvisa to work properly, and then you maybe/possibly need to give it another [resource identifier](https://pyvisa.readthedocs.io/en/latest/introduction/names.html#), probably `ASRL3::INSTR`, as VISA implementations differ in what they accept as a valid identifier (e.g. `COM3` would be fine with NI VISA, see https://github.com/pyvisa/pyvisa-py/issues/230).

wolfg-2020, 2020-12-22:
> I suspect this is because of a deprecation in pyvisa - they [deprecated](https://pyvisa.readthedocs.io/en/latest/faq/faq.html#error-visa-module-has-no-attribute-resourcemanager)/renamed the `visa` package (that one imports) to `pyvisa`. As I understand it, this was then removed in pyvisa 1.11, so our check in pymeasure 0.8 fails for the wrong reason. We have since fixed this in our master, but this is not yet released.
> 
> You can confirm my suspicion by running `import visa` and `visa.__version__` and post the result. If this fails, you can try downgrading to pyvisa 1.10 for now, which should avoid that problem.

Yes, you were right. Here below is the result as I ran it:

C:\Users\it\PycharmProjects\First_GUI\venv\Scripts\python.exe D:/Python/myProjects/InstrumentCont/Model2000.py
C:\Users\it\PycharmProjects\First_GUI\venv\lib\site-packages\visa.py:13: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  warnings.warn(

Process finished with exit code 0

I have PyVISA 1.11.3 installed on my computer. But I never communicated with an instrument successfully with it. Do you know what the problem is ?

bilderbuchi, 2020-12-22:
As I suggested above, please try
> downgrading to pyvisa 1.10 for now, which should avoid that problem.

and then we see if your problem is fixed, or else what kind of error is given.
If it still does not work see my [other comment](https://github.com/pymeasure/pymeasure/issues/381#issuecomment-745122342) for things to try.

wolfg-2020, 2021-01-12:
Hi, sorry for having suspended this topic for so long. But I've tried to downgrading pyvisa from 1.11.3 to 1.10, and run some lines: (As checked, visa 1.0.0 is on my computer, too. Is this a problem?)

import pyvisa as visa
rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")
rm.list_resources(). 

But I still have an "AttributeError: function 'viOpen' not found. 

If change the first line as "import visa", I'll get another attribute error: 
AttributeError: module 'visa' has no attribute 'ResourceManager'.

Actually now I have NI MAX installed, from which I can see my GPIB instrument from the list of VISA Interactive Control. At the same time, I also check the visa32.dll with a tool named eXescope, from which viOpen can be seen. So it's really confusing to have that attribute error. I guess this is not a problem of pymeasure, but a problem of visa or pyvisa. Can you give me some advice on this problem ? Thank you so much!

closed 2021-01-12 by bilderbuchi

wolfg-2020, 2021-01-13:
Thank you all the same for your advice~~I've updated more information to one of the issues you mentioned (pyvisa#385). Thanks again.

wolfg-2020, 2021-01-14:
Hi, I'm coming again.
Now that I'm able to communicate Keithley 2000 with pyvisa, but not with pymeasure. I'm wondering why.

Here's the lines I ran with pyvisa, and it works well with the os.add_dll_directory():

import pyvisa as visa
import os
os.add_dll_directory("C:/Program Files/Keysight/IO Libraries Suite/bin")
os.add_dll_directory("C:/Program Files (x86)/Keysight/IO Libraries Suite/bin")
rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")
print(rm.list_resources())
my_instrument = rm.open_resource('GPIB1::6::INSTR')
print(my_instrument.query('*IDN?'))

but when it comes to call pymeasure commands:

from pymeasure.adapters import VISAAdapter
from pymeasure.instruments.keithley import Keithley2000
adapter = VISAAdapter("GPIB::6")
multimeter = Keithley2000(adapter)
multimeter.id

I have this error message:
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

details:
Traceback (most recent call last):
  File "D:/Python/myProjects/InstrumentCont/Model2000.py", line 15, in <module>
    adapter = VISAAdapter("GPIB::6")
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pymeasure\adapters\visa.py", line 51, in __init__
    raise NotImplementedError("Please upgrade PyVISA to version 1.8 or later.")
NotImplementedError: Please upgrade PyVISA to version 1.8 or later.

My PyVISA was downgraded to 1.10 as you advised last time. But the error persists when it is upgraded to 1.11.3. It doesn't look like a problem from PyVISA any more. Would you please give me some suggestion ? Thank you.

adrzazga, 2021-01-14:
Dear Christoph,

Thank you very much for your feedback! I am really delighted that you answered me.
I am an Application Engineer at Tespol Engineering responsible for the Keithley's instrumentation.
As you probably know the KUSB from Keithley is the most popular GPIB to USB interface among the Keithley's community (but obviously not the only one). The problem is that the KUSB-488 native driver is not compatible with the native NI 488.2 drivers.
It means that if you have NI 488.2 installed you have to uninstall the NI 488.2 and install an original driver for the KUSB adapter (called NI Command Compatible as you can see below).
It has two different versions: NI 488 (with modifications x32, x64 bits) and Keithley (only 32 bits).
So, from my point of view the best way to check the pymeasure is to install only the driver for the KUSB and run the Anaconda enviroment with the pymeasure installed according to a proposal given by bilderbuchi.

Additionally, I would like to tell you that I will be testing this interface today. Please let me know if you have an additional questions about the instrumentation or documentation related to the Keithley's instruments.

Best regards

ADR

bilderbuchi, 2021-01-15:
hey folks! Why are you posting in two different discussions interchangeably (cf. #222)? This is becoming confusing to track...