pymeasure/pymeasure #382: Qthread destroyed error

neal-kepler, 2020-12-22:
I'm not sure if this has been raised before ( I seem to recall it but I can't find the issue). But, I not infrequently get `QThread: Destroyed while thread is still running` errors which cause the software to crash. I looked through the code and it looks like the only QThreads are the monitors. In addition I frequently end up with a bunch of stale python processes and I think that is related (like filling up 2 Gb of RAM of stale processes). So to get around this I added a flag to Monitor, `self.stop` which can break out of the run loop if set to True.

```
class Monitor(QtCore.QThread):
    """ Monitor listens for status and progress messages
    from a Worker through a queue to ensure no messages
    are losts
    """

    status = QtCore.QSignal(int)
    progress = QtCore.QSignal(float)
    log = QtCore.QSignal(object)
    worker_running = QtCore.QSignal()
    worker_failed = QtCore.QSignal()
    worker_finished = QtCore.QSignal()  # Distinguished from QThread.finished
    worker_abort_returned = QtCore.QSignal()

    def __init__(self, queue):
        super().__init__()
        self.queue = queue
        self.stop = False

    def run(self):
        while not self.stop:
            self.data = self.queue.get()
            if self.data is None:
                break
            topic, data = self.data
            if topic == 'status':
                self.status.emit(data)
                if data == Procedure.RUNNING:
                    self.worker_running.emit()
                elif data == Procedure.FAILED:
                    self.worker_failed.emit()
                elif data == Procedure.FINISHED:
                    self.worker_finished.emit()
                elif data == Procedure.ABORTED:
                    self.worker_abort_returned.emit()
            elif topic == 'progress':
                self.progress.emit(data)
            elif topic == 'log':
                self.log.emit(data)

        log.info("Monitor caught stop command")
```

Then where the Manager instance cleans up the self._monitor instance, I set the flag, give it a chance to stop (`self.monitor.wait(100)`), record if it finishes properly in wait for posterity, and then just terminate.  (I know it's dangerous, but I don't know why the QThread isn't exiting and I can't have my code crashing) :

```
    def _clean_up(self):
        self._worker.join()
        self._monitor.stop = True
        success = self._monitor.wait(100)
        if not success:
            raise ValueError('Monitor did not properly exit')
        else:
            self._monitor.terminate()
        del self._worker
        del self._monitor
        self._worker = None
        self._running_experiment = None
        log.debug("Manager has cleaned up after the Worker")
```

Any advice or thoughts are appreciated

neal-kepler, 2020-12-22:
I should add I am still testing if this solves the issue. Because this error occurs due to collisions between the garbage collection and the process, I can't deterministically reproduce the error. So far, though, it seems to have fixed it.

msmttchr, 2021-01-14:
I see also the error you reported from time to time. I will give a try at your tentative fix.

msmttchr, 2021-01-15:
I have changed the code as follow:
```
    def _clean_up(self):
        self._worker.join()
        del self._worker
        self._monitor.wait() # Added line
        del self._monitor
        self._worker = None

```
and I do not see the error anymore. Perhaps a new PR can be raised to fix this.

bilderbuchi, 2021-01-15:
@msmttchr did you find a repro procedure/test to trigger this bug?

bilderbuchi, 2021-01-15:
I'm wondering why that `_clean_up` method does not call `self._worker.shutdown()` before joining, that would cleanly clean up the monitor thread by enqueuing a `None` which makes the monitor stop cleanly.
@cjermain @CasperSchippers any ideas?

msmttchr, 2021-01-15:
> @msmttchr did you find a repro procedure/test to trigger this bug? At which point or user action do you get this bug? Alternatively, a log from when that bug happens would be interesting to be able to determine the sequence of events leading to this.

@bilderbuchi I think the issue is related to inherent asynchronous nature of thread execution. In fact, the worker in its `finally` clause send a `None` message to the monitor's queue, so we can't guarantee that when the monitor thread is deleted, he has already read the `None` message and exited. Adding the wait instruction makes the thread waiting for monitor to complete before deleting it. In my Windows environment, the problem is not deterministic and I could more easily reproduce it by launching a long sequence of experiments, so having more chance to hit the critical path. In my opinion, this is something that should be fixed by code inspection and making a regression test is not practical.

ph4as, 2021-01-20:
> I have changed the code as follow:
> 
> ```
>     def _clean_up(self):
>         self._worker.join()
>         del self._worker
>         self._monitor.wait() # Added line
>         del self._monitor
>         self._worker = None
> ```
> 
> and I do not see the error anymore. Perhaps a new PR can be raised to fix this.

I can confirm this also worked for me. I had the issue 9 out of 10 times while developing a new instrument control.
This fix solves it for me.

neal-kepler, 2021-01-21:
Thanks the more elegant fix and more proper description of the problem @msmttchr , I think I may have been conflating it with another issue with multiprocessing in an early version.  Thanks for also test the fix @ph4as .

bilderbuchi, 2021-01-22:
Thanks for checking, folks! I think adding this wait is probably the most sensible thing to do. Is anybody up for submitting a PR?

msmttchr, 2021-01-22:
I am going to do this, but the credit goes to @neal-kepler who found the root cause of the problem.

bilderbuchi, 2021-01-22:
Thank, merged!

closed 2021-01-22 by bilderbuchi