pymeasure/pymeasure #387: Driver for instrument without reading capability

pyMatJ, 2021-01-04:
Hi,
This is a general question regarding the best approach to create a driver for an instrument (here a current amplifier SR570 from Stanford Research Systems) that has no reading capabilities for its own state (AFAIK it does not even return a statement to acknowledge successfully setting a parameter). 
As stated in the manual, "the RS232 interface is configured as listen only", and all the functions only allow to set a given property. In my current implementation, I only use Instrument.setting() methods.  
This creates two problems: the user cannot check the state of a given property (in the eventuality that it was modified from the front interface), and there is no way to keep track of the current state of the instrument.

The first problem cannot be solved, obviously, unless the user manually checks and reset the properties through python if needed. I see however multiple ways to handle the second one, amongst which:
- creating instruments attributes in the code, and setting them accordingly in parallel to the Instrument.setting() 
- creating a dictionary of the instrument's properties
 
What would be the preferred way of dealing with this ? 
Trying to keep track of the changes with minimal intervention from the user implies maybe that the value should be stored during the Intrument.setting() call, maybe using set_processed ? Or using decorators (I'm new to these) to avoid the getter/setter scheme ?

I'm looking forward to your input !

bilderbuchi, 2021-01-04:
Hm, are not your two problems 2 sides of the same coin, really? What are you trying to accomplish with tracking the "state" of the instrument, when that seems impossible as the instrument cannot be queried, and you cannot confirm if a value was changed from the front panel? Are you purely trying to track the "state of remote user commands" so to say? Is that useful to you in operating the device?

I know that Lantz implements the concept of [caching](https://lantz.readthedocs.io/en/0.3/tutorial/using.html#the-cache), which could be similar to what you try to accomplish. If that sounds useful to you, depending on the complexity of the command set, one approach could be implement a custom [adapter](https://pymeasure.readthedocs.io/en/latest/api/adapters.html) based on the VISAAdapter, which records all the commands it receives and only passes on those whose values don't agree with the latest value. Pay attention that flip-flopping values get processed correctly (e.g. `GAIN 1`, `GAIN 5`, `GAIN1`). This does not afford you user-visible checks of set values, though, purely saves communication overhead on repeated commands.

Using `set_process` will be tricky because you will need a stateful check (keeping the last set value), there is currently no nice/elegant way to do this.

pyMatJ, 2021-01-05:
Thanks for the pointers on caching and adapters btw. Yes, these two things are somewhat related. 

> Are you purely trying to track the "state of remote user commands" so to say? Is that useful to you in operating the device?

Yes I think this is actually it, since obviously you can never get any kind of confirmation from the instrument.

The amplifier is usually in a signal processing chain with other instruments, which is why querying is not mandatory. Typically, tracking the "state" of the instruments would be useful in a procedure to know and re-use e.g. a gain value, to be able to provide a physically meaningful quantity during a measurement. This could be done in the procedure itself, but then it is up to every user to ensure that they implement it. Setting the state of the instrument at the beginning of the procedure, assuming the user does not change it _during_ the measurement, discards the need for querying. 

Since I have found no other similar instruments in pymeasure, I was not sure if there was any preferred way of dealing with this. Usually, useful properties like this are always accessible, so I was simply wondering if it was interesting to maintain this possibility within the driver itself. 
After reading your comment, I suppose the most conservative way is to keep the current philosophy of pymeasure (so I'll stick to the Instrument.setting()), and have the driver only do what is required for the instrument operation. It will be up to the user to ensure consistency in a measurement. 

Sorry if this is unclear...

bilderbuchi, 2021-02-16:
> After reading your comment, I suppose the most conservative way is to keep the current philosophy of pymeasure (so I'll stick to the Instrument.setting()), and have the driver only do what is required for the instrument operation. It will be up to the user to ensure consistency in a measurement.

> Sorry if this is unclear...

Not at all, I agree that that is probably the best way forward.

bilderbuchi, 2021-02-16:
This issue has more of a discussion/usage question character, so I will move it to our new discussion forum, which is a better fit than the issue tracker.

closed 2021-02-16 by bilderbuchi