pymeasure/pymeasure #389: PyVISA vs PyMeasure

froufrounette, 2021-01-11:
Hi all,

I am trying to read and write using PyMeasure since much of the functionalities are already available. However, when I want to write/read an instrument created in PyMeasure, I get a time out error. However, if I open my instrument by myself by using : 
rm=pyvisa.ResourceManager()
lock_in=rm.open_resource('TCPIPO:ipadress::port::SOCKET'), 
The write/read method works. 

What is wrong? And how can I use PyMeasure properly.

bilderbuchi, 2021-01-12:
Welcome!
Hm, it's hard to say at this point. Let me throw a couple of questions your way:
Do all the instruments not work? Even if you connect only one? 
What happens if, instead of the defined properties, you use the instrument.read(), .write(), .ask() methods directly, just as you do when doing it directly with pyvisa?
If that does not work, what if you use instrument.adapter.read(), .write(), .ask()?
If that does not work, what if you use instrument.adapter.connection.read(), .write(), .ask()?
Do you have settings regarding line termination characters? If yes, you can specifiy those when creating the instruments using the `read_termination` and  `write_termination` keyword arguments.

froufrounette, 2021-01-12:
Hi, 

thank you. It was missing the read/write termination

bilderbuchi, 2021-01-12:
So, did you solve your problem?

froufrounette, 2021-01-12:
Yes, for now it works.

Thank you again

bilderbuchi, 2021-01-12:
Great! You're welcome. :-)

closed 2021-01-12 by bilderbuchi