pymeasure/pymeasure #393: Information on plots

msmttchr, 2021-01-26:
I would like to plot two traces at the same time during an experiment, for example:
- plot a curve based on measurement at same time with a predicted one to appreciate the differences between model and measurement
- Plot a curve with experimental data and, when finished, make plot with a calculated interpolation curve 

Is this possible with pymeasure ?

CasperSchippers, 2021-02-11:
This can be quite challenging at the moment I think. As far as I know plotting two curves is not really possible.

It would of course be possible to add an additional data column to the experiment with a simulated curve, but then you'd still have to switch between the two columns (i.e. between the simulated and the measured curve). Another option would be to have first an "experiment" where you simulate the curve and then an experiment where you measure. But both are not ideal of course.

For the interpolated curve at the end, one possibility would be to add a column which will be NAN during the experiment, and then basically append the interpolation to the data (while then leaving the other real-data columns NaN). But this would of course pollute your data file with a lot of rows at the bottom just for the interpolated data (probably more rows than the data itself).

So for both cases there is i think not a really good solution at the moment.

msmttchr, 2021-02-11:
@CasperSchippers thanks for the feedback. I hope that #383 can progress in some way, so that people could redefine plot view.

closed 2021-02-11 by msmttchr