pymeasure/pymeasure #400: How to use sourcemeter more than one at the same time?

Byeoungju-Lee, 2021-02-15:
Hello.
I have two Keithley 2400 sourcemeters and they are connected by GPIB-ENET 100 of National Instruments.

I want to read current at the same time but there is a delay when I ran my code.
```
import numpy as np
import pandas as pd
from pymeasure.instruments.keithley import Keithley2400
from pymeasure.instruments import Instrument
from pymeasure.log import console_log
from pymeasure.display.Qt import QtGui
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import (
    Procedure, FloatParameter, unique_filename, Results
)

keithley = Keithley2400("GPIB::24")
keithley2 = Keithley2400("GPIB::25") #작동시키면 REM 모드 ON.
source_drain_volt=0.1
voltage_max=40
voltage_min=-60
gate_voltage_range=np.linspace(voltage_max, voltage_min,2*(voltage_max - voltage_min) + 1)
keithley.source_delay, keithley2.source_delay=[0.1, 0.1]

keithley.apply_voltage()
keithley2.apply_voltage()
for volt in gate_voltage_range:
    keithley.source_voltage=0.1
    
    keithley2.source_voltage=volt
    
    keithley.compliance_current = 1.05
    
    keithley2.compliance_current = 1e-6
    
    keithley.enable_source()
    
    keithley2.enable_source()
    
    keithley.measure_current(nplc=1)
    
    keithley2.measure_current(nplc=1)
    
    print(0.1,' ',keithley.current)
    print(volt,' ',keithley2.current)
keithley.shutdown()
keithley2.shutdown()
```
What shoud I do to work 2 sourcemeters simultaneously?

I attached two pictures about my settings.
I want to be it is helpful to advise my situation.
[image]
[image]

bilderbuchi, 2021-02-15:
Hi! What do you observe in detail when you say "there is a delay"? From looking over it this code looks alright to me. Does it work as expected when you use one or the other keithley alone?

Byeoungju-Lee, 2021-02-16:
Sorry for making you confused, @bilderbuchi .
My problem is when i run this code, there is a delay in
```
keithley.current
keithley2.current
```
and delay also occur between real sourcemeter panels,24 and 25.

But I want to make there is no delay between keithley.current and keithley2.current.
I attach a gif file and wish it to help you understand my problem. 
[image]

bilderbuchi, 2021-02-16:
OK, I see. As you do the executions sequentially, there is some natural delay as first one, then the other device is queried for a new value. 
You can benchmark how much time a "round trip" will take if you connect to one of the Keithleys in IPython, then executing with the [timeit magic](https://ipython.readthedocs.io/en/stable/interactive/magics.html#magic-timeit)
```
%timeit Keithley.current
```
Now you will know what kind of delay you incur from sequential execution. If that is unacceptable, maybe the triggering functions in a Keithley2400 are faster - you could configure a current measurement in each, an then trigger both, maybe this is then faster.

Finally, and this depends on how your GPIB adapter works and your Python skills, you could query both devices in separate threads, which means the device #2 query will not have to wait until the device #1 query is finished. I recommend using a [ThreadPoolExecutor](https://docs.python.org/3.9/library/concurrent.futures.html#threadpoolexecutor)

As this is more of a usage question, not a bug report, I will move this into our discussion forum

closed 2021-02-16 by bilderbuchi