pymeasure/pymeasure #409: Unify adapter classes

bilderbuchi, 2021-02-26:
As PyVISA has become more powerful, we should 
* check if we cannot unify VISAAdapter, SerialAdapter and PrologixAdapter, or at least make the latter two just thin wrappers around the former
* check if we cannot replace the VXI11Adapter with VISAAdapter, as pyvisa apparently supports [VXI instruments](https://pyvisa.readthedocs.io/en/latest/introduction/names.html#visa-resource-syntax-and-examples), and python-vxi11 seems [essentially unmaintaineed](https://github.com/python-ivi/python-vxi11/graphs/contributors)
* Adapt the API of the TelnetAdapter to the others.
* Add tests for the various adapters

The homogenised adapter API makes instrument support easier (e.g. regarding safe/valid keywords), and leaning onto PyVISA more would probably reduce our maintenance effort.

msmttchr, 2021-03-05:
Unfortunately Prologix is not supported by `pyvisa `and this is, I believe, the only reason why `SerialAdapter `exists.

bilderbuchi, 2021-12-03:
* [ ] Make VISAAdapter.ask also use our own write and read instead of delegating to pyvisa.query. This makes it easier to be homogeneous across adapters, and also enables cleaner use of custom read/write methods (see TopticaAdapter). [pyvisa docs](https://pyvisa.readthedocs.io/en/latest/_modules/pyvisa/resources/messagebased.html#MessageBasedResource.query)
* [ ] roll out the use of `query_delay` across all adapters.

bilderbuchi, 2022-05-14:
#621 adds an instrument that would also profit (i.e., eliminate duplication) from having a unified `ask` method.

PR #660 (2022-10-12): Adapter rework
Reworking the instrument/adapter. This pull request just focuses on the official adapters, Instrument class and keeping the code base in a working condition.

Fixes #567, fixes #512, fixes #409, fixes #219.

Roadmap of different parts (PRs)
----------
1. Preliminary tests of instruments/adapters in the current state #665 
2. This pull request with changes to the official adapters and minimum changes to the instruments to use these
3. A test code generator.
4. Pull requests for changes in individual instruments (removing custom adapters, where applicable)

Open questions
-------- [... cut]
closed 2022-10-12 by bilderbuchi