pymeasure/pymeasure #413: Welcoming message - Agilent33522A

CristoJV, 2021-03-12:
Hi!
I'm trying to use the class Agilent33500 to control the Agilent33522A using TelnetAdapter.
The issue I'm facing is the following.

When the connection is established the instrument returns a welcoming message:

> Welcome to Agilent's 33522A Waveform Generator
> 33522A>

This affects the following read operations generating inconsistency in the data received.
I modify the ```TelnetAdapter``` constructor to read in a non-blocking operation these few bytes using its self ```TelnetAdapter.read()``` function, and I worked like charm. I believe this change does not affect any other instrument.
```python
def __init__(self, host, port=0, query_delay=0, preprocess_reply=None,
                 **kwargs):
        super().__init__(preprocess_reply=preprocess_reply)
        self.query_delay = query_delay
        safe_keywords = ['timeout']
        for kw in kwargs:
            if kw not in safe_keywords:
                raise TypeError(
                    f"TelnetAdapter: unexpected keyword argument '{kw}', "
                    f"allowed are: {str(safe_keywords)}")
        self.connection = telnetlib.Telnet(host, port, **kwargs)
        #Added this two lines
        time.sleep(self.query_delay)
        self.read()
```

msmttchr, 2021-03-15:
Hello, have you tried using `VISAAdapter`, since you are using the device via LAN, you should supply an address like this: `TCPIP::<instrument ip address>` (see also https://pyvisa.readthedocs.io/en/latest/introduction/names.html), that is:
`generator = Agilent33500("TCPIP::<instrument ip address>")`.
This could probably solve your problem, assuming `pyvisa` discard the welcome string.

bilderbuchi, 2021-03-15:
If it's only this instrument that shows the welcome message, another option would be to add this startup treatment to the init of a specific instrument implementation, e.g. something like:
```
class Agilent33522A(Agilent33500):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        is isinstance(self.adapter, TelnetAdapter):
            #Discard welcome message of a telnet connection
            time.sleep(self.adapter.query_delay)
            msg = self.read()
            assert msg.startswith('Welcome to')  # make sure you discard the correct message
```